TheImagingSource/tiscamera #341: How to control the PixelFormat by USB3 Vision ?

oka1125, 2020-04-18:
I am using ''DFK37BUX273" camera connected to Ubuntu(18.04) PC.
The spec of this camera is as follows:
  interface=USB3.0
  width=1440
  height=1080
  fps=238
And the gstreamer's command is as follows:
  gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1440,height=1080 ! videoconvert ! ximagesink

Now I want to increase the real fps by cropping regions of interest.
Its manual saids that the pixel format is controlled through the PixelFormat when using USB3 Vision.
https://s1-dl.theimagingsource.com/api/2.5/packages/documentation/manuals-trm/trmdfk37bux273/b14c2a17-22f5-5f52-8884-c27cb993306d/trmdfk37bux273.en_US.pdf
  --> page 13 "5.1 Sensor Readout Control"

But how can I control the PixelFormat by USB3 Vision ?
Please let me know how to control the PixelFormat by USB3 Vision on Ubuntu 18.04 PC which is connected with DFK37BUX273 camera.

oka1125, 2020-04-19:
I also tried following command:
    1. gst-launch-1.0 tcamsrc serial=43810371 ! video/x-bayer,format=rggb,width=1440,height=1080 ! bayer2rgb ! videoconvert ! ximagesink
    2. gst-launch-1.0 tcamsrc serial=43810371 ! video/x-bayer,format=rggb,width=1280,height=720 ! bayer2rgb ! videoconvert ! ximagesink

1. is successed, but 2 is failed.
How can I change the resolution ?
How can I use USB3 Vision ?  (I guess that above gst pipeline use uvc backend, is it right ?)

TIS-Stefan, 2020-04-20:
If the camera is used in the way you showed, then the camera is driven by video for linux (v4l2) instead of USB3Vision.
You can call
tcam-ctrl -c 43810371 
which will give you a list of supported video format resolutions and their available frame rates:
```
video/x-bayer,format=rggb,width=1440,height=1080,framerate={2500000/10593,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=640,height=480,framerate={600/1,480/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=1440,height=1080,framerate={120/1,60/1,30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=640,height=480,framerate={600/1,480/1,5000000/20833,120/1,60/1,30/1,15/1,5/1,1/1}
```
As you can see, a 1280x720 format is not availabe, if v4l2 is used. In order to use the camera as USB3Vision device with more available frame rates and video formats, you do check out the development tree of tiscamera and set some variables for cmake:
```
git clone
git clone https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
sudo ./scripts/install-dependencies.sh --compilation --runtime
git checkout development
mkdir build
cd build
cmake -DBUILD_ARAVIS=ON -DTCAM_ARAVIS_USB_VISION=ON -DBUILD_V4L2=OFF ..
make -j 32
sudo make install
```
(The steps you already did do not need to be repeated) 
tcam-ctrl shows now:
```
$tcam-ctrl -l
Available devices:
Model		Type	Serial

DFK 37BUX273	Aravis	38910080

$ tcam-ctrl -c 38910080
Available gstreamer-1.0 caps:
video/x-bayer,format=rggb,width=256,height=4,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=320,height=240,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=360,height=280,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=544,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=640,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=352,height=288,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=576,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=720,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=960,height=720,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=1280,height=720,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=1440,height=1080,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb,width=[256,1440],height=[4,1080],framerate=[1/1,236667/1000];
video/x-bayer,format=rggb16,width=256,height=4,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=320,height=240,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=360,height=280,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=544,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=640,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=352,height=288,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
video/x-bayer,format=rggb16,width=576,height=480,framerate={1/1,2/1,3/1,4/1,5/1,6/1,7/1,8/1,9/1,10/1,11/1,12/1,13/1,14/1,15/1,16/1,17/1,18/1,19/1,20/1,30/1,40/1,50/1,60/1,70/1,80/1,90/1,100/1,150/1,200/1,236667/1000};
[... cut ...]
video/x-bayer,format=rggb16,width=[256,1440],height=[4,1080],framerate=[1/1,236667/1000]
```
You can select any resolution, even resolutions, that are not shown there. 
```
gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1440,height=720 ! videoconvert ! ximagesink
```
But I see, the maximum frame rate stays at 236 frames per seconds, regardless, which resolution is set. This will be fixed with the tiscamera version  0.12.0. It is already in the pipeline and I just confirmed, that 296 frames per second will be available.

Stefan

oka1125, 2020-04-20:
Dear Mr. Stefan

Thank you for the kind advice.
I'll try it soon.

Oka

oka1125, 2020-04-22:
Dear Mr. Stefan

I tried to rebuild tiscamera library, and I was able to get the same result with your "tcam-ctrl -c 38910080" log.

However, I tried the command "gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1280,height=720 ! videoconvert ! ximagesink",
the opened window freezes soon.

When I specify small width and height (e.g. width=400,height=400),
streaming frames were shown in the opened window.

In the tiscamera tutorial, it is described that Aravis works as user process (not kernel thread) and this results in not realtime cpu priority.
I changed the priority of gst-launch-1.0 command, but it had no effect (i.e. the opened window freezes soon).

What should I do ?

Please note that I am using a Jetson TX2 (not high-performance PC).

TIS-Stefan, 2020-04-22:
Hello

I suggest to try lower frame rates:
"gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1280,height=720, framerate =15/1 ! videoconvert ! ximagesink"
Please keep in mind, the main cpu power consumer is ximagesink. So if the video is not shown, you get all frames. So for live display you may rise the frame rate in steps until you get issues.
I also assume, you Jetson TX2 is powered by the "big" power supply and not by USB 2.0 input. 

Stefan

oka1125, 2020-04-22:
Dear Mr. Stefan

Thank you.
I was able to show the ximagesink window by your suggesting command:
  "gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=1280,height=720, framerate =15/1 ! videoconvert ! ximagesink"

But I met another issue.
I want to set properties (for example, Gain, Trigger Mode, and so on), and I was able to set such properties when backend was V4L2.

Now I can't set any property on Aravis backend. I tried following commands:
      "tcam-ctrl --load 43810371 '{\"properties\"{\"Gain\":\"0.2\"}}'"
        --> this resulted no effect.

And I also tried to set by using c program function (tcam_prop_set_tcam_property),
     but Segmentation fault occurred.

Best regards,
Hiroyuki Oka

TIS-Stefan, 2020-04-22:
Dear Mr Hiroyuki Oka-san
Please check with 
tcam-ctrl -p 43810371 
whether the property names and types are the same.

Sincerely yours
Stefan

oka1125, 2020-04-23:
Dear Mr. Stefan

I already have checked by "tcam-ctrl -p 43810371" whether the property names and types are the same.
But I could not set any property.
 (on the other hand, I could when I used v4l2 backend).

Best regards,
Hiroyuki Oka

TIS-Stefan, 2020-04-23:
Dear Mr Hiroyuki Oka-san

Using Aravis, it is checked, whether properties can be set or not. If Gain Auto is enabled, then setting gain has no effect. I do not dare to ask, but did you disable gain auto?

Best regards
Stefan

oka1125, 2020-04-24:
Dear Mr. Stefan

I did not disable gain auto.
But when I tried to set another property (Trigger Mode, etc.),
I was not able to change its value.
I will look for the cause a bit more.

Best regards,
Hiroyuki Oka

TIS-Stefan, 2020-04-27:
Dear Mr Hiroyuki Oka-san

Today I tested intensively and I found, as you:
`tcam-ctrl --load 38910080 '{"serial":"38910080","properties":{"Gain Auto":"Off","Gain":"0"}}'
`
The line has no effect. I made an error report.
For the development version I have currently no workaround, since the -s option was removed.

Stefan

oka1125, 2020-04-27:
Dear Mr. Stefan

Thank you for the test and making an error report.
I will wait for the fix.

Hiroyuki

TIS-Stefan, 2020-04-28:
Hello Hiroyuki

tcam-ctrl --load 38910080 '{"properties":{"Gain Auto":"Off"}}'
tcam-ctrl --load 38910080 '{"properties":{"Gain":0.2}}'

works. Currently only one property per command call.

Stefan

oka1125, 2020-05-04:
Hello Stefan

Thank you for the kind support.

(1) tcam-ctrl --load MY_SERIAL_ID '{"properties":{"Gain Auto":"Off"}}'
(2) tcam-ctrl --load MY_SERIAL_ID '{"properties":{"Gain":0.4}}'
  -->  (2) works, but (1) does not work.

Since this problem is different with this issue's purpose,
I will create an issue when I need.
Thank you so much.

Hiroyuki

closed 2020-05-04 by TIS-Stefan