TheImagingSource/tiscamera #342: tcamsrc reports 0 latency

dzhura, 2020-04-27:
Hello,

I have a pipeline:
`GST_DEBUG=2 
	gst-launch-1.0 compositor name=comp ! videoconvert ! ximagesink \
	tcamsrc serial=SERIAL_1 ! video/x-raw,format=BGR,width=640,height=480,framerate=1/1 ! comp. \
	tcamsrc serial=18914002 ! video/x-raw,format=BGR,width=640,height=480,framerate=1/1 ! comp. \
	videotestsrc is-live=1 ! video/x-raw,format=BGR,width=640,height=480,framerate=1/1 ! comp.`

When I run it, I get:
0:00:02.523371739   426   0x5565bdfad0 ERROR               GST_PADS gstpad.c:3352:gst_pad_query_latency_default:<comp:src> minimum latency bigger than maximum latency
0:00:02.523472158   426   0x5565bdfad0 WARN              aggregator gstaggregator.c:1733:gst_aggregator_query_latency_unlocked:<comp> warning: Impossible to configure latency: max 0:00:00.000000000 < min 0:00:01.000000000. Add queues or other buffering elements.
WARNING: from element /GstPipeline:pipeline0/GstCompositor:comp: GStreamer error: clock problem.

I found that tcamsrc does not have its own query function uses GstBaseSrc defalt query function, which returns 0 latency according to the documentation. Is there a way to make tcamsrc report the correct latency as done by videotestsrc is-live=1?

TIS-Edgar, 2020-04-28:
Hi,

I looked into this and can confirm this as a bug.

I will try to fix this within the week.
The maximum latency will be NONE, as all cameras have trigger and thus can be 'suspended' indefinitely.

Would it be sufficient for if I pushed the fix to the development branch?

dzhura, 2020-04-28:
Hi Edgar,

It will be great if I can run the pipeline with no issues.

Thank you,
Maxim

TIS-Edgar, 2020-04-28:
Ok,

I have added a query function to tcamsrc. You can find the changes (c69810f390943dede2fcf10a63e59c22cf94dcbd) on the development branch.

I do not get any warnings/error with your pipeline but I also do not get an image from a camera.
What is the result you are expecting?

dzhura, 2020-04-28:
Hi Edgar,

Your commit fixed the problem. To see the images, set alphas to 0.5 (e.g. sink_0::alpha=0.5) in compositor.

Thank you.
Cheers,
Maxim

closed 2020-04-28 by dzhura