TheImagingSource/tiscamera #343: tcamsrc num-buffers=1 does not produce a buffer

dzhura, 2020-04-28:
Hello,

I am trying to make a snapshot from DFK 38UX267 to process it further. It apperas, tcamsrc num-buffers=1 does not produce data.
There is an example pipeline:
`gst-launch-1.0 tcamsrc serial=SERIAL num-buffers=1 ! videoconvert ! jpegenc ! filesink location=image.jpg`
image.jpg is always empty! 

Thanks,
Maxim

TIS-Stefan, 2020-04-28:
Hello Maxim

The pipeline does not work, because the video formats can not be negotiated. Please use a pipeline like

`gst-launch-1.0 tcamsrc num-buffers=30 ! video/x-bayer,format=rggb,width=640,height=480,framerate=30/1 ! bayer2rgb ! videoconvert ! ximagesink`

Stefan

TIS-Stefan, 2020-04-28:
I am not sure about the "rggb". please use tcam-ctrl -c for a list of valid formats

dzhura, 2020-04-29:
I confirm that there is a bug in tcamsrc.

gst-launch-1.0 tcamsrc serial=SERIAL num-buffers=1 ! videoconvert ! jpegenc ! filesink location=image.jpg
Creates an empty file

but
gst-launch-1.0 v4l2src device=/dev/video0 num-buffers=1 ! videoconvert ! jpegenc ! filesink location=image.jpg
Works as expected.

TIS-Stefan, 2020-04-29:
There is no bug. The format provided from tcamsrc can not be read by videoconvert, as I already wrote. I also posted a working pipeline. Did you use that?

Stefan

dzhura, 2020-04-30:
Stefan,

Your pipeline sets num-buffers to 30, but I want to capture one buffer

Maxim

dzhura, 2020-04-30:
Stefan,

`gst-launch-1.0 tcamsrc num-buffers=30 ! video/x-bayer,format=rggb,width=640,height=480,framerate=30/1 ! bayer2rgb ! videoconvert ! ximagesink`
works fine, but

`gst-launch-1.0 tcamsrc num-buffers=1 ! video/x-bayer,format=rggb,width=640,height=480,framerate=30/1 ! bayer2rgb ! videoconvert ! jpegenc ! filesink location=image.jpg`

does not work at all

TIS-Stefan, 2020-04-30:
Set the buffers to 2 and see, what happens.

Use 

multifilesink location=image%d.jpg

TIS-Stefan, 2020-04-30:
Hello

we found, that the first provided image is damaged, therefore, please set num-buffers to 3. For currently unknown reason, this provides 2 images only and the second should be ok.

closed 2021-02-19 by TIS-Stefan