TheImagingSource/tiscamera #344: Syncing 6 DM-25G445-ML cameras to jetson TX2

f2013619, 2020-04-29:
Hi, 
I am trying to sync 6 cameras to jetson TX2. So I was trying to start by syncing two cameras first and then move on to 6. Can you please help me out as to how can I capture two images at the same time from the two cameras, i mean how can I sync both the cameras.
Regards, 
Shivani

TIS-Stefan, 2020-04-29:
Dear Mr. Shivani

Thank you for your email.
Which programming language do you use?
How did you start until now?

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

 On Wed, 29th Apr 2020 at 15:09, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I am trying to sync 6 cameras to jetson TX2. So I was trying to start by syncing two cameras first and then move on to 6. Can you please help me out as to how can I capture two images at the same time from the two cameras, i mean how can I sync both the cameras.
Regards,
Shivani 
—

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/166837?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/166837?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/166837?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/34176 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#BYOMT24NA2A633R3X5X)

f2013619, 2020-04-30:
Hi,
I am very new to this.
I am comfortable with the C language.
I was trying to use two tcam-capture windows and trying to capture the
photos at the same time.
Regards,
Shivani Choudhary

TIS-Stefan, 2020-04-30:
Dear Mr. Shivani Choudhary

Thank you for your email.
A good start would be https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Simple%20Snap%20Image%20in%20C%2B%2B
If you look into main.cpp, you see, that there is a line
TcamImage cam("42719953");

Simple approach is
TcamImage cam1("42719953");

TcamImage cam2("...");
TcamImage cam3("...");

and so on an use the code 6 times. (Better would be an array with pointers)
TcamImage *cam[6];

And use the cameras in a loop. Can you create that?

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

 On Thu, 30th Apr 2020 at 17:16, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I am very new to this.
I am comfortable with the C language.
I was trying to use two tcam-capture windows and trying to capture the
photos at the same time.
Regards,
Shivani Choudhary

-- 

Sincerely,

Shivani Choudhary

(+91)7728086646

Email id: f2013619@pilani.bits-pilani.ac.in

—

  

 On Wed, 29th Apr 2020 at 17:42, Stefan Geissler <stefan.geissler@theimagingsource.com> wrote: 

Dear Mr. Shivani

Thank you for your email.
Which programming language do you use?
How did you start until now?

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  

 On Wed, 29th Apr 2020 at 15:09, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I am trying to sync 6 cameras to jetson TX2. So I was trying to start by syncing two cameras first and then move on to 6. Can you please help me out as to how can I capture two images at the same time from the two cameras, i mean how can I sync both the cameras.
Regards,
Shivani 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/167248?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/167248?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/167248?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/34176 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#BYOMT24NA2A633R3X5X)

f2013619, 2020-04-30:
Hi,
Thanks a lot for your advice. I can do that.
Regards,
Shivani Choudhary

f2013619, 2020-05-05:
Hi,
I was trying to use with one camera the code and work my way up. I am using
frame rate 15/2. But I am getting timeout at snap image if I use 60 ms
timeout. can you please tell me about this.
Regards,
Shivani choudhary

TIS-Stefan, 2020-05-05:
Dear Mr. Shivani choudhary

Thank you for your email.
If you use 15/2 = 7,5 frames per second, then the frame delivery time is 133ms. Much longer, than you 60ms. Thus, set the timeout to 266ms in order to be on the safe side.

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

 On Tue, 5th May 2020 at 15:11, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I was trying to use with one camera the code and work my way up. I am using
frame rate 15/2. But I am getting timeout at snap image if I use 60 ms
timeout. can you please tell me about this.
Regards,
Shivani choudhary

>
>
>
> Dear Mr. Shivani Choudhary
>
> Thank you for your email.
> A good start would be
> https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Simple%20Snap%20Image%20in%20C%2B%2B
> If you look into main.cpp, you see, that there is a line
> TcamImage cam("42719953");
>
> Simple approach is
> TcamImage cam1("42719953");
>
> TcamImage cam2("...");
> TcamImage cam3("...");
>
> and so on an use the code 6 times. (Better would be an array with pointers)
> TcamImage *cam[6];
>
> And use the cameras in a loop. Can you create that?
>
> Yours sincerely
>
> Stefan Geissler
> --------------------------------------------------------------------
> The Imaging Source Europe GmbH
> Ueberseetor 18
> D-28217 Bremen
> Germany
> Tel +49/421/33591-0
> Fax +49/421/33591-80
> Registergericht: Amtsgericht Bremen HRB 11 927
> Geschäftsführer: Rolf Bollhorst
> --------------------------------------------------------------------
>
> E-MAIL
>
> On Thu, 30th Apr 2020 at 17:16, TheImagingSource/tiscamera <
>
[... 515 lines cut ...]
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  

 On Wed, 29th Apr 2020 at 15:09, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I am trying to sync 6 cameras to jetson TX2. So I was trying to start by syncing two cameras first and then move on to 6. Can you please help me out as to how can I capture two images at the same time from the two cameras, i mean how can I sync both the cameras.
Regards,
Shivani 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/168243?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/168243?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/168243?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/34176 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#BYOMT24NA2A633R3X5X)

f2013619, 2020-05-07:
Hi Stefan,
I had some different doubt. Can you give me an idea of power consumption of
the cameras. I am using 5 port intellinet Poe switch which has maximum 30 W
per port.
Regards,
Shivani Choudhary

TIS-Stefan, 2020-05-08:
Dear Mr. Shivani Choudhary

Thank you for your email.
The PoE controller of our camera is set to class 2 of IEEE 802.3af which will describe that the expected power consumption of the connected device will not exceed 6.49W on the PSE side. So the mentioned PoE switch with 30W per port will be easily able to drive our camera.

Yours sincerely

Stefan Geissler
--------------------------------------------------------------------
The Imaging Source Europe GmbH
Ueberseetor 18
D-28217 Bremen
Germany
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

 On Thu, 7th May 2020 at 18:39, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi Stefan,
I had some different doubt. Can you give me an idea of power consumption of
the cameras. I am using 5 port intellinet Poe switch which has maximum 30 W
per port.
Regards,
Shivani Choudhary

>
>
>
> Dear Mr. Shivani choudhary
>
> Thank you for your email.
> If you use 15/2 = 7,5 frames per second, then the frame delivery time is
> 133ms. Much longer, than you 60ms. Thus, set the timeout to 266ms in order
> to be on the safe side.
>
> Yours sincerely
>
> Stefan Geissler
> --------------------------------------------------------------------
> The Imaging Source Europe GmbH
> Ueberseetor 18
> D-28217 Bremen
> Germany
> Tel +49/421/33591-0
> Fax +49/421/33591-80
> Registergericht: Amtsgericht Bremen HRB 11 927
> Geschäftsführer: Rolf Bollhorst
> --------------------------------------------------------------------
>
> E-MAIL
>
> On Tue, 5th May 2020 at 15:11, TheImagingSource/tiscamera <
>
> Hi,
> I was trying to use with one camera the code and work my way up. I am using
> frame rate 15/2. But I am getting timeout at snap image if I use 60 ms
> timeout. can you please tell me about this.
> Regards,
> Shivani choudhary
>
>
> >
> >
> >
> > Dear Mr. Shivani Choudhary
[... 1298 lines cut ...]
Tel   +49/421/33591-0
Fax   +49/421/33591-80
Registergericht: Amtsgericht Bremen HRB 11 927
Geschäftsführer: Rolf Bollhorst 
--------------------------------------------------------------------

E-MAIL 

  

 On Wed, 29th Apr 2020 at 15:09, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hi,
I am trying to sync 6 cameras to jetson TX2. So I was trying to start by syncing two cameras first and then move on to 6. Can you please help me out as to how can I capture two images at the same time from the two cameras, i mean how can I sync both the cameras.
Regards,
Shivani 

  
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/169361?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/169361?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/TORK-9660-RVGJ/T24NA2A633R3X5X/169361?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/34176 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#BYOMT24NA2A633R3X5X)

f2013619, 2020-06-24:
Hi Stefan,
I was trying to use the method you told to connect 6 cameras to the board.
When I tried with three cameras it worked but now when I have connected all
the cameras and extended the code for all the six cameras its not working.
I get the attached output.
Can you please help me with this.
 Thanks & Regards,
Shivani

TIS-Stefan, 2020-06-24:
Hello
I can not see the output.

Stefan

f2013619, 2020-06-24:
Hi,
I have attached the image again.
Regards,
Shivani

TIS-Stefan, 2020-06-24:
Aha. Where is the image?

Stefan

TIS-Stefan, 2020-06-24:
If you send an email to the github issues, the attachments seem to be solved. Thus open the URL https://github.com/TheImagingSource/tiscamera/issues/344 in your browser and add your comment and images there.

Stefan

f2013619, 2020-06-24:
Hi, 
For two cameras the code worked perfectly. when I increase the code for three cameras it gives the output in the image. unless I close the snaps that display the images the code doesn't go to Snap Image.  After I close all the displays the output in the attached image appears.
[image]

TIS-Stefan, 2020-06-24:
Timeout means, no images. Therefore, you may try lower frame rates. 

Stefan

f2013619, 2020-06-29:
Hi Stefan,
I had a few questions. Sorry to bother you.

When I changed the frame rate from 15/2 to 15/4 I get a warning in make
command that clock skew is detected. Your build may be incomplete. File
depend.make in simple-snapimage.dir has modification time 13403 s in the
future.
So when I run the code I get timeout at snap even when I have used 1024 as
timeout in the code
Can you please share your thoughts on this?

Also I wanted to ask if can I go below 15/4 frame rate as I have 10 cameras
connected to the board. So I just wanted to confirm if I can use 15/8 as my
frame rate.

Thanks & Regards,
Shivani Choudhary

TIS-Stefan, 2020-06-29:
Can you please send me the make output?

Stefan

f2013619, 2020-07-08:
Hi Stefan,
I understood the problem with the clock skew and fixed it.
I am using the attached code for using 7 cameras 10 cameras at a time. But
after 4 snaps it stops working. Can you please take a look and tell me
where I am going wrong?
Regards,
Shivani Choudhary

TIS-Stefan, 2020-07-08:
Hello Shivani

You can not sent attachments by email into the github issues list. You must write the issue directly at https://github.com/TheImagingSource/tiscamera/issues/344

Stefan

f2013619, 2020-07-08:
Hi , Please find attached the code I am using.
////////////////////////////////////////////////////////////////////////
/* Simple snapimage Example
This sample shows, how to snap a image from the live stream.

It uses the the examples/cpp/common/tcamcamera.* files as wrapper around the
GStreamer code and property handling. Adapt the CMakeList.txt accordingly.

As sample image processing an OpenCV cv::Mat is generated and saved as JPEG
*/

#include <stdio.h>
#include <stdlib.h>

#include <unistd.h>
 
#include "tcamimage.h"

#include "opencv2/opencv.hpp"

using namespace gsttcam;

///////////////////////////////////////////////////////////////////////////////////////////////////
int main(int argc, char **argv)
{
    gst_init(&argc, &argv);

    cv::Mat OpenCVImage;
    // Initialize our TcamCamera object "cam" with the serial number
    // of the camera, which is to be used in this program.
    //TcamImage cam1("13020368");
    TcamImage cam2("13020378");
    TcamImage cam3("13020379");
    TcamImage cam4("13020364");
    TcamImage cam5("13020365");
    TcamImage cam6("13020373");
    TcamImage cam7("13020371");
    TcamImage cam8("13020374");
    TcamImage cam9("13020377");

    
    // Set a color video format, resolution and frame rate
    //cam1.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});

    cam2.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam3.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam4.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam5.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam6.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam7.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam8.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    cam9.set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});

    // Comment following line, if no live video display is wanted.
    //cam1.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam2.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam3.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam4.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam5.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam6.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam7.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam8.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    cam9.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    // Start the camera
    //cam1.start();
    cam2.start();
    cam3.start();
    cam4.start();
    cam5.start();
    cam6.start();
[... 141 lines cut ...]
        
        cv::imwrite("test_cam9.jpg",OpenCVImage);
    }
    else
    {
        printf("Timeout at snapImage()\n");
    }

    printf("Press enter key to end program.");
    // Simple implementation of "getch()", wait for enter key.
    char dummyvalue[10];
    scanf("%c",dummyvalue);

    //cam1.stop();
    cam2.stop();
cam3.stop();
cam4.stop();
cam5.stop();
cam6.stop();
cam7.stop();
cam8.stop();
cam9.stop();

    return 0;
}

TIS-Stefan, 2020-07-08:
Hello

First of all, you should shorten your code by using an error for the camera objects:

```
TcamImage *pCam[10];
int numcameras = 9;

pCam[0] = new TcamImage("13020368");
pCam[1] = new TcamImage("13020378");
pCam[2] = new TcamImage("13020379");
pCam[3] = new TcamImage("13020364");
pCam[4] = new TcamImage("13020365");
pCam[5] = new TcamImage("13020373");
pCam[6] = new TcamImage("13020371");
pCam[7] = new TcamImage("13020374");
pCam[8] = new TcamImage("13020377");

for( int i = 0; i < numcameras; i++)
{
    pCam[i]->set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    pCam[i]->enable_video_display(gst_element_factory_make("ximagesink", NULL));
    pCam[i]->start();
}

for( int i = 0; i < numcameras; i++)
{
    if( pCam[i]->snapImage(533) )
    {
        // On succes do something with the image data. Here we create
        // a cv::Mat and save the image
        OpenCVImage.create( cam2.getHeight(),pCam[i]->getWidth(),CV_8UC(pCam[i]->getBytesPerPixel()));
        memcpy( OpenCVImage.data, pCam[i]->getImageData(), pCam[i]->getImageDataSize());
        char filename[256];
        sprintf(filename,"test_cam%d.jpg",i);
        cv::imwrite(filename,OpenCVImage);
    }
    else
    {
        char errortext[256];
        sprintf(errortext,"Timeout at snapImage() of camera %d\n",i);
        printf(errortext);
    }
}

for( int i = 0; i < numcameras; i++)
{
    pCam[i]->stop();
}
```
This makes the code better to read and you do not have potential copy & paste errors and changes need to be done only once.
You start each cameras twice, which has no effect. 

When running your or the newer code, which error message or message do you receive?

Stefan

f2013619, 2020-07-08:
Hi stefan, 
thanks a lot for the guidance in writing the code.
i did notget an error in either of the codes. But the issue is that after the 5th Snap image window that opens it stops working. unless I close one or two windows the next window does not come and the images are not captured giving the following output-
Timeout at snapImage() of camera 0
Timeout at snapImage() of camera 1
Timeout at snapImage() of camera 2
Timeout at snapImage() of camera 3
Timeout at snapImage() of camera 4
Timeout at snapImage() of camera 5
Timeout at snapImage() of camera 6
Timeout at snapImage() of camera 7
Timeout at snapImage() of camera 8
Press enter key to end program.
[image]
[image]

TIS-Stefan, 2020-07-08:
Then there may is a bandwidth issue.
So you mach change the source to

```
TcamImage *pCam[10];
int numcameras = 9;

pCam[0] = new TcamImage("13020368");
pCam[1] = new TcamImage("13020378");
pCam[2] = new TcamImage("13020379");
pCam[3] = new TcamImage("13020364");
pCam[4] = new TcamImage("13020365");
pCam[5] = new TcamImage("13020373");
pCam[6] = new TcamImage("13020371");
pCam[7] = new TcamImage("13020374");
pCam[8] = new TcamImage("13020377");

for( int i = 0; i < numcameras; i++)
{
    pCam[i]->set_capture_format("BGRx", FrameSize{1280,960}, FrameRate{15,4});
    pCam[i]->enable_video_display(gst_element_factory_make("ximagesink", NULL));
}

for( int i = 0; i < numcameras; i++)
{
    pCam[i]->start();
    if( pCam[i]->snapImage(533) )
    {
        // On succes do something with the image data. Here we create
        // a cv::Mat and save the image
        OpenCVImage.create( cam2.getHeight(),pCam[i]->getWidth(),CV_8UC(pCam[i]->getBytesPerPixel()));
        memcpy( OpenCVImage.data, pCam[i]->getImageData(), pCam[i]->getImageDataSize());
        char filename[256];
        sprintf(filename,"test_cam%d.jpg",i+1)
        cv::imwrite(filename,OpenCVImage);
    }
    else
    {
        char errortext[256];
        sprintf(errortext,"Timeout at snapImage() of camera %d\n",i+1)
        printf(errortext);
    }
    pCam[i]->stop();
}
```
The cameras will not run in parallel with that, but we will enhance that in the next step. It is just to figure out, whether it is working and you get images from all cameras.
Unfortunatelly you use DMM 25G445-ML cameras and not DMK 33G445. The latter one memory on board, so you can trigger tons of them simultaneously and then get the images one after another... The DMM 25G445-ML has no memory on board, so the images must be picked up in time, otherwise they are lost.

Stefan

f2013619, 2020-07-08:
Hi Stefan,
Thanks a lot. I am now able to capture images from all cameras.
Is there anyway in which I can increase the bandwidth so that they can run
in parallel?
Regards,
Shivani

TIS-Stefan, 2020-07-08:
Hello Shivani

There is no way to increase the bandwidth, it is limited by the GigE connector. 
The simple solution would be using a different more sophisticated camera model, as I wrote. 

I would like to know, what is the maximum time allowed between the image of the first and of the last camera. Knowing this, we can try to make something more "smart". 

In any case, we have to rewrite your program in some parts, e.g. enable trigger mode, use software trigger, so all cameras can be started in parallel, but do not deliver images (attention: Automatics must be turned off), Then we need an image callback, because we use trigger. Image delivery will be faster.
The trick is to trigger, e.g. 4 cameras, wait for the images and then trigger the next 4 cameras and so on. 

So I wait for your answer, before we code...

Stefan

f2013619, 2020-07-09:
Hi stefan,
The basic idea is to create a 3D model from the 2D images taken from the
cameras at once. So I think the maximum allotted time difference between
the first and last camera can be around 10s.
Best regards,
Shivani

TIS-Stefan, 2020-07-09:
Hello Shivani

I think, we are done with the code then.

Stefan

f2013619, 2020-07-10:
Hi Stefan,
It will be great if you could help me out with the parallel cameras even 4
at a time.
If you could guide me how to do it as you mentioned in one of the previous
mails.

Best regards,
Shivani

TIS-Stefan, 2020-07-10:
Hello Shivani

I am not allowed to make software development for free.
Therefore, please check trigger mode and software trigger samples at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples
Keep in mind, the trigger mode property is a string on gige cameas, like "On".

Stefan

TIS-Stefan, 2020-07-10:
Hello Shivani

using trigger mode, callback and image saving by OpenCV is much more complex. You need to create a callback for each camera, which gets its own customdata passed.
This sample triggers many cameras by software trigger and does all the callback and opencv stuff.
[trigger-many-cameras.zip](https://github.com/TheImagingSource/tiscamera/files/4903599/trigger-many-cameras.zip)

Currently, there is one loop for all cameras, but I am pretty sure you can make two or three loops, one for camera 1-4, then 5-8 and then 9-10.

The wait for images has no time out. so if a camera does not sent an image or the image received is incomplete, the program will hang. You may create your own time out.

Stefan

TIS-Stefan, 2020-07-13:
Hello Shivani

I would like to know, whether you had a chance to check the project, I sent to you.

Stefan

f2013619, 2020-07-14:
Hi Stefan,
I am sorry for the delay but there are issues at my office because of
Covid-19 case detection in the office.
I am trying to test the project as soon as possible and thanks a lot for
all your help.
I will let you know as soon as I try it.
Thanks again.
Regards,
Shivani Choudhary

TIS-Stefan, 2020-07-14:
Hi Shivany

even if this is not the platform for personal messages, please take care and stay safe!
I suppose, if you understood the new project, the next question will be properties like exposure, gain etc.

Stefan

f2013619, 2020-07-14:
Hi stefan,
I had a doubt regarding connecting 10 cameras at once. I am using two PoE
switch and have connected 7 cameras to one switch and 3 cameras to another.
I have given different ipv4 settings to each camera. After doing this when
I use the tcam-capture either 3 cameras connected to one Poe switch works
or 7 cameras connected to another one works but not both at the same time.
can you guide me as to what I am doing wrong?

TIS-Stefan, 2020-07-14:
Hi Shivani

how are the switches connected to the computer and what are the Ip addresses of the cameras and network controller?

Stefan

f2013619, 2020-07-14:
Hi Stefan,
I resolved that issue.
I tried the trigger many cameras zip. After I ran the program the cameras
are staying in trigger mode only but they are not capturing any image.
Afterwards I checked in tcam_capture, the cameras were staying in trigger
mode itself but not capturing anything. Before I ran this the cameras were
capturing images in tcam-capture.
Can you tell me how to resolve this?

TIS-Stefan, 2020-07-14:
Hi Shivani

The cameras should have provided images with each software trigger. Did you got any output from the program? Did the program end unexpected? 
Do you get images in tcam-capture? 

Stefan

f2013619, 2020-08-06:
Hi Stefan,
The window is opening for all the 10 cameras. But it saves the image for
only 6 the and 4th camera and the it hangs. Can you  please tell me what
can be done?
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-06:
Hello Shivani

I am sorry, but I can not guess, what happens on your computer.

All cameras are in trigger mode and you use software trigger to acquire an image?

Stefan

f2013619, 2020-08-06:
Hi Stefan,
Yes I am using software trigger as you mentioned in your zip file.
Also can you tell me how to set the exposure settings when I am using code
to capture images as the images are not coming as it comes using tcam
capture. They are yellowish.
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-06:
Hello Shivani

Please understand, that I do not have the code in memory anymore. (I am an old man...)

Please let us handle the issue of the 10 images first. 
If you do not get an image, is the timeout error shown in the terminal?

Stefan

f2013619, 2020-08-06:
Hi Stefan,
If i use the single image code from Linux sample programming examples I am
getting images but they are not parallell.
Hence I was trying to use the zip code that you sent to trigger many
cameras at once. When I use that the cameras are triggered but the images
from all the cameras are not saved.
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-06:
Hi Shivani

we miss each other... you do not understand or answer my questions. 

If I am looking into the correct source code, then there should be an output "Saved image" plus image file name. Do you get that?

Stefan

f2013619, 2020-08-06:
Hi Stefan,
Yeah I get output like Saved image  camera06.001.jpg.
But I get that from only two cameras.
I have increased the number of cameras to 10 still the images get saved
from only 6 the and 4th camera.
Regards,
Shivani Choudhary

TIS-Stefan, 2020-08-06:
Did you debug your code and check, whether all cameas are handled?

f2013619, 2020-08-06:
Yeah, the image window opens for every camera that is 10 windows open but
the issue is the image is not getting saved for every camera.

TIS-Stefan, 2020-08-06:
Did you debut your code? e.g. with gdb?

f2013619, 2020-08-06:
No. I will try to do that.

TIS-Stefan, 2020-08-07:
I suppose, you work on your TX hardware. 
You can make things much easier for the camera development, if you use a laptop or desktop computer and use e.g. Visual Studio Code for programming and debugging.  However, even here you have to learn how to use the tools. 

I do not know a nice C++ IDE for ARM with good gdb integration. Maybe you can find one in the internet.

You may send me your project source code, so I can see, which changes you did.

Stefan

f2013619, 2020-08-10:
Hi Stefan, 
Please find attached my source code.
[main_trigger many cameras.txt](https://github.com/TheImagingSource/tiscamera/files/5049729/main_trigger.many.cameras.txt)

TIS-Stefan, 2020-08-10:
Hello Shivana

your code can not work:

```
    strcpy(CustomData[0].serialnumber,"33819936");
    strcpy(CustomData[1].serialnumber,"22819913");
strcpy(CustomData[2].serialnumber,"22819913");
strcpy(CustomData[3].serialnumber,"22819913");
strcpy(CustomData[4].serialnumber,"22819913");
strcpy(CustomData[5].serialnumber,"22819913");
strcpy(CustomData[6].serialnumber,"22819913");
strcpy(CustomData[7].serialnumber,"22819913");
strcpy(CustomData[8].serialnumber,"22819913");
strcpy(CustomData[9].serialnumber,"22819913");
```

It is always the same serial number. If you use 10 cameras, then there should be 10 different serial numbers.

Stefan

f2013619, 2020-08-10:
Hi Stefan,
Yeah I have changed that to 10 different serial numbers. But still the
images for only two cameras are getting saved.
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-10:
The problem is, that the code fires software trigger for all cameras at once. This works fine with newer camera models, such as 33G, but not with your 25G series. The 25G series has no memory on board, therefore, we must get the images one by one from each camera:

```
    // Take  images from all camras.
    for( int images = 0; images < 1; images++)
    {
        // Trigger cameras
        for( int i = 0; i < numcameras; i++)
        {
            CustomData[i].SaveNextImage = true;
            CustomData[i].SoftwareTrigger->set(*CustomData[i].pCam,1);

            int tries = 20;
            while( CustomData[i].SaveNextImage && tries > 20)
            {
                usleep( 33000); // Wait for 33ms
                tries--;
            }
            if( CustomData[i].SaveNextImage) // Oh oh, we got no image
            {
                printf("No image from camera %s\n",CustomData[i].serialnumber );
            }
        }
    }
```
That is the change loop. One software trigger, wait for image, repeat for all cameras. 

Stefan

f2013619, 2020-08-10:
Hi Stefan,
I tried the code you sent and now 1 to 8 camera is saving images but for
some reason 0 and 9 camera is not saving the image. They are working
perfectly if I try to capture images in tcam- capture.
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-10:
Hello Shivani
Just move these cameras to other positions in the array and see, what happens
Do you see, that each camera is handled in exactly the same way?

Stefan

f2013619, 2020-08-12:
Hi,
I moved the cameras in the array.
There's one thing that i am getting that is each time different number if
cameras are saving images. When I tried today 7 cameras saved images.
Sometimes only two cameras save images.
Do you have an idea why this might be happening?
Best regards,
Shivani Choudhary

f2013619, 2020-08-12:
Hi Stefan,
When I moved the cameras in the array then also those two particular
serial numbers are capturing the image.
Can you please tell me the settings to get the proper images as now I am
obtaining yellowish images.
Best regards,
Shivani Choudhary

TIS-Stefan, 2020-08-13:
Hello Shivani 

Are all cameras powered correctly?
Do the green LEDs on all cameras light?
Do the yellow LEDs on all cameras light?

Do you get any timeout outputs from the software? The message at printf("No image from camera %s\n",CustomData[i].serialnumber ); in the code change I sent to you?

Stefan

f2013619, 2020-08-24:
Hi stefan, 
The cameras are powered correctly and the all the \Green and Yellow lights are on.
Individually if I take photos it is working perfectly. 
But in the code change that you sent the attached error is happening.
Also the images are not proper they have yellowish tone to them.
Can you please give me the settings to take proper images by putting proper exposure and gain.
[image]

closed 2021-02-19 by TIS-Stefan