TheImagingSource/tiscamera #345: DMK23U274 Error: Property Gain Max reported a range that could not be handled

strolch101, 2020-05-15:
System: LMDE4, Buster, 64bit, gcc 8.3.0, cmake 3.13.4
Depencies are all present and were checked by running repo-included install scripts.

CMake build flags:
-DBUILD_ARAVIS=OFF
-DBUILD_GST_1_0=ON
-DBUILD_TOOLS=ON
-DBUILD_V4L2=ON
-DBUILD_LIBUSB=ON
-DBUILD_DOCUMENTATION=ON
-DCMAKE_INSTALL_PREFIX=/home/xxx/TISCamera

Build went fine beside some warning about missing logo-images location related to documentation-build. `./tcam-capture` runs flawless with another TIS camera (D*F*K23U274). However, title-mentioned D*M*K23U274 shows no image (all white), most controls are greyed out and keeps giving the following error in terminal:

```
xxx@xxx:~/TISCamera/bin$ ./tcam-capture 
qt5ct: using qt5ct plugin
15-05-2020:21:28:01,181 ERROR    [PropertyWidget.py:94] Property Gain Max reported a range that could not be handled
15-05-2020:21:28:01,202 WARNING  [TcamView.py:490] get_tcam_property failed for 'Trigger Mode'
15-05-2020:21:28:01,421 ERROR    [TcamView.py:460] ERROR: bin : Could not create element 'tcamautoexposure'.
15-05-2020:21:28:03,243 ERROR    [PropertyWidget.py:189] A slider had a value outside of the integer range. That should no happen.
```
The last message appears repeatedly. Both cameras are detected as V4L2 devices in tcam-capture. VLC Media Player recognizes and shows both camera streams fine and just has issues displaying the correct video format (no rggb-color, artifacts related to resolution).

How do I fix this?

TIS-Stefan, 2020-05-18:
Hello

It seems, you use the "development" branch of tiscamera. If I am right, please checkout the "master" branch and build again.

tcam-ctrl shows the properties of the "naked" camera only. The automatics are made in software by our GStreamer modules. But I agree with you, the error of ranges from 0~ max-double should not happen. This was invented in development branch 0.12.

Stefan

strolch101, 2020-05-18:
Thanks for your answer. I cloned the repository via `git clone https://github.com/TheImagingSource/tiscamera.git` again. UnfortunateIy the error still persists. 

I went back to my pre-TIS-Camera system with Timeshift to have the most "fresh" install as possible. Thus all the (then missing) depencies were newly installed via your install-script too. In addition, I left the install directory untouched, i.e. default, and followed strictly your github readme step-by-step (only exception: newly created build folder is empty, so build was done from ~/tiscamera)

This was my new build configuration:
```
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found GObject: /usr/include/glib-2.0  
-- Found PYTHON_MODULE_PyQt5: /usr/lib/python3/dist-packages/PyQt5  
-- Found Sphinx: /usr/bin/sphinx-build  
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   OFF
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            ON
-- Build tests                    OFF
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /etc/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing data files to:      /usr/share/theimagingsource/tiscamera
-- Installing gstreamer-1.0 to:   /usr/lib/x86_64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/x86_64-linux-gnu/girepository-1.0
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/tiscamera
```

The only noticeable events while building were the following:

```
[...]
[ 17%] Linking CXX shared library libtcam.so
[ 17%] Built target tcam
Scanning dependencies of target tcamprop
[ 18%] Building C object src/gobject/CMakeFiles/tcamprop.dir/tcamprop.c.o
[ 19%] Linking C shared library libtcamprop.so
[ 19%] Built target tcamprop
Scanning dependencies of target create_gobject
[ 20%] Generating Tcam-0.1.gir
x86_64-linux-gnu-gcc -pthread -E -Wdate-time -D_FORTIFY_SOURCE=2 -I. -I/home/xxx/tiscamera/src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -o g-ir-cpp-44nmij8p.i -C /home/xxx/tiscamera/src/gobject/g-ir-cpp-44nmij8p.c
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/xxx/tiscamera/src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/uuid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -c /home/xxx/tiscamera/src/gobject/tmp-introspectoavhfm6k/Tcam-0.1.c -o /home/xxx/tiscamera/src/gobject/tmp-introspectoavhfm6k/Tcam-0.1.o -Wno-deprecated-declarations -pthread
g-ir-scanner: link: cc -o /home/xxx/tiscamera/src/gobject/tmp-introspectoavhfm6k/Tcam-0.1 /home/xxx/tiscamera/src/gobject/tmp-introspectoavhfm6k/Tcam-0.1.o -L. -Wl,-rpath,. -Wl,--no-as-needed -L/home/xxx/tiscamera/src/gobject -Wl,-rpath,/home/xxx/tiscamera/src/gobject -ltcamprop -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
[ 21%] Generating Tcam-0.1.typelib
[ 21%] Built target create_gobject
Scanning dependencies of target tcam-algorithms
[...]
```
and
```
[100%] Generate HTML Documentation with Sphinx
WARNING: logo file '../../../data/images/logo_small.jpg' does not exist
WARNING: favicon file '../../../data/images/logo.jpg' does not exist
[100%] Built target doc
```
The result has not changed. Both cameras are connected but tcam-capture can only handle the D_F_K23U247 correctly, not the D_M_K23U274. Error is same as initially.

TIS-Stefan, 2020-05-19:
Hello

can you please change into your tiscamera directory and run
git status
and show me the output?

guvcview creates a lot of output...

The DxK 23 series has only a few properties, that are in the hardware. Most of the other properties are generated by the Gstreamer Modules. In particular it is the automatics for exposure and gain. Therefore, third party v4l2 programs do not show these properties. 

The values for gain min and max are available, after the pipeline has changed into state PLAYING. Before this happens, the max and min are the max double or max in and 0. The max double and max int values were invented with the development branch. They should not be in the master branch of tiscamera.
Therefore the Python sample 03 shows nonsense ranges. 
You may try the sample at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Save%20an%20image%20on%20trigger%20in%20Python
It lists the properties after the pipeline has been started and then there should be meaningful min and max values. 
You must uncomment line 64:
```
# Remove comment below in oder to get a property list.
# Tis.List_Properties()
```

I wonder, whether the white image is simply the Window background and no frames are delivered. What happens if you use the simple
`
gst-launch-1.0 tcambin ! videoconvert ! ximagesink
`
with the DMK.

Stefan

strolch101, 2020-05-19:
Hello,

here is the output of git status (in german though):

```
~/tiscamera$ git status
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

Unversionierte Dateien:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)

	CPackConfig.cmake
	CPackSourceConfig.cmake
	cmake_uninstall.cmake
	data/tcam.pc
	data/udev/80-theimagingsource-cameras.rules
	doc/_build/
	doc/_doctrees/
	doc/html/
	env.sh
	packaging/deb/postinst
	packaging/deb/prerm
	src/algorithms/libtcam-algorithms.so.0
	src/algorithms/libtcam-algorithms.so.0.11.1
	src/gobject/Tcam-0.1.gir
	src/gobject/Tcam-0.1.typelib
	src/gobject/libtcamprop.so.0.1
	src/libtcam.so.0
	src/libtcam.so.0.11.1
	src/libusb/libtcam-libusb.so.0
	src/libusb/libtcam-libusb.so.0.11.1
	src/v4l2/libtcam-dfk73.so.0
	src/v4l2/libtcam-dfk73.so.0.11.1
	src/v4l2/libtcam-v4l2.so.0
	src/v4l2/libtcam-v4l2.so.0.11.1
	tools/dfk73udev/dfk73udev
	tools/firmware-update/firmware-update
	tools/tcam-capture/tcam-capture.desktop
	tools/tcam-ctrl/tcam-ctrl
	tools/tcam-uvc-extension-loader/tcam-uvc-extension-loader

nichts zum Commit vorgemerkt, aber es gibt unversionierte Dateien
(benutzen Sie "git add" zum Versionieren)

```
Just a guess, but at an very early stage I tried to install the ubuntu-libraries and dutils from your [support website](https://www.theimagingsource.de/support/software-f%C3%BCr-linux/) which didn't work. It might be that some remainder of that attempt is causing the issues. 
I also tried `gst-launch-1.0 tcambin ! videoconvert ! ximagesink`. If only DMK is connected then I get the following output:
```
/usr/bin$ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Leitung wird auf PAUSIERT gesetzt ...
FEHLER: Leitung möchte nicht pausiert werden.
FEHLER: Von Element /GstPipeline:pipeline0/GstTcamBin:tcambin0: Could not create element 'tcamautoexposure'.
Zusätzliche Fehlerdiagnoseinformation:
Could not create element 'tcamautoexposure'.
Leitung wird auf NULL gesetzt ...
Leitung wird geleert ...
```

Trying the programming examples was more successful. As you mentioned I got more meaningful values for the DMK and the program ran fine. If you need the terminal output to take a look at, just let me know, and I will insert here later. Actually these examples are a partial solve of my problem, since I only need the SDK being able to talk to the camera and not tcam-capture in particular.

TIS-Stefan, 2020-05-20:
Hello

Did the Python sample show exposure and gain auto properties? If the module tcamautoexposure is not found, I guess, rather not.
I suggest to reinstall:
```
cd tiscamera/build
sudo gige-deamon stop
sudo make uninstall
rm * -rf
cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON ..
make -j 48
sudo make install
```
The gige-daemon stop may says, there is no process, which is ok.
The make -j 48 tells make to use 48 thread max.

After installation the pipeline
`gst-launch-1.0 tcambin ! videoconvert ! ximagesink`
should run without error message. If there are still missing modules, you may run

`gst-inspect-1.0 | grep tcam`

It should list tcamsrc, tcambin, tcamwhitebalance, tcamautoexposure, tcambiteater and so on.
If it does not show tcamautoexposure, then I wonder, whether you change the GST search path.

BTW: German output is ok.

Stefan

strolch101, 2020-05-21:
Hello,

it seems the python sample does find tcamautoexposure. Here is the output without Tis.List_Properties()
```
~/TISCamera/programming-samples/Save an image on trigger in Python$ python3 Program.py 
tcambin serial="39910311" name=source ! video/x-raw,format=GRAY8,width=1600,height=1200,framerate=20/1 ! videoconvert ! appsink name=sink
Gain Auto : True 
Gain : 176
Gain Auto now : False 
Exposure Auto now : False 
Press Esc to stop

(python3:3875): Gtk-WARNING **: 09:25:44.253: Theme parsing error: gtk.css:661:53: Expected ',' in color definition
```
Beside the stream being a bit slow, the image is shown in good quality and is regularly updated. 

Opening tcam-capture immediately after running the python-script made the all-white background for DMK disappear in tcam-capture. Now there is an all-grey window with following message: *"In Trigger Mode. Waiting for first image."* but the error message in terminal is still the same (couldn't create tcamautoexposure / Gain Max / Slider value) and the drop-down menu for *Format, Resolution* and *FPS* are empty/unresponsive.

I haven't run a new build as you suggested yet. I will do that later. However, here is the **current**  output of `gst-inspect-1.0 | grep tcam`

~/TISCamera/programming-samples/Save an image on trigger in Python$ gst-inspect-1.0 | grep tcam
tcamautofocus:  tcamautofocus: The Imaging Source auto focus Element
tcamautoexposure:  tcamautoexposure: The Imaging Source Auto Exposure Element
tcamsrc:  tcamsrc: Tcam Video Source
tcamwhitebalance:  tcamwhitebalance: The Imaging Source White Balance Element
tcambiteater:  tcambiteater: Bit depth reduction module
tcambin:  tcambin: Tcam Video Bin

The terminal output for `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` has not changed over the run of the python-script.

TIS-Stefan, 2020-05-25:
Hallo

ich muss gestehen, daß ich ein wenig am Ende mit meinem Latein bin. 
Wenn tcamautoexposure bei der Farbkamera gefunden wird, gibt es absolut keinen Grund, warum das nicht auch be der Schwarzweißkamera gefunden wird.
Bitte prüfen Sie, ob bei der Farbkamera der Fehler auch wirklich nicht auftritt.

gst-inspect sagt, daß die Installation korrekt ist.

Bitte machen Sie 
export TCAM_LOG=DEBUG
gst-launch-1.0 tcambin ! videoconvert ! ximagesink

Bitte prüfen Sie auch, ob das Livebild bei Helligkeitsänderungen auch seine Helligkeit anpaßt.

In tcam-capture verwenden Sie bitte eine niedrigere Bildwiederholrate. Allerdings reagiert es manchmal etwas empfindlich. Es gibt in den Samples ein QT5 Beispiel, das etwas weniger empfindlich ist, weil man die Kamera konfigurieren kann, bevor das Livevideo gestartet wird. https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/QT5%20Device%20Selection%20and%20Property%20Dialogs%20in%20C%2B%2B

strolch101, 2020-05-27:
Hallo,

also ich habe nochmal alles neu installiert, wie im vorangegangenen Post vorgeschlagen. Diesmal mit Aravis (gige-daemon war vorher nicht installiert). Hatte leider keine Auswirkungen, das Problem besteht immer noch.

Wenn beide Kameras angeschlossen sind, öffnet `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` stets die DFK (Farbkamera). Die Farbkamera reagiert **nicht** auf Helligkeitsänderung hinsichtlich _autoexposure_ (Belichtungszeit wird nicht angepasst, wenn dunkler). Die entprechenden Slider (exposure min/max) sind in tcam-capture auch ausgegraut. 

Im Folgenden ist immer nur jeweils eine der beiden Kameras angeschlossen. Hier ist der Output mit DEBUG-Flag für die (teilweise) funktionierende DFK:

```
~$ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Leitung wird auf PAUSIERT gesetzt ...
56086      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.11.1_master/2042bba_rev_2617
	Aravis:	0.6_version_1758883
59912      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.11.1_master/2042bba_rev_2617
	Aravis:	0.6_version_1758883
59948      <ERROR> ~/tiscamera/build/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
60030      <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
62518      <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:467: Aravis gave us 2
62669      <ERROR> ~/tiscamera/build/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
62746      <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
65418      <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:467: Aravis gave us 2
84697      <DEBUG> Logger:107: 
The following library versions are used:
	Tcam:	0.11.1_master/2042bba_rev_2617
	Aravis:	0.6_version_1758883
84783      <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:426: Current framerate is 20 / 1 fps
85159      <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
85664      <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Gain (dB/100)'(199e204)
85749      <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1207: Encountered write only control.
85947      <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:220: Detected stepsize 0 for property Strobe Delay. Setting to 1.
86076      <DEBUG> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:270: Enum Trigger Polarity does not have any entries. Ignoring.
86116      <ERROR> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1245: Property 'Trigger Polarity' is null
86329      <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: RGGB
86464      <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y800
86577      <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y16
86635      <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:257: Returning 3 formats.
86679      <DEBUG> ~/tiscamera/build/src/PipelineManager.cpp:127: Received 3 formats.
86783      <DEBUG> ~/tiscamera/build/src/PropertyHandler.cpp:273: Created 14 properties
86989      <INFO> ~/tiscamera/build/src/ImageSource.cpp:68: Source changed to state STOPPED
87030      <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:567: Stopped stream
140483     <INFO> ~/tiscamera/build/src/CaptureDeviceImpl.cpp:159: Closed device DFK 23U274.
140816     <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:426: Current framerate is 20 / 1 fps
141453     <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
141555     <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Gain (dB/100)'(199e204)
141634     <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1207: Encountered write only control.
141822     <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:220: Detected stepsize 0 for property Strobe Delay. Setting to 1.
141945     <DEBUG> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:270: Enum Trigger Polarity does not have any entries. Ignoring.
141986     <ERROR> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1245: Property 'Trigger Polarity' is null
142144     <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: RGGB
142254     <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y800
142366     <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y16
142440     <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:257: Returning 3 formats.
142483     <DEBUG> ~/tiscamera/build/src/PipelineManager.cpp:127: Received 3 formats.
142559     <DEBUG> ~/tiscamera/build/src/PropertyHandler.cpp:273: Created 14 properties
150768     <INFO> ~/tiscamera/build/src/gstreamer-1.0/tcamgstbase.cpp:659: Fixating assumed largest caps: video/x-bayer, format=(string)rggb, width=(int)1600, height=(int)1200, framerate=(fraction){ 20/1, 15/1, 10/1, 15/2, 5/1, 15/4 }, pixel-aspect-ratio=(fraction)1/1
151951     <INFO> ~/tiscamera/build/src/gstreamer-1.0/tcamgstbase.cpp:659: Fixating assumed largest caps: video/x-bayer, framerate=(fraction)20/1, width=(int)1600, height=(int)1200, format=(string)rggb
Leitung ist aktiv und erfordert keinen VORLAUF …
INFO:
[... cut ...]
656039     <ERROR> ~/tiscamera/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
656069     <INFO> ~/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1093: GstObject tcamsrc does not exist. Messages will not be sent to the GstBus.
656118     <ERROR> ~/tiscamera/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods
656134     <INFO> ~/tiscamera/src/gstreamer-1.0/gsttcamsrc.cpp:1093: GstObject tcamsrc does not exist. Messages will not be sent to the GstBus.
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

TIS-Stefan, 2020-05-27:
Hallo

es sieht so aus, als hätten Sie da eine Fehler gefunden. Die Ursache scheint zu sein, daß wir statt "Exposure" "Exposure (us)" bekommen und das scheint das tcamautoexposure nicht zu mögen. 

Vielleicht können Sie das erstmal manuell im tcamautoexposure code ändern. 

Stefan

TIS-Stefan, 2020-05-27:
Hello Poul

The only situation I could imagine is, that somewhat blocks the stream. 
IC Express starts the live video as follows:

        private bool startVideo()
        {
            try
            {
                icImagingControl1.LiveStart();
                _GraphStartTime = icImagingControl1.ReferenceTimeStart;
                return true;
            }
            catch (ICException iEx)
            {
                MessageBox.Show("Error starting camera live stream:" + iEx.Message, "ic-express", MessageBoxButtons.OK, MessageBoxIcon.Error);
            }
            return false;
        }
As you can see, there is nothing special.

E-MAIL 

 On Wed, 27th May 2020 at 5:01, TheImagingSource/tiscamera <reply@reply.github.com> wrote: 

Hallo, 
also ich habe nochmal alles neu installiert, wie im vorangegangenen Post vorgeschlagen. Diesmal mit Aravis (gige-daemon war vorher nicht installiert). Hatte leider keine Auswirkungen, das Problem besteht immer noch. 
Wenn beide Kameras angeschlossen sind, öffnet gst-launch-1.0 tcambin ! videoconvert ! ximagesink stets die DFK (Farbkamera). Die Farbkamera reagiert NICHT auf Helligkeitsänderung hinsichtlich _autoexposure_ (Belichtungszeit wird nicht angepasst, wenn dunkler). Die entprechenden Slider (exposure min/max) sind in tcam-capture auch ausgegraut. 
Im Folgenden ist immer nur jeweils eine der beiden Kameras angeschlossen. Hier ist der Output mit DEBUG-Flag für die (teilweise) funktionierende DFK: 
~$ gst-launch-1.0 tcambin ! videoconvert ! ximagesink Leitung wird auf PAUSIERT gesetzt ... 56086 <DEBUG> Logger:107: The following library versions are used: Tcam: 0.11.1_master/2042bba_rev_2617 Aravis: 0.6_version_1758883 59912 <DEBUG> Logger:107: The following library versions are used: Tcam: 0.11.1_master/2042bba_rev_2617 Aravis: 0.6_version_1758883 59948 <ERROR> ~/tiscamera/build/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock" 60030 <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods 62518 <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:467: Aravis gave us 2 62669 <ERROR> ~/tiscamera/build/src/utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock" 62746 <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:465: Unable to connect to gige-daemon. Using internal methods 65418 <ERROR> ~/tiscamera/build/src/aravis/aravis_utils.cpp:467: Aravis gave us 2 84697 <DEBUG> Logger:107: The following library versions are used: Tcam: 0.11.1_master/2042bba_rev_2617 Aravis: 0.6_version_1758883 84783 <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:426: Current framerate is 20 / 1 fps 85159 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201) 85664 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Gain (dB/100)'(199e204) 85749 <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1207: Encountered write only control. 85947 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:220: Detected stepsize 0 for property Strobe Delay. Setting to 1. 86076 <DEBUG> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:270: Enum Trigger Polarity does not have any entries. Ignoring. 86116 <ERROR> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1245: Property 'Trigger Polarity' is null 86329 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: RGGB 86464 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y800 86577 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y16 86635 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:257: Returning 3 formats. 86679 <DEBUG> ~/tiscamera/build/src/PipelineManager.cpp:127: Received 3 formats. 86783 <DEBUG> ~/tiscamera/build/src/PropertyHandler.cpp:273: Created 14 properties 86989 <INFO> ~/tiscamera/build/src/ImageSource.cpp:68: Source changed to state STOPPED 87030 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:567: Stopped stream 140483 <INFO> ~/tiscamera/build/src/CaptureDeviceImpl.cpp:159: Closed device DFK 23U274. 140816 <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:426: Current framerate is 20 / 1 fps 141453 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201) 141555 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Gain (dB/100)'(199e204) 141634 <INFO> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1207: Encountered write only control. 141822 <WARNING> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:220: Detected stepsize 0 for property Strobe Delay. Setting to 1. 141945 <DEBUG> ~/tiscamera/build/src/v4l2/v4l2_utils.cpp:270: Enum Trigger Polarity does not have any entries. Ignoring. 141986 <ERROR> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:1245: Property 'Trigger Polarity' is null 142144 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: RGGB 142254 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y800 142366 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:710: Found format: Y16 142440 <DEBUG> ~/tiscamera/build/src/v4l2/V4l2Device.cpp:257: Returning 3 formats. 142483 <DEBUG> ~/tiscamera/build/src/PipelineManager.cpp:127: Received 3 formats. 142559 <DEBUG> ~/tiscamera/build/src/PropertyHandler.cpp:273: Created 14 propertie [... cut ...]
 Was this message helpful?   Yes   [https://theimagingsource.deskpro.com/ticket-rate/NCBJ-8207-YTVQ/83RC6M959KSHQKW/174795?setrating=1]   It was OK   [https://theimagingsource.deskpro.com/ticket-rate/NCBJ-8207-YTVQ/83RC6M959KSHQKW/174795?setrating=0]   No   [https://theimagingsource.deskpro.com/ticket-rate/NCBJ-8207-YTVQ/83RC6M959KSHQKW/174795?setrating=-1] 

View and manage this ticket online: https://theimagingsource.deskpro.com/tickets/35643 

  The Imaging Source [https://www.theimagingsource.com] · https://theimagingsource.deskpro.com/ 

 (#CASX83RC6M959KSHQKW)

strolch101, 2020-05-27:
> Hallo
> 
> es sieht so aus, als hätten Sie da eine Fehler gefunden. Die Ursache scheint zu sein, daß wir statt "Exposure" "Exposure (us)" bekommen und das scheint das tcamautoexposure nicht zu mögen.
> 
> Vielleicht können Sie das erstmal manuell im tcamautoexposure code ändern.
> 
> Stefan

Hallo Stefan,

Ich bin nicht sicher ob ich richtig verstanden habe. Wo genau soll ich "Exposure" zu "Exposure (us)" ändern? In der Datei `~/tiscamera/src/gstreamer-1.0/gsttcamautoexposure.cpp` ist entweder nur "Exposure Auto" zu finden oder die folgenden Codezeilen:

```
if (category)
        {
            g_value_init(category, G_TYPE_STRING);
            g_value_set_string(category, "Exposure");
        }
        if (group)
        {
            g_value_init(group, G_TYPE_STRING);
            g_value_set_string(group, "Exposure");
        }
        return TRUE;
```

Letztere habe ich in der gesamten Datei zu "Exposure (us)" geändert, allerdings ohne Erfolg.

TIS-Stefan, 2020-05-28:
Hallo

tcam-capture zeigt es als "Exposure Time (us)" an. Da habe ich leider nicht aufgepasst.
Allerdings ist die Stelle der Änderung falsch, was ich bisher nicht wußte, da ich nicht der Programmierer bin. 

Ich habe selbst folgende Änderungen versucht:
In property_identifactions.h
```
//const TCAM_PROPERTY_ID TCAM_PROPERTY_EXPOSURE      = 0x199e0002;
const TCAM_PROPERTY_ID TCAM_PROPERTY_EXPOSURE      = 0x009a0902;
```
In standard_properties.h
 ```
       TCAM_PROPERTY_EXPOSURE,
        "Exposure (Absolute)",
        //"Exposure Time (us)",
        TCAM_PROPERTY_TYPE_INTEGER,
        { TCAM_PROPERTY_CATEGORY_EXPOSURE, TCAM_PROPERTY_EXPOSURE, 0 }, 
```
Leider erfolglos. Aber immerhin ist die Warnung nun:
`<WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201) `
Wenn ich eine DxK 23U274 anschließe, dann wird mit tiscamera 0.12.0, heute morgen released, immer ein Livevideo in gst-launch und tcam-capture angezeigt. 
Die Exposure Automatik funktioniert nicht korrekt, da sie nur Gain, nicht aber die Belichtungszeit ändert.

Wie gesagt, ich kenne den Sourcecode selbst nicht so genau. als daß ich wüßte, wo ich da genau eingreifen muss. Ich habe aber den Kollegen bereits informiert.

Stefan

TIS-Stefan, 2020-05-29:
Hallo
Folgende Änderung löst das Problem für die 23U:

tiscamera 0.12.0, eventuell ein git pull machen, vielleicht alles erst löschen und dann komplett neu bauen.
In tiscamera/src/v4l2/v4l2_property_mapping.h ab Zeile 40:
```
{
	TCAM_PROPERTY_EXPOSURE,
	{ V4L2_CID_EXPOSURE_ABSOLUTE, V4L2_CID_EXPOSURE,
	// TCAM_V4L2_EXPOSURE_TIME_US
	},
},
```
zu

```
{
	TCAM_PROPERTY_EXPOSURE,
	{ V4L2_CID_EXPOSURE_ABSOLUTE, V4L2_CID_EXPOSURE,
	 TCAM_V4L2_EXPOSURE_TIME_US
	},
},
```
ändern. Neu bauen, installieren und ausprobieren.

Stefan

strolch101, 2020-07-22:
Hello,

sorry for the late response / late issue closing. New tiscamera version 0.12 fixed the problem. It wasn't even necessary to uncomment the mentioned section, but it is good to know - just in case. Thank you for your support!

closed 2020-07-22 by strolch101