TheImagingSource/tiscamera #346: gst-launch-1.0 tcambin ! videoconvert ! ximagesink

ericherbert, 2020-05-27:
Dear TIS-Stefan

I'm experiencing trouble with the video stream using a fresh install of Ubuntu 18.04 and Ubuntu 20.04. the camera DMK 23UM021 is well recognised and it works with GuvcView.

kernel version is 
> uname -rs

After compilation, the command  gst-launch-1.0 tcambin ! videoconvert ! ximagesink gives:

>gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Définition du pipeline à PAUSED...
ERREUR : le pipeline refuse de se mettre en pause.
ERREUR : de l’élément /GstPipeline:pipeline0/GstTcamBin:tcambin0 : Could not create element 'tcamautoexposure'.
Information de débogage supplémentaire :
Could not create element 'tcamautoexposure'.
Définition du pipeline à NULL…
Libération du pipeline…

pardon my french :)

Because of the python dependency I give you the package version of the Python 3.8.2 installation: 

>Package                Version      
alabaster              0.7.8        
appdirs                1.4.4        
apturl                 0.5.2        
attrs                  19.3.0       
Babel                  2.6.0        
bcrypt                 3.1.7        
beautifulsoup4         4.8.2        
blinker                1.4          
Brlapi                 0.7.0        
certifi                2019.11.28   
chardet                3.0.4        
command-not-found      0.3          
cryptography           2.8          
cupshelpers            1.0          
cycler                 0.10.0       
Cython                 0.29.19      
dbus-python            1.2.16       
decorator              4.4.2        
defer                  1.0.6        
distro                 1.4.0        
distro-info            0.23ubuntu1  
docutils               0.16         
duplicity              0.8.12.0     
entrypoints            0.3          
fasteners              0.14.1       
future                 0.18.2       
gmsh-interop           2019.1       
html5lib               1.0.1        
httplib2               0.14.0       
idna                   2.8          
imageio                2.8.0        
imagesize              1.2.0        
Jinja2                 2.10.1       
keyring                18.0.1       
kiwisolver             1.2.0        
language-selector      0.1          
launchpadlib           1.10.13      
lazr.restfulclient     0.14.2       
lazr.uri               1.0.3        
lockfile               0.12.2       
louis                  3.12.0       
lxml                   4.5.0        
macaroonbakery         1.3.1        
Mako                   1.1.0        
Markdown               3.1.1        
MarkupSafe             1.1.0        
matplotlib             3.2.1        
MeshPy                 2018.2.1     
monotonic              1.5          
more-itertools         8.3.0        
natsort                7.0.1        
netifaces              0.10.4       
networkx               2.4          
numpy                  1.18.4       
oauthlib               3.1.0        
olefile                0.46         
opencv-python          3.4.8.29     
packaging              20.4         
pandas                 1.0.3        
paramiko               2.6.0        
pause                  0.2          
pexpect                4.6.0        
Pillow                 7.1.2        
pip                    20.0.2       
PIPython               2.1.1.2      
pluggy                 0.13.1       
protobuf               3.6.1        
py                     1.8.1        
pycairo                1.16.2       
pycups                 1.9.73       
Pygments               2.3.1        
PyGObject              3.36.0       
PyJWT                  1.7.1        
pymacaroons            0.13.0       
PyNaCl                 1.3.0        
pyparsing              2.4.7        
PyQt5                  5.14.1       
pyRFC3339              1.1          
pyserial               3.4          
pytest                 5.4.2        
python-apt             2.0.0        
python-dateutil        2.8.1        
python-dbusmock        0.19         
python-debian          0.1.36ubuntu1
pytools                2020.2       
pytz                   2020.1       
pyudev                 0.21.0       
PyWavelets             1.1.1        
pyxdg                  0.26         
PyYAML                 5.3.1        
reportlab              3.5.34       
requests               2.22.0       
requests-unixsocket    0.2.0        
roman                  2.0.0        
scikit-image           0.17.2       
scipy                  1.4.1        
SecretStorage          2.3.1        
setuptools             45.2.0       
Shapely                1.7.0        
simplejson             3.16.0       
sip                    4.19.21      
six                    1.15.0       
soupsieve              1.9.5        
Sphinx                 1.8.5        
systemd-python         234          
tifffile               2020.5.25    
tqdm                   4.46.0       
trackpy                0.4.2        
ubuntu-advantage-tools 20.3         
ubuntu-drivers-common  0.0.0        
ufw                    0.36         
unattended-upgrades    0.1          
urllib3                1.25.8       
usb-creator            0.3.7        
wadllib                1.3.3        
wcwidth                0.1.9        
webencodings           0.5.1        
wheel                  0.34.2       
xkit                   0.0.0        
Python 3.8.2 (default, Apr 27 2020, 15:53:34) 
[GCC 9.3.0] on linux

TIS-Stefan, 2020-06-15:
Hello

The issue "could not create element 'tcamautoexposure'." is caused by by the exposure property name of the camera. 
In order to solve the issue for your 23U perform following steps:
Perform git pull on your tiscamera in order to get tiscamera 0.12.0 latest version.

Change in file  tiscamera/src/v4l2/v4l2_property_mapping.h starting a line  40:
```
{
	TCAM_PROPERTY_EXPOSURE,
	{ V4L2_CID_EXPOSURE_ABSOLUTE, V4L2_CID_EXPOSURE,
	// TCAM_V4L2_EXPOSURE_TIME_US
	},
},
```
to

```
{
	TCAM_PROPERTY_EXPOSURE,
	{ V4L2_CID_EXPOSURE_ABSOLUTE, V4L2_CID_EXPOSURE,
	 TCAM_V4L2_EXPOSURE_TIME_US
	},
},

```
Then rebuild, install and test.

Stefan

ericherbert, 2020-06-19:
ok thanks, the bug have been fixed by the firmware udate.

closed 2020-06-19 by ericherbert