TheImagingSource/tiscamera #349: regression `tcamsrc` *V0.12.0* : hangs again on `set_state(Gst.State.NULL)` reporting "Error closing udev monitoring pipe."

jraygauthier, 2020-06-02:
#288 again (or a variant of). Observed this on both start and stop. 

On 0.12.0 tag:

```bash
$ tcam-ctrl --version
Versions: 
        Tcam:   0.12.0_/_rev_
        Aravis: not_configured_version_not_configured
```

Was previously fixed as part 0.11.1 unreleased changesets (was at `2042bba574c2b4280027b041881afae5a9e000d8`).

Device is:

```bash
$ tcam-ctrl --list
Available devices:
Model           Type    Serial

DFM 37UX264-ML  V4L2    25910066
```

Both c3e3d26 and 628b6ff changesets fixing the issue are indeed part of the release. It must be some new code that reintroduce the issue.

Here's some (truncated) log (on start) (`TCAM_LOG=DEBUG GST_DEBUG=WARNING`):

```bash
$ 2>&1 ./reproduce_bug.sh
57:55.873 INFO    test_035_tiscamera_raw.py          391  expected_frame_count: 12, actual_frame_count: 12, acceptable_frame_count: 9
57:55.873 INFO    test_035_tiscamera_raw.py          321  ====== Iteration 3 =======
57:55.873 INFO    test_035_tiscamera_raw.py          30   tcam_src = Gst.ElementFactory.make("tcamsrc")
57:55.873 INFO    test_035_tiscamera_raw.py          34   tcam_src.get_device_serials()
57:55.875 INFO    test_035_tiscamera_raw.py          39   tcam_src.set_property("serial", 25910066)
2020.06.01T19:57:55:876        <INFO> V4l2Device.cpp:436: Current framerate is 35 / 1 fps
2020.06.01T19:57:55:876        <WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
2020.06.01T19:57:55:876        <INFO> V4l2Device.cpp:1222: Encountered write only control.
2020.06.01T19:57:55:877        <INFO> V4l2Device.cpp:891: Detected multiple exposure interfaces. Simplifying
2020.06.01T19:57:55:877        <DEBUG> V4l2Device.cpp:725: Found format: RGGB
2020.06.01T19:57:55:877        <DEBUG> V4l2Device.cpp:725: Found format: RG16
2020.06.01T19:57:55:877        <DEBUG> V4l2Device.cpp:267: Returning 2 formats.
2020.06.01T19:57:55:877        <DEBUG> PipelineManager.cpp:127: Received 2 formats.
2020.06.01T19:57:55:877        <DEBUG> PropertyHandler.cpp:273: Created 42 properties
57:55.877 INFO    test_035_tiscamera_raw.py          225  _configure_tcam_source: {use_soft_trig: True}
2020.06.01T19:57:55:878        <DEBUG> V4l2Device.cpp:1411: Changed ctrl Strobe Enable to value 1.
# ..
2020.06.01T19:57:55:959        <DEBUG> V4l2Device.cpp:1411: Changed ctrl Trigger Mode to value 1.
57:55.959 INFO    test_035_tiscamera_raw.py          102  _make_pipeline
57:55.970 INFO    test_035_tiscamera_raw.py          335  pipeline.set_state(Gst.State.PLAYING)
2020.06.01T19:57:55:972        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2020.06.01T19:57:55:973        <DEBUG> V4l2Device.cpp:577: Stopped stream
2020.06.01T19:57:55:973        <DEBUG> V4l2Device.cpp:181: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
2020.06.01T19:57:55:973        <DEBUG> V4l2Device.cpp:403: Setting framerate to '35.000000' =  35 / 1
2020.06.01T19:57:55:974        <INFO> V4l2Device.cpp:436: Current framerate is 35 / 1 fps
2020.06.01T19:57:55:974        <DEBUG> V4l2Device.cpp:247: Active format is: 'format=RGGB,width=2448,height=2048,framerate=35.000000'
2020.06.01T19:57:55:974        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RGGB' - 42474752
2020.06.01T19:57:55:974        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RG16' - 36314752
2020.06.01T19:57:55:974        <DEBUG> V4l2Device.cpp:181: Requested format change to 'format=RGGB,width=2448,height=2048,framerate=35.000000' 42474752
2020.06.01T19:57:55:975        <DEBUG> V4l2Device.cpp:403: Setting framerate to '35.000000' =  35 / 1
2020.06.01T19:57:55:976        <INFO> V4l2Device.cpp:436: Current framerate is 35 / 1 fps
2020.06.01T19:57:55:976        <DEBUG> V4l2Device.cpp:247: Active format is: 'format=RGGB,width=2448,height=2048,framerate=35.000000'
2020.06.01T19:57:55:976        <INFO> ImageBuffer.cpp:38: allocating data buffer
# ..
2020.06.01T19:57:55:976        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2020.06.01T19:57:55:976        <INFO> PipelineManager.cpp:486: source ->  sink
2020.06.01T19:57:55:976        <INFO> ImageSink.cpp:51: Pipeline started playing
2020.06.01T19:57:55:976        <DEBUG> V4l2Device.cpp:1606: Will use 10 buffers
2020.06.01T19:57:55:976        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f42f4006af0) with length 5013504
# ..
2020.06.01T19:57:55:987        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
[... 143 lines cut ...]
0:00:26.310620397 23515      0x2755160 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<bayer2rgb5> current READY pending VOID_PENDING, desired next NULL
0:00:26.310650293 23515      0x2755160 INFO              GST_STATES gstelement.c:2676:gst_element_continue_state:<bayer2rgb5> completed state change to NULL
0:00:26.310663357 23515      0x2755160 INFO              GST_STATES gstelement.c:2579:_priv_gst_element_state_changed:<bayer2rgb5> notifying about state-changed READY to NULL (VOID_PENDING pending)
0:00:26.310680265 23515      0x2755160 INFO              GST_STATES gstbin.c:2951:gst_bin_change_state_func:<pipeline5> child 'bayer2rgb5' changed state to 1(NULL) successfully
0:00:26.310696006 23515      0x2755160 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<src_video_caps_filter> current READY pending VOID_PENDING, desired next NULL
0:00:26.310709037 23515      0x2755160 INFO              GST_STATES gstelement.c:2676:gst_element_continue_state:<src_video_caps_filter> completed state change to NULL
0:00:26.310724006 23515      0x2755160 INFO              GST_STATES gstelement.c:2579:_priv_gst_element_state_changed:<src_video_caps_filter> notifying about state-changed READY to NULL (VOID_PENDING pending)
0:00:26.310742025 23515      0x2755160 INFO              GST_STATES gstbin.c:2951:gst_bin_change_state_func:<pipeline5> child 'src_video_caps_filter' changed state to 1(NULL) successfully
0:00:26.310758316 23515      0x2755160 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<queue5> current READY pending VOID_PENDING, desired next NULL
0:00:26.310774978 23515      0x2755160 INFO              GST_STATES gstelement.c:2676:gst_element_continue_state:<queue5> completed state change to NULL
0:00:26.310786377 23515      0x2755160 INFO              GST_STATES gstelement.c:2579:_priv_gst_element_state_changed:<queue5> notifying about state-changed READY to NULL (VOID_PENDING pending)
0:00:26.310804415 23515      0x2755160 INFO              GST_STATES gstbin.c:2951:gst_bin_change_state_func:<pipeline5> child 'queue5' changed state to 1(NULL) successfully
0:00:26.310819554 23515      0x2755160 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<tcamsrc31> current READY pending VOID_PENDING, desired next NULL
0:00:26.310836840 23515      0x2755160 INFO                 tcamsrc gsttcamsrc.cpp:1297:gst_tcam_src_close_camera: Closing device
2020.06.01T20:02:05:700        <INFO> V4l2Device.cpp:601: notifying callbacks about lost device
0:00:26.311937510 23515      0x2c75a40 WARN                 tcamsrc gsttcamsrc.cpp:1139:gst_tcam_src_device_lost_callback:<tcamsrc31> error: Device lost (25910066)
0:00:26.311981012 23515      0x2c75a40 INFO        GST_ERROR_SYSTEM gstelement.c:2141:gst_element_message_full_with_details:<tcamsrc31> posting message: Device lost (25910066)
0:00:26.312042797 23515      0x2c75a40 INFO        GST_ERROR_SYSTEM gstelement.c:2168:gst_element_message_full_with_details:<tcamsrc31> posted error message: Device lost (25910066)
0:00:26.312056044 23515      0x2c75a40 WARN                 tcamsrc gsttcamsrc.cpp:1147:gst_tcam_src_device_lost_callback:<tcamsrc31> error: Device lost
0:00:26.312076833 23515      0x2c75a40 INFO        GST_ERROR_SYSTEM gstelement.c:2141:gst_element_message_full_with_details:<tcamsrc31> posting message: Device lost
0:00:26.312097261 23515      0x2c75a40 INFO        GST_ERROR_SYSTEM gstelement.c:2168:gst_element_message_full_with_details:<tcamsrc31> posted error message: Device lost
# Hung there again...
```

This prevent us from upgrading from 0.11.1 (w unreleased patches)  to 0.12.

jraygauthier, 2020-06-02:
A little experiment revealed that the issue is also fixed (on the 0.11.1 `2042bba574c2b4280027b041881afae5a9e000d8` changset) by cherry picking the c3e3d26
and 628b6ff changeset: <https://github.com/jraygauthier/tiscamera/commits/jrg/attempt_fix_hang_0_11>.

This goes a long way in the direction that the changesets were equivalent to the above mentionned patch.

 
Another experiment reveal that I was able to reproduce the issue on top of e72a8757e8b1b37a41cccd619bcd42af17e3ea41:

```bash
$ git log -4
commit e72a8757e8b1b37a41cccd619bcd42af17e3ea41 (HEAD)
Author: Edgar Thier <edgar.thier@theimagingsource.com>
Date:   Thu Feb 27 10:14:27 2020 +0100

    Update changelog

commit 34cee19aca2a985dfd7a9a27206f77b0358bb2de
Author: Edgar Thier <edgar.thier@theimagingsource.com>
Date:   Thu Feb 27 10:13:26 2020 +0100

    examples: Fix indentation

commit c3e3d26d4595dc6861440221010660af02987cc4
Author: Edgar Thier <edgar.thier@theimagingsource.com>
Date:   Thu Feb 27 10:11:38 2020 +0100

    gst: tcamsrc: do not call stop when device is lost
    
    Let the gstreamer main thread call gst_tcam_src_stop.

commit 628b6ff55885e31fe381e52712d5c2e08f5a961f
Author: Edgar Thier <edgar.thier@theimagingsource.com>
Date:   Thu Feb 27 09:57:41 2020 +0100

    v4l2: Fix thread lock-up during destruction
```

So the problem must have been introduced in between 2042bba574c2b4280027b041881afae5a9e000d8 and 628b6ff55885e31fe381e52712d5c2e08f5a961f.

TIS-Edgar, 2020-06-04:
Hi there,

Sorry for the late reply.
Your reproduce bug script does what exactly? The same as the scripts in 'reproduce_bug_51_384_hangs_on_set_state_null.zip'?

As you might have guessed I am still not able to reproduce this bug locally.
I would really like to know your hardware setup to finally reproduce it myself.

> So the problem must have been introduced in between 2042bba and 628b6ff.

But 628b6ff fixes the problem, correct? If so, that means we are dealing with a regression.
There were no changes that should have caused that to my knowledge but lets see what we find.

Could you run a git bisect?
```
git checkout master
git bisect start
git bisect bad
git bisect good 628b6ff
```

The commit you end up with should tell us what causes the problem.

jraygauthier, 2020-06-08:
Hi @TIS-Edgar : We hit this problem through a higher priority task (retrieving meta data) which we finally decided as a fallback to stay on 0.11.1 which is fine for the client at this time.

We raised another separate issue our client's bug tracker for addressing the current situation with 0.12. The client establish that the issue is to be treated as low to med priority. We should hopefully come back to you at some point later with the required details and some more time to work on this particular issue.

> The same as the scripts in 'reproduce_bug_51_384_hangs_on_set_state_null.zip'?

In part, but it does a bit more which is currently part of the client's private repository. We might later on give you some privileged access (requiring some refactor on our part) so that you can reproduce using the same code using the same environment.

> Could you run a git bisect?

Out of time right now. Will do once the issue come to top of priority stack.

closed 2021-02-19 by TIS-Stefan

jraygauthier, 2022-07-12:
No longer an issue with tiscamera v1.