TheImagingSource/tiscamera #350: How to make the live streaming in my GUI

Izzatbrg, 2020-06-05:
Hey,
I made my own GUI and I'm trying to make the live streaming of tiscamera in my GUI without appearing of the automatic tiscamera GUI  how can I make both of them.Thanks in advance

TIS-Stefan, 2020-06-15:
I am sorry, but I do not really understand your question. It seems you having a C++ or Python program already, so you may look at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples. There is shown how to get image in memory for rendering on a GUI.

For better help I need to know which GUI and programming language and also which camera model you use.

Stefan

Izzatbrg, 2020-06-15:
yes,I have a QtPy5 GUI this GUI content a push-Button to start the streaming and stop button to end it what I'm trying to do is to stream the live streaming form tis camera to my pyqt5 gui

TIS-Stefan, 2020-06-15:
You must pass the winid() of a widget as overlay to the ximagesink:

http://gstreamer-devel.966125.n4.nabble.com/Qt5-gstreamer-and-QWidget-td4660183.html

Or you may look at https://gstreamer.freedesktop.org/data/doc/gstreamer/head/qt-gstreamer/html/qtvideosink_overview.html
Which looks for Gstreamer 1.2...

Izzatbrg, 2020-06-15:
thanks a lot

TIS-Stefan, 2020-06-15:
Reopen the issue, if there are more questions.

Stefan

Izzatbrg, 2020-06-18:
I tried to make it happened but till now I still facing problems,hier is my code,can you pleas tell me where is my mistake
`from PyQt5 import QtCore, QtGui, QtWidgets
from PyQt5.QtWidgets import *
import time
import sys
import gi
gi.require_version('Gst', '1.0')
gi.require_version('GstVideo', '1.0')
gi.require_version("Tcam", "0.1")
from gi.repository import GObject, Gst, GstVideo,Tcam

class Ui_MainWindow(object):
    def setupUi(self, MainWindow):
        MainWindow.setObjectName("MainWindow")
        MainWindow.resize(1234, 600)
        self.WinId = MainWindow.winId()
        camera=Cmaera(self.WinId)
        
        self.centralwidget = QtWidgets.QWidget(MainWindow)
        self.centralwidget.setObjectName("centralwidget")
        self.pushButton = QtWidgets.QPushButton(self.centralwidget)
        self.pushButton.setGeometry(QtCore.QRect(190, 300, 131, 81))
        self.pushButton.setObjectName("pushButton")
        self.pushButton.clicked.connect(camera.start_streaming)
        self.pushButton_2 = QtWidgets.QPushButton(self.centralwidget)
        self.pushButton_2.setGeometry(QtCore.QRect(850, 290, 131, 81))
        self.pushButton_2.setObjectName("pushButton_2")
        self.pushButton_2.clicked.connect(camera.stop_streaming)
        MainWindow.setCentralWidget(self.centralwidget)
        self.menubar = QtWidgets.QMenuBar(MainWindow)
        self.menubar.setGeometry(QtCore.QRect(0, 0, 1234, 26))
        self.menubar.setObjectName("menubar")
        MainWindow.setMenuBar(self.menubar)
        self.statusbar = QtWidgets.QStatusBar(MainWindow)
        self.statusbar.setObjectName("statusbar")
        MainWindow.setStatusBar(self.statusbar)
        self.retranslateUi(MainWindow)
        QtCore.QMetaObject.connectSlotsByName(MainWindow)

    def retranslateUi(self, MainWindow):
        _translate = QtCore.QCoreApplication.translate
        MainWindow.setWindowTitle(_translate("MainWindow", "MainWindow"))
        self.pushButton.setText(_translate("MainWindow", "Start"))
        self.pushButton_2.setText(_translate("MainWindow", "Stop"))

class Cmaera():
    def __init__(self,windowId):
        self.serial=None
        self.windowId=windowId
        Gst.init(sys.argv)
        self.pipeline = Gst.parse_launch('tcambin name=bin'
                                         '! videoconvert'
                                         '! ximagesink name=sink')
        self.camera = self.pipeline.get_by_name("bin")
        self.sink=self.pipeline.get_by_name("sink")
        self.setup_pipeline()
      #  Gst.set_window_handle (GST_VIDEO_OVERLAY(self.sink),self.windowId)
      #  Gst.gst_video_overlay_set_render_rectangle (GST_VIDEO_OVERLAY(self.sink),100,500,50,50)
    
    def setup_pipeline(self):
        self.state = Gst.State.NULL
        if self.serial is not None:
            camera.set_property("serial", self.serial)
        self.sink.set_window_handle(self.windowId)

    def start_streaming(self):
        print("ranning")
        self.pipeline.set_state(Gst.State.PLAYING)
        
    def stop_streaming(self):
        self.pipeline.set_state(Gst.State.NULL)
        
if __name__ == "__main__":
    import sys
    app = QtWidgets.QApplication(sys.argv)
    MainWindow = QtWidgets.QMainWindow()
    ui = Ui_MainWindow()
    ui.setupUi(MainWindow)
    MainWindow.show()
    sys.exit(app.exec_())`

TIS-Stefan, 2020-06-18:
Hello

Do you intend to display the live video on the mainwindow? Would'nt it be better to create an own widget 

video = Qwigdet 

and place that on the mainwindow with your other widgets and use that video.winID()?

Stefan

Izzatbrg, 2020-06-18:
yes for sure I will make it as a QtWidget, and to be sure do I use pass the overlay to the ximagesink correctly?
self.sink.set_window_handle(self.windowId)

TIS-Stefan, 2020-06-18:
I am not sure, when the winID() is valid. Maybe it is valid after you called 

 MainWindow.show()

closed 2021-02-19 by TIS-Stefan