TheImagingSource/tiscamera #351: Triggered camera and "select-timeout" errors

blazini36, 2020-06-07:
This is something I've dealt with for sometime but haven't addressed just yet and it's not an actual issue with tiscamera itself.

I have 2 DFK33UX264's (or something like that) and I've been using them in a machine under Linux that I have a programmer helping me with software for. The intended use of the camera's is always to have them be externally triggered by GPIO but there is a selectable mode to operate the cameras untriggered (normal video). They are used with OpenCV and v4l2.

That's all fine, that works....the issue is opening the capture device and not receiving an image causes this "select-timeout" error that will print a couple of times in the terminal then kill the program. Being that I use the cameras triggered, and they are triggered by motion. At the moment the "power" button on the GUI opens the capture device. If the capture device is opened, but no image has been received because the camera has not been triggered (nothings moving) then I will get the select timeout crash without fail.

Now my thought at the moment is to use the motion controller to pipe back a signal for Python to open or close the capture device if it's in triggered mode. I can do that, but I'm wondering if there is something else I can do about this before I do that. No uvcvideo quirk adresses this type of issue.

TIS-Stefan, 2020-06-15:
Hello

The timeout is of course expected, if no images come in. Since OpenCV has no callback mechanism, you can create an error handler for that timeout, if possible, ignore the timeout and restart the capture.
However, using an image callback for the incoming images is always better, since you do not have a blocking image polling loop. You may look at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples for some trigger samples in Python and C++.

Stefan

ShadeTechnik, 2020-06-28:
Is there any suggestion for a purely openCV fix. I more or less understand the callback mechanism in: https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/Using%20external%20trigger%20and%20callback%20in%20Python

But I don't use the tiscamera libraries as Gstreamer is incompatable with my application. cv2.videocapture.read() is what is opening the cap (and starting the countdown to a select-timeout error). But the cap needs to be opened to detect that image to start the callback. Not sure if I'm missing an opencv function or what.

TIS-Stefan, 2020-06-29:
Hello 

OpenCV has no callback mechanism, so you can not miss anything. 
If the camera runs in trigger mode, I would run the cv::videocapture::read() in a loop and ignore the timeouts. If there is a timeout, then the returned image is None. So your program can handle that.

Stefan

ShadeTechnik, 2020-07-01:
OK, I've been at the same project for over 2 years now. Running the cameras in trigger mode has always been the intention. The select-timeout has always been printing though we generally did what you said and "ignored it", didn't really seem to hurt anything.

The problem with that is that "select-timeout" became impossible to ignore. Without fail, if read() was started there would be 2 prints of "select-timeout"  and a crash taking down the program immediately after the 2nd,  and it wasn't clear to use whether it was our program or something else. So the intention was to avoid the case that would lead to the error.....hence the post. 1 single image read from the camera would prevent this behavior but I don't think it's good practice to apply hacky workarounds like swapping modes programatically to ingest an image. Now, in tracking the behavior we added a print statement to the case that cv2.videocapture.read() returned false and it turned out that it never did return false, even while select-timeout was printing. This makes it hard to handle the exception.

The programmer I work with found what he believes is a bug in OpenCV (though I don't know the details) from the latest version at least back to 3.2. I believe the bug was specifically in the grab function (read is grab and retrieve). He made a change to the source and recompile now this does not seem to be an issue. It's quite possible (and likely) that I was previously using an older version of opencv that at least did not crash the program and the grab/read function did not contain this bug.

closed 2021-02-19 by TIS-Stefan