TheImagingSource/tiscamera #352: tcam on Raspberry Pi not seeing camera

ben00052, 2020-06-15:
I am running into an odd behavior, and I am not sure the next step to troubleshoot.  I am trying to get a DFK27AUJ003 running on a Raspberry pi 3 B+ running Raspbian with a GUI desktop, and while the camera works fine with a v4l2 gstreamer pipeline, using any of the tcam utilities do not seem to be able to find the camera, preventing me from using exposure controls.

I had built the tcam utilities from the same source I used for a successful x86 centos build, and I am using the same camera.  Do I need to get a different code base or set a different cmake setting for an ARM system?

Some command line dumps to try and provide relevant information:

```
pi@raspberrypi:/usr/local/tiscamera/bin $ gst-launch-1.0 v4l2src ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstXImageSink:ximagesink0: Output window was closed
Additional debug info:
ximagesink.c(697): gst_x_image_sink_handle_xevents (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0
Execution ended after 0:00:02.201895810
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

pi@raspberrypi:/usr/local/tiscamera/bin $ gst-launch-1.0 tcambin ! videoconvert ! ximagesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

pi@raspberrypi:/usr/local/tiscamera/bin $ ./tcam-ctrl -l
No devices found.

pi@raspberrypi:/usr/local/tiscamera/bin $ lsusb
Bus 001 Device 009: ID 199e:8619 The Imaging Source Europe GmbH 
Bus 001 Device 008: ID 04e7:1c00 Elo TouchSystems 
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 413c:2111 Dell Computer Corp. 
Bus 001 Device 004: ID 045e:0040 Microsoft Corp. Wheel Mouse Optical
Bus 001 Device 007: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

TIS-Stefan, 2020-06-15:
Hello 

The DMK 27 does not work reliably on USB 2.0 ports. The PI3 has USB 2.0 only. 
The tambin sets the biggest possible video format and highest possible frame rate. I would guess, yo do not get even one complete image with these settings, therefore, you do not see any images at all.

You may try
`
tcam-ctrl -c <serial number of your camera> 
`
and start with a lower video format:
`
gst-launch-1.0 tcambin ! video/x-raw,format=BGRx,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink sync=false
`

Stefan

ben00052, 2020-06-15:
tcam-ctrl -c <serial> returns 
Unable to open device with serial "<serial>"

That being said, I suspect the USB 2.0 is the core of the issue, even with v4l2, I was not able to get full resolution images.  Are there any other model cameras to weigh against looking into a Raspberry pi 4 with USB 3.0 ?

TIS-Stefan, 2020-06-15:
Therefore, I wrote

tcam-ctrl -c <serial number of your camera>

where "<serial number of your camera>" has to be replaced by the serial number of your camera.

You have a 10 megapixel camera. That does not really makes sense on USB 2.0, because the camera sends the image data uncompressed, original sensor raw data. 
However, the DFK 37AUX178, 6MP or DFK 37AUX226 12MP could do the job. These cameras are intended to work on USB 2.0 too. I must admit, I only tested the DFK 37AUX178 on USB 2.0 intesively.

ben00052, 2020-06-15:
Yes, i did use 17714160, not '<'serial '>'.  sorry, forum ate my formatting, apparently it does not like less than and greater than characters.

TIS-Stefan, 2020-06-15:
tcam-ctrl -c 17714160

is the correct command.

ben00052, 2020-06-15:
just for clarity, i navigated to /usr/local/tiscamera/bin and ran
./tcamm-ctrl =c 17714160
and got
Unable to open device with serial "17714160"

TIS-Stefan, 2020-06-15:
Oho, that is weird. What gives
tcam-ctrl -l 
?
You may dis- and reconnect the camera.

stefan

ben00052, 2020-06-15:
that was back in the original post

pi@raspberrypi:/usr/local/tiscamera/bin $ ./tcam-ctrl -l
No devices found.
It seemed odd that the v4l2src stream worked (allbeit at reduced resolution) and lsusb saw the camera even when tcam-ctrl could not find the device

ben00052, 2020-06-15:
Camera still not seen after unplugging/re-plugging the usb or rebooting the pi

TIS-Edgar, 2020-06-16:
A few questions:

You are running version 0.12.0 (current master) correct?

Is there a reason you installed to /usr/local/tiscamera/ ? How are you adding the required custom settings to the environment?

Does TCAM_LOG=DEBUG add anything?

`Do I need to get a different code base or set a different cmake setting for an ARM system?`

Assuming native compilation no changes what so ever are required.
What cmake configuration do you use? Maybe we can replicate it.

TIS-Stefan, 2020-06-16:
Hello

It seems your camera uses an older firmware.
Please download the file(s)
 - Usb3Cam.508.fwpack
 from https://we.tl/t-8CixtxH6wr
For firmware update please look at https://www.theimagingsource.com/documentation/tiscamera/firmware-update.html

I am not sure, whether this helps, but this is the only difference I can find.

Stefan

ben00052, 2020-06-16:
Good call Edgar, the CMakeLists.txt has version 0.8.0 in it, getting that up to date should be my next step.  Unfortunately I do not have hardware access today, so I will have to get back to you tomorrow with further information.

I did compile on the pi itself, using a build script a former coworker originally set up for a Centos box, apparently longer ago than I realized.

ben00052, 2020-06-17:
Ok, the camera was found with 0.12.0.  I am getting a few issues since rebuilding finding the libraries, but that is a different issue.

Thank you for your patience, and pointing out the USB issue.  I suspect that is going to be important, even if it was not the root cause of the problem asked about.

closed 2021-02-19 by TIS-Stefan