TheImagingSource/tiscamera #355: Gstreamer can't reach above 30 fps on TIS cameras

hamedty, 2020-07-02:
I have TIS camera "DFK 27BUR0135" and I followed installation steps for linux. My camera has 120 fps capability:

`video/x-raw,format=GRAY8,width=640,height=480,framerate=120/1` (from tcam-ctrl).

I wrote a simple fps benchmark for the camera based in python and Gstreamer and it can never reach above 33 fps. While Gstreamer built-in "videotestsrc" easily reaches 120 fps:

`import time
import gi

gi.require_version("Tcam", "0.1")
gi.require_version("Gst", "1.0")
gi.require_version("GstVideo", "1.0")

from gi.repository import Tcam, Gst, GstVideo

Gst.init()

def calculate_pipeline_fps(pipeline_string):
	pipeline = Gst.parse_launch(pipeline_string + " ! appsink name=sink")
	pipeline.set_state(Gst.State.PLAYING)
	sink = pipeline.get_by_name("sink")

	print("Start benchmarking " + pipeline_string)
	N = 100
	t0 = time.time()
	for i in range(N):
		sample = sink.emit("pull-sample")
	t1 = time.time()
	calculated_fps = N / (t1 - t0)
	print("FPS = %.2f" % calculated_fps)
	pipeline.set_state(Gst.State.NULL)

def main():
	# Calculating TIS @120FPS
	calculate_pipeline_fps(
		"tcambin"
		" ! video/x-raw,format=GRAY8,width=640,height=480,framerate=120/1"
	)

	# Calculating Test Video @120FPS
	calculate_pipeline_fps(
		"videotestsrc"
		" ! video/x-raw,format=GRAY8,width=640,height=480,framerate=120/1"
	)

if \_\_name\_\_ == "\_\_main\_\_":
	main()`

PS: I did similar test for python3+CV2+V4L2. When I ask for upto 90 fps everything was running smoothly. When I switched to 120 fps it rolled back to 60 fps.

hamedty, 2020-07-02:
I guess it has something to do with exposure time. I finally was able to get 120 fps using v4l2 by setting "CAP_PROP_EXPOSURE" to 84. Perhaps 84 is mapped to 8.333 ms in TIS firmware (=120fps).

TIS-Stefan, 2020-07-02:
You are right, if the exposure time is too long, then the frame rate goes down. I must admit, I am puzzled about the values for exposure time.

hamedty, 2020-07-04:
I am not agree that this issue is closed. This seems to me to be a deeper issue. I am not able to set CAP_PROP_EXPOSURE to 83. Only 84 works with 119 fps. 83 yield back to 60 fps! Very wired behavior. Also FPS priority must be above EXPOSURE while I explicitly set FPS and not EXPOSURE.

I spent many years of my professional life developing graphics driver at AMD at this driver for such simple hardware doesn't look good to me (nor for TIS reputation).

TIS-Stefan, 2020-07-06:
Hello
I tested the camera DFK 27BUR0135, firmware version 508  with pure v4l2 usage in a C++ program on an x86 computer, without GStreamer. I got following result:

```
$ ./v4l2-frameratetest 
Available cameras:
 1 : DFK 27BUR0135	13719906	/dev/video0
 2 : DFK 27BUR0135	13719906	/dev/video1
Your choice:1
Testing : DFK 27BUR0135	13719906	/dev/video0
Start capturing GRBG 1280x960 at 60.000000 fps
[PASSED] 59.959127 fps

Start capturing GRBG 1280x960 at 50.000000 fps
[PASSED] 49.949704 fps

Start capturing GRBG 1280x960 at 40.000000 fps
[PASSED] 40.023831 fps

Start capturing GRBG 1280x960 at 30.000000 fps
[PASSED] 29.979733 fps

Start capturing GRBG 1280x960 at 20.000000 fps
[PASSED] 20.019678 fps

Start capturing GRBG 1280x960 at 10.000000 fps
[PASSED] 9.989954 fps

Start capturing GRBG 1280x720 at 80.000000 fps
[PASSED] 80.127684 fps

Start capturing GRBG 1280x720 at 60.000000 fps
[PASSED] 60.032216 fps

Start capturing GRBG 1280x720 at 50.000000 fps
[PASSED] 50.013723 fps

Start capturing GRBG 1280x720 at 40.000000 fps
[PASSED] 39.959797 fps

Start capturing GRBG 1280x720 at 30.000000 fps
[PASSED] 30.036410 fps

Start capturing GRBG 1280x720 at 20.000000 fps
[PASSED] 20.031882 fps

Start capturing GRBG 1280x720 at 10.000000 fps
[PASSED] 9.989935 fps

Start capturing GRBG 1024x768 at 80.000000 fps
[PASSED] 80.041530 fps

Start capturing GRBG 1024x768 at 70.000000 fps
[PASSED] 70.000490 fps

Start capturing GRBG 1024x768 at 60.000000 fps
[PASSED] 59.985480 fps

Start capturing GRBG 1024x768 at 40.000000 fps
[PASSED] 40.020746 fps

Start capturing GRBG 1024x768 at 30.000000 fps
[PASSED] 30.012847 fps

Start capturing GRBG 1024x768 at 20.000000 fps
[PASSED] 20.010378 fps

Start capturing GRBG 1024x768 at 10.000000 fps
[PASSED] 9.997603 fps

Start capturing GRBG 640x480 at 120.000000 fps
[PASSED] 120.079309 fps

Start capturing GRBG 640x480 at 90.000000 fps
[PASSED] 90.059578 fps

Start capturing GRBG 640x480 at 60.000000 fps
[PASSED] 59.988561 fps

Start capturing GRBG 640x480 at 30.000000 fps
[PASSED] 30.019793 fps

Start capturing GRBG 640x480 at 15.000000 fps
[PASSED] 14.997137 fps

Start capturing GRBG 640x480 at 10.000000 fps
[PASSED] 9.998156 fps

Start capturing GREY 1280x960 at 60.000000 fps
[PASSED] 59.959411 fps

Start capturing GREY 1280x960 at 50.000000 fps
[PASSED] 49.949600 fps

Start capturing GREY 1280x960 at 40.000000 fps
[PASSED] 40.023738 fps

Start capturing GREY 1280x960 at 30.000000 fps
[PASSED] 29.979700 fps

Start capturing GREY 1280x960 at 20.000000 fps
[PASSED] 20.019465 fps

Start capturing GREY 1280x960 at 10.000000 fps
[PASSED] 9.989921 fps

Start capturing GREY 1280x720 at 80.000000 fps
[PASSED] 80.127612 fps

Start capturing GREY 1280x720 at 60.000000 fps
[PASSED] 60.032090 fps

Start capturing GREY 1280x720 at 50.000000 fps
[PASSED] 50.013660 fps

Start capturing GREY 1280x720 at 40.000000 fps
[PASSED] 39.959800 fps

Start capturing GREY 1280x720 at 30.000000 fps
[PASSED] 30.036430 fps

Start capturing GREY 1280x720 at 20.000000 fps
[PASSED] 20.031901 fps

Start capturing GREY 1280x720 at 10.000000 fps
[PASSED] 9.989938 fps

Start capturing GREY 1024x768 at 80.000000 fps
[PASSED] 80.041676 fps

Start capturing GREY 1024x768 at 70.000000 fps
[PASSED] 69.999821 fps

Start capturing GREY 1024x768 at 60.000000 fps
[PASSED] 59.985465 fps

Start capturing GREY 1024x768 at 40.000000 fps
[PASSED] 40.020557 fps

Start capturing GREY 1024x768 at 30.000000 fps
[PASSED] 30.012889 fps

Start capturing GREY 1024x768 at 20.000000 fps
[PASSED] 20.010374 fps

Start capturing GREY 1024x768 at 10.000000 fps
[PASSED] 9.997585 fps

Start capturing GREY 640x480 at 120.000000 fps
[PASSED] 120.079382 fps

Start capturing GREY 640x480 at 90.000000 fps
[PASSED] 90.059384 fps

Start capturing GREY 640x480 at 60.000000 fps
[PASSED] 59.988532 fps

Start capturing GREY 640x480 at 30.000000 fps
[PASSED] 30.019812 fps

Start capturing GREY 640x480 at 15.000000 fps
[PASSED] 14.997144 fps

Start capturing GREY 640x480 at 10.000000 fps
[PASSED] 9.998091 fps

Start capturing Y16  1280x960 at 70.000000 fps
[PASSED] 70.035192 fps

Start capturing Y16  1280x960 at 60.000000 fps
[PASSED] 59.959516 fps

Start capturing Y16  1280x960 at 50.000000 fps
[PASSED] 49.949551 fps

Start capturing Y16  1280x960 at 40.000000 fps
[PASSED] 40.023868 fps

Start capturing Y16  1280x960 at 30.000000 fps
[PASSED] 29.979695 fps

Start capturing Y16  1280x960 at 20.000000 fps
[PASSED] 20.019504 fps

Start capturing Y16  1280x960 at 10.000000 fps
[PASSED] 9.989938 fps

Start capturing Y16  640x480 at 120.000000 fps
[PASSED] 120.079109 fps

Start capturing Y16  640x480 at 90.000000 fps
[PASSED] 90.059320 fps

Start capturing Y16  640x480 at 60.000000 fps
[PASSED] 59.988511 fps

Start capturing Y16  640x480 at 30.000000 fps
[PASSED] 30.019802 fps

Start capturing Y16  640x480 at 15.000000 fps
[PASSED] 14.997179 fps

Start capturing Y16  640x480 at 10.000000 fps
[PASSED] 9.997997 fps

SUCCESS: 65, FAIL: 0
The Test has [PASSED]

```
Sorry for the long output. However, it shows, that v4l2 and camera provide the wanted frame rates on my computer. 

If I use your source unchanged, I can reproduce the issue. If I pass  "sync=false" to the appsink, then I get ~116 fps
`
	pipeline = Gst.parse_launch(pipeline_string + " ! appsink sync=false name=sink")

`
What do you think about that?

Stefan

hamedty, 2020-07-07:
Then I would apologize and get my words back :). Sorry and thank you for your investigation.

hamedty, 2020-07-07:
I honestly moved on this part by using CAP_PROP_EXPOSURE and I am developing the rest of my project. If I had anything else to add later on, I'll comment here. Thanks.

closed 2020-07-07 by hamedty

TIS-Stefan, 2020-07-07:
Hello Hamedty

No need to apologize, because I also do not know all and need to ask my team often.

Stefan