TheImagingSource/tiscamera #356: Cannot set camera image properties

lmollinedo, 2020-07-07:
Hello,

We are using a DMK 37BUX178 camera, with serial number 07020663.

Via tcam-capture we can set image sharpness, denoise, contrast & gamma properties:

[image]
 
But these parameters are not present in the camera properties list:
```
$ tcam-ctrl --properties 07020663
Found 33 propert(y/ies)
Brightness                              (int) min=0 max=4095 step=1 default=200 value=800 category=Exposure
Gain                                    (int) min=0 max=480 step=1 default=0 value=15 category=Exposure
Exposure Time (us)                      (int) min=20 max=60000000 step=1 default=333 value=30000 category=Unknown
Exposure Auto                           (bool) default=true value=false category=Exposure
Exposure Auto Reference                 (int) min=0 max=255 step=1 default=128 value=0 category=Exposure
Gain Auto                               (bool) default=true value=false category=Exposure
Trigger Mode                            (bool) default=false value=false category=Special
Software Trigger                        (button) category=Special
Trigger Delay (us)                      (int) min=0 max=1000000 step=1 default=0 value=31 category=Special
Strobe Enable                           (bool) default=false value=false category=Special
Strobe Polarity                         (bool) default=false value=false category=Special
Strobe Exposure                         (bool) default=true value=false category=Special
Strobe Duration                         (int) min=0 max=32767 step=1 default=100 value=0 category=Special
Strobe Delay                            (int) min=0 max=32767 step=1 default=0 value=0 category=Special
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special
GPIn                                    (int) min=0 max=1 step=1 default=0 value=0 category=Special
Offset X                                (int) min=0 max=2816 step=4 default=0 value=0 category=Partial Scan
Offset Y                                (int) min=0 max=2044 step=4 default=0 value=0 category=Partial Scan
Offset Auto Center                      (bool) default=true value=false category=Partial Scan
Trigger Polarity                        (enum)  default=Rising Edge category=Special
						value=Rising Edge
							Falling Edge
							Rising Edge
Trigger Operation                       (enum)  default=Default category=Special
						value=Global Reset Release
							Default
							Global Reset Release
Auto Functions ROI Control              (bool) default=true value=false category=Special
Reverse X                               (bool) default=false value=false category=Image
Reverse Y                               (bool) default=false value=false category=Image
Highlight Reduction                     (bool) default=false value=false category=Exposure
Exposure Auto Upper Limit Auto          (bool) default=true value=false category=Exposure
Exposure Auto Lower Limit               (int) min=20 max=1000000 step=1 default=333 value=333 category=Exposure
Exposure Auto Upper Limit               (int) min=20 max=1000000 step=1 default=333 value=333 category=Exposure
Override Scanning Mode                  (int) min=1 max=16 step=1 default=1 value=1 category=Partial Scan
Auto Functions ROI Preset               (enum)  default=Center 50% category=Special
						value=Full Sensor
							Center 50%
							Full Sensor
Gain Auto Lower Limit                   (int) min=0 max=480 step=1 default=0 value=0 category=Exposure
Gain Auto Upper Limit                   (int) min=0 max=480 step=1 default=480 value=0 category=Exposure
Trigger Global Reset Release            (bool) default=false value=false category=Special
```
And we cannot set them via 'set_tcam_property method' method in our Python code:
```
set_tcam_property("Sharpness", 7)
set_tcam_property("Denoise", 5)
set_tcam_property("Contrast", 140)
set_tcam_property("Gamma", 1.62)
```
No changes are seen in the resulting image.

Maybe we are missing a step during the image properties setting?

Thanks in advance,

Luis

TIS-Stefan, 2020-07-08:
Hello 

These properties are generated by software after a Gstreamer pipeline has been built and set to state PLAYING. tcam-ctrl does not use GStreamer and shows the pure camera properties only. Therefore, it does not show the software generated properties.

Stefan

lmollinedo, 2020-07-08:
Hi Stefan,

Thanks for your fast and useful answer! 

Changing Gstreamer pipeline state to PLAYING state before listing the parameters list or setting the parameters solves the problem.

Kind regards,

Luis

closed 2020-07-08 by lmollinedo