TheImagingSource/tiscamera #357: tcam-capture won't work on fresh install on TX2

Rdelux, 2020-07-10:
I tried to install the tools and resources from TIS' github on a DJI Manifold 2, which runs a Jetson TX2 processor but I got an error when I ran tcam-capture.
I had done this before on a laptop and I had no issue.  I followed the instruction here:
https://github.com/TheImagingSource/tiscamera

Doing the following provides a camera feed from my DFK 33UX265 camera:
gst-launch-1.0 tcambin ! videoconvert ! ximagesink

Tcam-ctrl -l also works, which returns the model and serial number.  However, when I ran tcam-capture, I got the following error
`dji@manifold2:~$ tcam-capture
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 33, in <module>
    from tcam_capture.TcamView import TcamView
  File "/usr/lib/python3/dist-packages/tcam_capture/TcamView.py", line 72
    self.caps_desc: CapsDesc = None
                  ^
SyntaxError: invalid syntax
`
What can I do to fix this issue?

Thanks

TIS-Stefan, 2020-07-10:
Hello

I would say, it is a Python configuration error. Which Python version is installed on your computer?

Stefan

Rdelux, 2020-07-10:
Hello Stefan,

Thanks for the reply.  The computer has Python 2.7.12 and 3.5.2, defaulting to Python 2 in a Ubuntu 16 environment.

Richard

TIS-Stefan, 2020-07-10:
Hi Richard

so it is normal Python and not Anaconda?
There must be a difference between your laptop and your DJI Manifold 2. The very same code runs fine on the laptop, but not on the DJI Manifold 2.  That proves, that the code is ok.  Unfortunately I never saw a DJI Manifold 2 on my own, therefore, I can not be of good help here, so you use find the differences.
I am sorry for my lack of knowledge here.

Stefan

closed 2021-02-19 by TIS-Stefan