TheImagingSource/tiscamera #359: some issues when i have installed two version of tiscamera

Jared-Shen, 2020-07-16:
When i have installed the latest version of tiscamera,as i run ‘tcam-ctrl’ ,i can find devices.

as my code has something incompatible with new version of tiscamera,so i installed the last version of tiscamera,but after i have finished it and run “tcam-ctrl -l”,i get only “No devices found!”

i have tried it two times, and i get entire same results.

maybe you can give me some advice on what has changed on these two versions and how to solve this problem. 

[image]

camera is shown above.

shy

TIS-Stefan, 2020-07-16:
Hello did you run cmake with

`cmake -DBUILD_ARAVIS=ON ..`

?

Stefan

Jared-Shen, 2020-07-16:
‘cmake -DBUILD_ARAVIS=OFF ..’

so you mean ARAVIS is needed? Sorry i am not clear about this. Would you give me some advice on what this is?

Thanks.

shy

TIS-Stefan, 2020-07-16:
Yes! You have a GigE camera. That needs Aravis GigEVision library.

closed 2021-02-19 by TIS-Stefan