TheImagingSource/tiscamera #360: How can I set software trigger using tcam-ctrl?

cocopa, 2020-07-30:
I confirmed the property of software trigger with "tcam-ctrl -p".
I couldn't setting software trigger  with "tcam-ctrl ".
How can I set software trigger using tcam-ctrl?

tcam-ctrl --load serial  '{"Trigger Mode": "true"}'
tcam-ctrl --load serial  '{"Software Trigger": "1"}' ->error!!

TIS-Stefan, 2020-07-30:
Hello 

Which camera model do you use?
Does tcam-ctrl -p list the software trigger property?
Stefan

cocopa, 2020-07-31:
Thank you for your reply.
It is as below.

xxxx@xxxx-MacBookPro:~$ tcam-ctrl -l
Available devices:
Model Type Serial

DMK 33UX264 V4L2 16******

xxxx@xxxx--MacBookPro:~$ tcam-ctrl --load 16710581 '{"Software Trigger": "1"}'
Loading: {"Software Trigger": "1"}
Unabe to apply '"1"' to 'Software Trigger'. Ignoring.

cocopa, 2020-07-31:
Thank you for your reply.
It is as below.

xxxx@xxxx-MacBookPro:~$ tcam-ctrl -l
Available devices:
Model Type Serial

DMK 33UX264 V4L2 16******

xxxx@xxxx--MacBookPro:~$ tcam-ctrl --load 16******  '{"Software Trigger": "1"}'
Loading: {"Software Trigger": "1"}
Unabe to apply '"1"' to 'Software Trigger'. Ignoring.

TIS-Stefan, 2020-07-31:
Hello

I am able to reproduce the issue and notified the programmers.
Background: It does not make sense to save one push button properties like Software Trigger and White Balance One Push. Unfortunately this lead to the situation, that they also can not be set by tcam-ctrl.

Is it possible to use software trigger in a program code as shown in the samples at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples ?

Stefan

cocopa, 2020-07-31:
Dear Stefan

Thank you for your reply.
As there was the property of software trigger in tcam-ctrl -p, it seemed working software trigger in tcam-ctrl .
So I had this issue. 
Of course, It could work sample program of Linux-tiscamera-Programming-Samples.
If possible, please fix to be set by tcam-ctrl.

Best Regards

TIS-Stefan, 2020-07-31:
It should be fixed in the development branch (at least on our local gitlab). 
The change is  (0e930237)  and all button properties shall work.

Stefan

davidsean, 2020-08-14:
Not sure if this can be useful to you @cocopa , but this is how I get it working on my end.
First make sure trigger mode is enabled, you can check using `tcam-ctrl -p `
If it is not set to true, change it with:
```bash
tcam-ctrl -p <PUT SERIAL HERE> -s "Trigger Mode=true"
```
afterwards, you can trigger it using 
```bash
tcam-ctrl -p <PUT SERIAL HERE> -s "Software Trigger=1"
```
and look for get a confirmation of `Found property!` printed on the console.  I set `Software Trigger=1` in this example, but I found other arguments to also work (like `Software Trigger=true`). 
In my case, this emits a single frame into the pipeline (that I setup in the background via another process).

closed 2021-02-19 by TIS-Stefan