TheImagingSource/tiscamera #363: tcam on Debian Buster (amd64 architecture on VirtualBox) fails for GRAY8 format: DBK 33UX178.AS

andreok, 2020-08-19:
Hello,

I tried tcam-capture on an amd64 architecture Debian Buster virtual machine using VirtualBox, and I can see the gray8/y800 format available (but not gray16/y16) for a DBK 33UX178.AS, but I'm getting the following errors when I try to use it and no frames are displayed:
"tcam-capture
[2020-08-17:14:48:29:129] [   ERROR] [         TcamView.py:259] - Given format caps could not be found in caps descriptions. video/x-raw,format=GRAY8,width=3072,height=2048,framerate=5/1
[2020-08-17:14:48:29:129] [   ERROR] [         TcamView.py:260] - Falling back to default behavior.
[2020-08-17:14:49:10:861] [   ERROR] [         TcamView.py:259] - Given format caps could not be found in caps descriptions. video/x-raw,format=GRAY8,width=3072,height=2048,framerate=5/1
[2020-08-17:14:49:10:862] [   ERROR] [         TcamView.py:260] - Falling back to default behavior.
[2020-08-17:14:49:45:414] [   ERROR] [         TcamView.py:259] - Given format caps could not be found in caps descriptions. video/x-raw,format=GRAY8,width=3072,height=2048,framerate=1/1
[2020-08-17:14:49:45:415] [   ERROR] [         TcamView.py:260] - Falling back to default behavior."

A similar no frames issue occurs with GStreamer and gray8 formats.

Regards,
Andre

TIS-Edgar, 2020-08-21:
Hi,

What pipeline are you using?

Could you run the pipeline with --gst-debug=tcamsrc:5,tcambin:5 so we can see what is going wrong.
tcam-capture uses gstreamer internally so fixing gstreamer should also fix tcam-capture.

TIS-Edgar, 2020-08-24:
Hi Andre,

The debug output indicates that the camera you are using does not support GRAY8 out of the box.
This means you will have to convert it to GRAY8.

One such pipeline would be:
```
gst-launch-1.0 tcamsrc serial=26020406 ! bayer2rgb ! videoconvert ! video/x-raw,format=GRAY8,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
```
To get a static overview over the caps you camera offers enter `tcam-ctrl -c 26020406 ` into a console.

andreok, 2020-08-24:
Hello Edgar,

Thank you for the information. The pipeline you suggested worked just fine.
The same approach could be applied to GRAY16?

Also, I want to be able to capture raw bayered frames, for later on-software debayering (outside GStreamer). In this case, I could drop the bayer2rgb and use the following pipeline as a general solution (for both cameras supporting natively GRAY8 and not supporting it)?
`gst-launch-1.0 tcamsrc serial=26020406 ! videoconvert ! video/x-raw,format=GRAY8,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink`

Regards,
Andre

TIS-Edgar, 2020-08-25:
Hi Andre,

> The same approach could be applied to GRAY16?
Yes.
> In this case, I could drop the bayer2rgb and use the following pipeline as a general solution (for both cameras supporting natively GRAY8 and not supporting it)?
That should be possible, yes. Check the gstreamer output like we did above to ensure the source sets the correct output format and it work fine.

Regards,
Edgar

andreok, 2020-08-25:
Hello Edgar,

Both GRAY8 and GRAY16_LE formats worked with the videoconvert before the caps format data, but I'm getting only monochrome output with this pipeline and not the raw bayered output as expected for a DBK camera:
`gst-launch-1.0 tcamsrc serial=26020406 ! videoconvert ! video/x-raw,format=GRAY8,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink`

I want to support only Case B and Case C debayering techniques, as described [here](https://www.astronomycameras.com/blog/archive/20080515/dbk-and-dfk-series-debayering-techniques/). 
I expect all cameras to output GRAY8/Y800 and GRAY16_LE/Y16 frames so I can receive bayered raw data from color cameras (DBK and DFB series), and debayering should be done on-software, accordingly.

Regards,
Andre

TIS-Edgar, 2020-08-26:
That is a rather old article you found there. All current cameras we have do not offer bayer declared as mono (to my knowledge).

To manually redeclare the caps to mono use the capssetter element.
`gst-launch-1.0 -v tcamsrc ! video/x-bayer,width=640,height=480,framerate=30/1 ! capssetter replace=true join=false caps="video/x-raw,format=GRAY8" ! video/x-raw,format=GRA
Y8,width=640,height=480,framerate=30/1  ! videoconvert ! ximagesink `

Alternatively feed your application with bayer and only declare it mono when you process it.

`tcamsrc ! video/x-bayer ! appsink` and then something like `appsrc ! video/x-raw,format=GRAY8 ! <encoder> ! multifilesink`

andreok, 2020-08-26:
Hello Edgar,

Unfortunately, I'm porting an application from IC Imaging Control C Library on Windows to tiscam on Linux, so I would like to keep things as similar as possible.

Your first option worked just fine. 
In this case, programmatically, I could replace both caps filters by `capsfilter name="..."` in the pipeline and proceed as usual, but how could I adjust the capsstter's caps format accordingly (GRAY8/GRAY16_LE)?

Regards,
Andre

TIS-Edgar, 2020-08-27:
You require 12 bit bayer right?

In that case we need to switch to a different backend. The uvc/v4l2 kernel modules already received a patch from us to add 12 bit bayer support. Sadly it is not yet merged. That is why we need to switch to aravis usb3vision.

You need to rebuild with the cmake flag TCAM_ARAVIS_USB_VISION=ON.
After that you can simply append `-aravis` to the serial the switch between backends.

`tcam-ctrl -c 26020406-aravis` should also list bayer 12/16 caps.

```
gst-launch-1.0 -v tcamsrc serial=16020406-aravis ! video/x-bayer,format=<format for your dev here>,width=640,height=480,framerate=30/1 ! capssetter replace=true join=false caps="video/x-raw,format=GRAY16_LE" ! video/x-raw,format=GRAY16_LE,width=640,height=480,framerate=30/1 ! videoconvert ! ximagesink
```

andreok, 2020-08-27:
Hello Edgar,

Correct, I need to provide an option to use 12-bit bayer from the camera and output as 16-bit bayered frames (GRAY16_LE).

Unfortunately, I cannot build tiscam anymore to include the cmake flag TCAM_ARAVIS_USB_VISION=ON.
Looks like a commit to the master branch of [Aravis](https://github.com/AravisProject/aravis) from 4 days ago is causing some issues:
`make
[  0%] Performing build step for 'aravis'
Making all in src
  CC       arvuvcp.lo
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c: In function ‘arv_uvcp_packet_type_to_string’:
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c:127:29: error: ‘ARV_TYPE_UVCP_PACKET_TYPE’ undeclared (first use in this function); did you mean ‘ARV_TYPE_GVCP_PACKET_TYPE’?
  return arv_enum_to_string (ARV_TYPE_UVCP_PACKET_TYPE, value);
                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                             ARV_TYPE_GVCP_PACKET_TYPE
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c:127:29: note: each undeclared identifier is reported only once for each function it appears in
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c: In function ‘arv_uvcp_command_to_string’:
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c:133:29: error: ‘ARV_TYPE_UVCP_COMMAND’ undeclared (first use in this function); did you mean ‘ARV_TYPE_GVCP_COMMAND’?
  return arv_enum_to_string (ARV_TYPE_UVCP_COMMAND, value);
                             ^~~~~~~~~~~~~~~~~~~~~
                             ARV_TYPE_GVCP_COMMAND
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c: In function ‘arv_uvcp_packet_type_to_string’:
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c:128:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c: In function ‘arv_uvcp_command_to_string’:
/home/andre/tiscamera/dependencies/aravis/src/arvuvcp.c:134:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[5]: *** [Makefile:952: arvuvcp.lo] Error 1
make[4]: *** [Makefile:702: all] Error 2
make[3]: *** [Makefile:583: all-recursive] Error 1
make[2]: *** [dependencies/CMakeFiles/aravis.dir/build.make:112: dependencies/aravis-prefix/src/aravis-stamp/aravis-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:125: dependencies/CMakeFiles/aravis.dir/all] Error 2
make: *** [Makefile:152: all] Error 2`

It is possible to force the ARAVIS_0_8_0 tag or even the older ARAVIS_0_7_5 tag via a cmake flag?

Regards,
Andre

TIS-Edgar, 2020-09-01:
Hi Andre,

Switching to aravis 0.7 or aravis 0.8 is planned but not yet implemented.
It requires some changes in our backend as aravis had some API changes with 0.7.
The project also switched from autotools to meson/ninja making automated builds on older systems a bit more difficult.

If you are interested I can make a cleanup of our current test code and drop it on a separate branch.
That would, however be without any guarantees as we did not yet have time to properly test everything and are currently focused on other things.

andreok, 2020-09-08:
Hello Edgar,

Sorry, looks like the Aravis build issue was fixed now, and I finally managed to get tiscam built with it enabled.
But, unfortunately, I could get any frame displayed with the following command:
`gst-launch-1.0 --gst-debug=tcamsrc:5,tcambin:5 tcamsrc serial=26020406-aravis ! video/x-bayer,width=640,height=480,framerate=1/1 ! capssetter replace=true join=false caps="video/x-raw,format=GRAY8" ! video/x-raw,format=GRAY8,width=640,height=480,framerate=1/1 ! videoconvert ! ximagesink`
Here're the last lines of the console output:
`0:00:10.240279042 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:799:gst_tcam_src_negotiate:<tcamsrc0> caps of peer: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1; video/x-bayer, format=(string)rggb16, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.240313505 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:803:gst_tcam_src_negotiate: Peer gave us something to work with.
0:00:10.240341117 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:823:gst_tcam_src_negotiate:<tcamsrc0> peer: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.240385745 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:834:gst_tcam_src_negotiate:<tcamsrc0> intersect: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.240467989 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:1681:gst_tcam_src_fixate_caps:<tcamsrc0> Fixated caps to video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.240536234 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:949:gst_tcam_src_negotiate:<tcamsrc0> fixated to: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.240556864 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:1013:gst_tcam_src_set_caps: In tcam_set_caps
0:00:10.240577550 22127 0x55e4527c3d40 INFO                 tcamsrc gsttcamsrc.cpp:1025:gst_tcam_src_set_caps: Requested caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
0:00:10.348379668 22127 0x55e4527c3d40 DEBUG                tcamsrc gsttcamsrc.cpp:1074:gst_tcam_src_set_caps:<tcamsrc0> Frame rate = 1 Hz
0:00:10.348445764 22127 0x55e4527c3d40 INFO                 tcamsrc gsttcamsrc.cpp:1109:gst_tcam_src_set_caps: Start acquisition
0:00:10.555696847 22127 0x55e4527c3d40 INFO                 tcamsrc gsttcamsrc.cpp:1126:gst_tcam_src_set_caps: Successfully set caps to: video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1
`
Do I need to install something else specific for the Aravis backend?

Regards,
Andre

TIS-Edgar, 2020-09-09:
Hi Andre,

The output is what I would expect. Since you are getting a caps description by tcamsrc the camera seems to have been correctly identified. The same pipeline with another 33U works fine on my end, so lets see if we can find the differences.

You could try adding `-v` to gst-launch-1.0 to get more information.
Set the environment variable TCAM_LOG=DEBUG to see what the backend reports.

I also recommend adding `sync=false` to the ximagesink to ensure better pipeline handling.

-Edgar

andreok, 2020-09-09:
Hello Edgar,

According to the TCAM_LOG output the gige-daemon was not running:
`TCAM_LOG=DEBUG gst-launch-1.0 tcamsrc serial=26020406-aravis ! video/x-bayer,width=640,height=480,framerate=1/1 ! capssetter replace=true join=false caps="video/x-raw,format=GRAY8" ! video/x-raw,format=GRAY8,width=640,height=480,framerate=1/1 ! videoconvert ! ximagesink sync=false
2020.09.08T15:02:43:100        <DEBUG> Logger:108: 
The following library versions are used:
	Tcam:	0.12.0_HEAD/2ce3ee84_rev_2777
	Aravis:	0.7_version_cc14855
2020.09.08T15:02:43:105        <ERROR> utils.cpp:305: Could not open file "/var/lock/gige-daemon.lock"
2020.09.08T15:02:43:105        <ERROR> aravis_utils.cpp:474: Unable to connect to gige-daemon. Using internal methods
2020.09.08T15:02:44:237        <ERROR> aravis_utils.cpp:476: Aravis gave us 1
...`
So, I enabled the gige-daemon as described in the readme file, and now the issue stopped.

But, to be able to capture the bayered output as raw mono frames I had to increase the framerate to 2fps, and for higher framerates I'm getting some dropped frames errors:
`TCAM_LOG=DEBUG gst-launch-1.0 -v tcamsrc serial=26020406-aravis ! video/x-bayer,format=rggb,width=640,height=480,framerate=5/1 ! capssetter replace=true join=false caps="video/x-raw,format=GRAY8" ! video/x-raw,format=GRAY8,width=640,height=480,framerate=5/1 ! videoconvert ! ximagesink sync=false
2020.09.08T16:14:39:156        <DEBUG> Logger:108: 
The following library versions are used:
	Tcam:	0.12.0_HEAD/2ce3ee84_rev_2777
	Aravis:	0.7_version_cc14855
...
2020.09.08T16:14:40:223        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BayerRG8 (42474752) 
2020.09.08T16:14:40:223        <ERROR> AravisDevice.cpp:1190: Input format no supported! "10c0059" - BayerRG12p
2020.09.08T16:14:40:321        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BayerRG16 (36314752) 
2020.09.08T16:14:40:417        <DEBUG> AravisDevice.cpp:1219: Adding format desc: Mono8 (30303859) 
2020.09.08T16:14:40:513        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BGR8 (33524742) 
2020.09.08T16:14:40:513        <ERROR> AravisDevice.cpp:1190: Input format no supported! "210003b" - YCbCr422_8
2020.09.08T16:14:40:513        <ERROR> AravisDevice.cpp:1190: Input format no supported! "20c005a" - YCbCr411_8
2020.09.08T16:14:40:513        <ERROR> AravisDevice.cpp:1190: Input format no supported! "20c003c" - YCbCr411_8_CbYYCrYY
2020.09.08T16:14:40:513        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=4294966720,height=4294966992,framerate=5.000000'
2020.09.08T16:14:40:631        <DEBUG> AravisDevice.cpp:205: Returning 71 properties
2020.09.08T16:14:40:631        <DEBUG> PipelineManager.cpp:127: Received 4 formats.
2020.09.08T16:14:40:631        <DEBUG> AravisDevice.cpp:205: Returning 71 properties
2020.09.08T16:14:40:632        <DEBUG> PropertyHandler.cpp:273: Created 71 properties
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
2020.09.08T16:14:47:227        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2020.09.08T16:14:47:233        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=640,height=480,framerate=5.000000'
2020.09.08T16:14:47:336        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RGGB' - 42474752
2020.09.08T16:14:47:337        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RG16' - 36314752
2020.09.08T16:14:47:337        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'Y800' - 30303859
2020.09.08T16:14:47:337        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'BGR24' - 33524742
2020.09.08T16:14:47:337        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=640,height=480,framerate=5.000000'
2020.09.08T16:14:47:439        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:439        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T16:14:47:440        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2020.09.08T16:14:47:440        <INFO> PipelineManager.cpp:486: source ->  sink
2020.09.08T16:14:47:441        <INFO> ImageSink.cpp:51: Pipeline started playing
2020.09.08T16:14:47:516        <INFO> AravisDevice.cpp:636: Starting actual stream...
2020.09.08T16:14:47:535        <WARNING> AravisDevice.cpp:586: Aravis capture thread is running in high priority mode
2020.09.08T16:14:47:542        <DEBUG> ImageSource.cpp:57: PLAYING....
2020.09.08T16:14:47:542        <INFO> PipelineManager.cpp:83: All pipeline elements set to PLAYING.
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0.GstPad:src: caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)5/1
[... cut ...]
2020.09.08T16:14:50:800        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
2020.09.08T16:14:50:826        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
2020.09.08T16:14:53:877        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
2020.09.08T16:14:53:903        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
...`
Should I adjust some packet size or buffer size?

Regards,
Andre

TIS-Edgar, 2020-09-10:
Why do you have to increase the framerate? That should not be a requirement. Are there any error messages?
If you only rarely require an image maybe triggering them might be a good idea.

Adjusting buffers/packets should not be required.
The backend seems to set the correct size.

Does a normal image stream work?

`TCAM_LOG=DEBUG gst-launch-1.0 -v tcamsrc serial=26020406-aravis ! video/x-bayer,format=rggb,width=640,height=480,framerate=5/1 ! bayer2rgb ! videoconvert ! ximagesink sync=false`?

I will try to reproduce this...

andreok, 2020-09-10:
Hello Edgar,

I had to increase the frame rate from 1fps to at least 2fps to be able to get frames displayed or get dropped frames messages. Here's an example call with 1fps, where you can see that it hangs trying to capture a frame:
`TCAM_LOG=DEBUG gst-launch-1.0 -v tcamsrc serial=26020406-aravis ! video/x-bayer,format=rggb,width=640,height=480,framerate=1/1 ! bayer2rgb ! videoconvert ! ximagesink sync=false
2020.09.08T17:21:07:093        <DEBUG> Logger:108: 
The following library versions are used:
	Tcam:	0.12.0_HEAD/2ce3ee84_rev_2777
	Aravis:	0.7_version_cc14855
...
2020.09.08T17:21:08:151        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BayerRG8 (42474752) 
2020.09.08T17:21:08:151        <ERROR> AravisDevice.cpp:1190: Input format no supported! "10c0059" - BayerRG12p
2020.09.08T17:21:08:245        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BayerRG16 (36314752) 
2020.09.08T17:21:08:342        <DEBUG> AravisDevice.cpp:1219: Adding format desc: Mono8 (30303859) 
2020.09.08T17:21:08:439        <DEBUG> AravisDevice.cpp:1219: Adding format desc: BGR8 (33524742) 
2020.09.08T17:21:08:439        <ERROR> AravisDevice.cpp:1190: Input format no supported! "210003b" - YCbCr422_8
2020.09.08T17:21:08:439        <ERROR> AravisDevice.cpp:1190: Input format no supported! "20c005a" - YCbCr411_8
2020.09.08T17:21:08:439        <ERROR> AravisDevice.cpp:1190: Input format no supported! "20c003c" - YCbCr411_8_CbYYCrYY
2020.09.08T17:21:08:439        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=4294966720,height=4294966992,framerate=5.000000'
2020.09.08T17:21:08:559        <DEBUG> AravisDevice.cpp:205: Returning 71 properties
2020.09.08T17:21:08:559        <DEBUG> PipelineManager.cpp:127: Received 4 formats.
2020.09.08T17:21:08:559        <DEBUG> AravisDevice.cpp:205: Returning 71 properties
2020.09.08T17:21:08:559        <DEBUG> PropertyHandler.cpp:273: Created 71 properties
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
2020.09.08T17:21:15:022        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2020.09.08T17:21:15:028        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=640,height=480,framerate=1.000000'
2020.09.08T17:21:15:131        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RGGB' - 42474752
2020.09.08T17:21:15:131        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RG16' - 36314752
2020.09.08T17:21:15:131        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'Y800' - 30303859
2020.09.08T17:21:15:131        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'BGR24' - 33524742
2020.09.08T17:21:15:131        <DEBUG> AravisDevice.cpp:431: Setting format to 'format=RGGB,width=640,height=480,framerate=1.000000'
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:231        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:232        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:232        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:232        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.09.08T17:21:15:232        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2020.09.08T17:21:15:232        <INFO> PipelineManager.cpp:486: source ->  sink
2020.09.08T17:21:15:232        <INFO> ImageSink.cpp:51: Pipeline started playing
2020.09.08T17:21:15:307        <INFO> AravisDevice.cpp:636: Starting actual stream...
2020.09.08T17:21:15:326        <WARNING> AravisDevice.cpp:586: Aravis capture thread is running in high priority mode
2020.09.08T17:21:15:333        <DEBUG> ImageSource.cpp:57: PLAYING....
2020.09.08T17:21:15:333        <INFO> PipelineManager.cpp:83: All pipeline elements set to PLAYING.
/GstPipeline:pipeline0/GstTcamSrc:tcamsrc0.GstPad:src: caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstBayer2RGB:bayer2rgb0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1, format=(string)BGRx
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1, format=(string)BGRx
/GstPipeline:pipeline0/GstXImageSink:ximagesink0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1, format=(string)BGRx
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)1/1, pixel-aspect-ratio=(fraction)1/1, format=(string)BGRx
[... cut ...]
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-bayer, format=(string)rggb, width=(int)640, height=(int)480, framerate=(fraction)5/1, pixel-aspect-ratio=(fraction)1/1
2020.09.08T17:16:27:065        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
2020.09.08T17:16:30:115        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
2020.09.08T17:16:30:160        <WARNING> AravisDevice.cpp:865: The received image did not fit in the buffer data space
...`
As you can see, all the frames were dropped due to problems with the size of the buffer data space.

Regards,
Andre

TIS-Edgar, 2020-09-11:
Hi Andre,

We are unable to reproduce this problem.
Debian Buster, amd64 as a virtualbox machine works fine on our end.

So lets try to narrow it down:
1. Can you uninstall, delete the build dir, compile and try again.
  I would be extremely surprised if a reconfiguration could cause such an issue but better safe than sorry.
2. You can compile aravis separately and try the arv-viewer. If it displays an image tiscamera is the problem.
3. Does the problem occur when you interact with the camera on a native system.
   If this works the virtualization causes the issues.
4. Does the VM work on another machine? Maybe the host is causing problems with passing the camera through. 

If those do not work I will prepare a debug version of tiscamera with additional debug logs so that we can identify this issue.

I am sorry if this whole conundrum causes unnecessary delays for you.

Regards,

Edgar

andreok, 2020-09-15:
Hello Edgar,

Thank you very much for the information.

Here are the results for the tests:
1- Uninstalled tiscamera, removed the build directory, recreated the build directorry and configured it (`cmake -DBUILD_ARAVIS=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DBUILD_LIBUSB=ON -DTCAM_ARAVIS_USB_VISION=ON ..`), built and installed, and reconfigured the gige-daemon: 
Problem persisted
2- Built Aravis from the dependencies/aravis directory with `./configure --enable-viewer` (had to install gtk+-3.0 and libnotify-dev using apt):
Same issue with arv-viewer. I tried also [Aravis 0.6.4 from the Gnome repository](http://ftp.gnome.org/pub/GNOME/sources/aravis/0.6/) but the results were the same.
3- The host machine is a MacBook with USB 2.0 port, so I tried also [Aravis 0.6.4 from the Gnome repository](http://ftp.gnome.org/pub/GNOME/sources/aravis/0.6/), and with arv-viewer worked just fine with any fps setting;
4- Looks like there's an Aravis compatibility issue related to VirtualBox 6.1.12 r139181 when the USB controller is set to USB 2.0 (OHCI + EHCI) controller. After changing it to USB 3.0 (xHCI) controller Aravis and tiscam started to work perfectly with any valid fps.

I could not make tiscam to work with my USB3Vision camera using the Aravis backend without the gige-daemon running. I expected that the gige-daemon would be necessary only for GigE cameras, not also for USB3Vision cameras.
Is it possible to use the Aravis backend for USB3Vision cameras without requiring the configuration of the gige-daemon?

Regards,
Andre

TIS-Edgar, 2020-09-16:
Hi Andre,

Glad to hear it finally works.

At this point in time the gige-daemon is required. The backend was not designed to address a camera through multiple backends,
which is why we currently have to go the additional steps through the daemon.
It is recommended to use the daemon nonetheless as it speeds up device opening.
Without the daemon aravis has to query the network for new devices before returning an up to date device list.
The daemon holds a list that is updated every 2 seconds but available immediately.

I can look into making the gige-daemon optional if you want,
but that is something you should not expect this year to happen.

Take care,

Edgar

closed 2021-02-19 by TIS-Stefan