TheImagingSource/tiscamera #365: runtime dependencies are insufficient for exampels

midjji, 2020-09-07:
so if you just follow the guide and install the tiscamera and then run the tiscamera examples or tools, 
most do not work on ubuntu 2004. 
for example tcam-capture requires pyqt5 etc. 

the gst_launch examples do not work due to requiring bayer2rgb which is probably in one of gstreamer1.0-nice gstreamer1.0-plugins-good gstreamer1.0-qt5  gstreamer1.0-plugins-bad
so figure out which one it is for certain and add to runtime dependencies. 

Also the error you get if trying to use the gst examples is a bunch of 
(ncamera_viewer:1627393): GStreamer-WARNING **: 20:34:06.825: Name 'tcambin-src_caps' is not unique in bin 'tcambin-source', not adding
followed by a stall, so its not trivial to se that missing gst plugins are the problem...

TIS-Edgar, 2020-09-09:
Hi,
I am assuming you are referring to this guide: https://www.theimagingsource.com/documentation/tiscamera/tutorial.html correct?
install-dependencies.sh --runtime should install all potentially needed packages.

`(ncamera_viewer:1627393): GStreamer-WARNING **: 20:34:06.825: Name 'tcambin-src_caps' is not unique in bin 'tcambin-source', not adding`

That is a warning I have not been able to observe on multiple systems. 
Would it be possible to see the entire output?

midjji, 2020-09-11:
Yes I ran that script for runtime, it does not install the bayer2rgb fat
module which is in gstreamer-bad or one of the others not installed by the
script. I found the tip in your issues list, so you have litteraly helped
someone else solve this issue by instructing them to install an extra
dependency, but then not added it to the dependency script before.

The full output is that message repeated 20 or so times. I compiled Aravis
and installed it, then it stopped, but I don't know why. Especially since
it's still in Libusb mode, I think. Both show up in tcam-ctrl -l now, but
not before

TIS-Stefan, 2020-09-14:
Hello

I use Ubuntu 20.04 for some time too and I can compile and run all samples.
According to the GStreamer documentation (https://gstreamer.freedesktop.org/documentation/bayer/bayer2rgb.html) the bayer2rgb should be in GStreamer Bad Plugins. I suggest to install this package manually and check with 
gst-inspect-1.0 bayer2rg
whether it was installed. You should see following:

```
Factory Details:
  Rank                     none (0)
  Long-name                Bayer to RGB decoder for cameras
  Klass                    Filter/Converter/Video
  Description              Converts video/x-bayer to video/x-raw
  Author                   William Brack <wbrack@mmm.com.hk>

Plugin Details:
  Name                     bayer
  Description              Elements to convert Bayer images
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstbayer.so
  Version                  1.16.2
  License                  LGPL
  Source module            gst-plugins-bad
  Source release date      2019-12-03
  Binary package           GStreamer Bad Plugins (Ubuntu)
  Origin URL               https://launchpad.net/distros/ubuntu/+source/gst-plugins-bad1.0

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstBaseTransform
                         +----GstBayer2RGB
```

I have to admit, the " GStreamer-WARNING **: 20:34:06.825: Name 'tcambin-src_caps' is not unique in bin 'tcambin-source', not adding" is new to me. 

Stefan

midjji, 2020-09-14:
Yes, thats right, any particular reason why the script said to install the
runtime dependencies does not install the runtime dependencies?

TIS-Stefan, 2020-09-14:
Well, it seemed, the bayer2rgb has been moved from one plugin to another, when there was a change from 16.04 to 18.04. 
Therefore, at least I am not really where the bayer2rgb is in.
Sorry for that.

midjji, 2020-09-14:
No problem, it was a bugreport after all.

closed 2021-02-19 by TIS-Stefan