TheImagingSource/tiscamera #366: Binning and ROI with DBK 33UX178.AS

andreok, 2020-09-30:
Hello,

I would like to know how I should proceed to enable binning and ROI using the C API and USB3.0 cameras, DBK 33UX178.AS camera in particular, on amd64 Debian Buster.
I tried also using tcam-capture, but setting the Override Scanning Mode parameter to 2 had no apparent effect on the screen, and I only managed to get a ROI region outlined in green on the screen when I adjusted the Auto Functions ROI (custom widthxheight or preset).

Regards,
Andre

TIS-Stefan, 2020-10-06:
Hello

The Override Scanning Mode has different values to be set for each sensor. Also the Override Scanning Mode has no effect, if the pipeline is not restarted. So you may set a value and change the frame rate for restarting the pipeline in tcam-capture. Check the results of the different values you can set. The camera supports Binning 2x only.

Stefan

andreok, 2020-10-08:
Hello Stefan,

Thank you very much for your reply.

Just to give you a bit more context, I'm porting an application from IC Imaging Control on Windows to tiscam on Linux.

Sorry, but I still have so many questions:
1- To programmatically enable 2x2 binning with tiscam, I would need to loop through the list of the Override Scanning Mode parameter values (restarting the pipeline) until I find the one that gives me expected frame size (half the full frame size)?
2- Alternatively, to test it from the command line, I should set the Override Scanning Mode and start the pipeline passing the full resolution to tcambin?
`tcam-ctrl 26020406 -s "Override Scanning Mode=2"
gst-launch-1.0 tcambin serial=26020406 ! video/x-bayer,format=rggb,width=3072,height=2048,framerate=5/1 ! bayer2rgb ! videoconvert ! ximagesink` 

Finally, I tried changing the fps after setting the ROI region, but I could not see any difference in tcam-capture.
To emulate it from the command line, I should do?
`tcam-ctrl 26020406 -s "Auto Functions ROI Preset=4"
tcam-ctrl 26020406 -s "Auto Functions ROI Width=300"
tcam-ctrl 26020406 -s "Auto Functions ROI Height=300"
tcam-ctrl 26020406 -s "Auto Functions ROI Left=0"
tcam-ctrl 26020406 -s "Auto Functions ROI Top=0"
tcam-ctrl 26020406 -s "Auto Functions ROI Control=true"
gst-launch-1.0 tcambin serial=26020406 ! video/x-bayer,format=rggb,width=3072,height=2048,framerate=5/1 ! bayer2rgb ! videoconvert ! ximagesink`

Regards,
Andre

TIS-Stefan, 2020-10-09:
Hi Andre

> 1- To programmatically enable 2x2 binning with tiscam, I would need to loop through the list of the Override Scanning Mode parameter values (restarting the pipeline) until I find the one that gives me expected frame size (half the full frame size)?

Unfortunately, yes. You would need a list containing the camera model and its supported Override Scanning modes.

> 2- Alternatively, to test it from the command line, I should set the Override Scanning Mode and start the pipeline passing the full resolution to tcambin?
> tcam-ctrl 26020406 -s "Override Scanning [...]
If you want to use Binning, then you must use a lower resolution, e.g. max pixels / 2. In your pipeline, you should use 1536x1024 pixel resolution, otherwise the binned resolution is too big for the sensor. 
I made a QT5 sample for testing binning for another camera model. It will work with your camera too.
[dfk-afu420.zip](https://github.com/TheImagingSource/tiscamera/files/5353342/dfk-afu420.zip)
But it might be helpful for you, It simplyfies the Override Scanning testing. I would start with a low resolution for testing, e.g. 640x480, so the binning 2x and if available 4x fits on the sensor. 

> Finally, I tried changing the fps after setting the ROI region, but I could not see any difference in tcam-capture.
> To emulate it from the command line, I should do?
Using the DFK 33UX178 with v4l2 driver, you have only the resolutions offered in the list of video formats by
`tcam-ctrl -c`
Setting a lower resolution makes an ROI on the sensor. The sample above shows, how to move that region around on the sensor. This can be done, while the camera is streaming.
If you want to use any resolution (usually step size 8) and not only a selected list of resolutions, then you must use Aravis/USB3Vision. For this task, you "git checkout development" and rebuild with
```
cd tiscamera/build
sudo make uninstall
rm * -rf
cmake -DBUILD_ARAVIS=ON -DTCAM_ARAVIS_USB_VISION ..
make 
sudo make install
```
Then the "backend" is selected with the serial number:
 - 26020406-aravis for Aravis/USB3Vision
 - 26020406-v4l2 for vl42

If using Aravis/USB3Vision, the camera property set changes. Also only the 33U, 37U and 38U camera series supports USB3Vision.

Stefan

andreok, 2020-10-13:
Hello Stefan,

Thank you very much for the information, it's now clear to me why I was getting the results I was getting.

Regarding backward compatibility with older USB2.0 cameras, like your dfk72auc02, that have the "Binning" property and doesn't have a half full size resolution, I should set Binning=2 and keep the resolution at maximum?

Regards,
Andre

TIS-Stefan, 2020-10-14:
Hi Andre

if you want to use Binning, you must always set a half resolution video format.

Stefan

closed 2021-02-19 by TIS-Stefan