TheImagingSource/tiscamera #367: Tone Mapping in python code in Ubuntu

Moj-Dev, 2020-10-02:
The tonemapping option is available in GUI, but there is no property for enabling this feature in SDK. How can I enable it in my python code?
I'm using Tis.Set_Property("", ) for other properties. Can I use this function to enable the tonemapping?

TIS-Stefan, 2020-10-06:
Hello

Tone Mapping is available, after the pipeline is in state PLAYING. So start the stream first, then you can set Tonemapping. 
There is also a "listPropertiers" function in TIS class. You may call this function before and after you started the pipeline in order to see the differences.

Stefan

Moj-Dev, 2020-10-06:
Worked for me when I called the List_Properties() in PLAYING state. 
I ran this code based on [example code](https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/blob/master/Save%20an%20image%20on%20trigger%20in%20Python/Program.py) . How can I set properties when I use gstreamer. Actually, I mean the [08-save-stream.py ](https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/08-save-stream.py).

TIS-Stefan, 2021-02-19:
You can do that only, after the state is set to PLAYING. It is kind of design flaw of our GStreamer modules. The additional modules for properties done in software are initialized correctly after the first frame went through them.

closed 2021-02-19 by TIS-Stefan