TheImagingSource/tiscamera #368: Blue/Purple Dots in the image

Moj-Dev, 2020-10-03:
I'm using the DFM 37UX226-ML camera. For image capturing, I used "IC Capture 2.5" in Windows and "tcam-capture" in Ubuntu 18.04, but images are so noisy, and there are some dots in the image. What causes this noise and how we can solve it?
[image]

TIS-Stefan, 2020-10-06:
Hello

First of all, reduce gain. Gain amplifies hotpixels.
Do you see them in Windows and Linux?

Stefan

Moj-Dev, 2020-10-06:
I set the gain to zero and also exposure to 24000-66000. It works well in full resolution 15fps, but I have dots in dark areas when the fps is 30 and also when the exposure is auto. 
How can I fix this issue?

TIS-Stefan, 2020-10-07:
Hello

You mean, you have the dots only at 30 fps, not at 15? I must admit, that is new to me. However, I think, that is an issue of hot pixels on the sensor. We do not have a hot pixel fix for Linux, as we have it in Windows.
You may need to save the hot pixel coordinates in a list and interpolate them by using the neighbors.

I must admit, I am somewhat puzzled.

Moj-Dev, 2020-10-08:
Hey
Using the T-cam Capture in Ubuntu, I set the format in rggb16 and everything is fine. But there are two points:
1- I can't set this format in my code:
pipeline = Gst.parse_launch("tcambin name=bin"
                                " ! video/x-raw,format=BGRx,width=4000,height=3000,framerate=15/1"
                                " ! tee name=t"
                                " ! queue"
                                " ! videoconvert"
                                " ! ximagesink"
                                " t."
                                " ! queue"
                                " ! videoconvert"
                                " ! avimux"
                                " ! filesink name=fsink")
When I change the format, I get error.

2. I need 30fps capturing at 4000x3000 and 15fps is low for me and causes motion blurry. So any idea to get images in 30fps and full resolution without hot pixels (those dots)?

TIS-Stefan, 2020-10-09:
Unfortunately, I can not test that on my own, because I am in home office without that camera model.
But motion blur is created by exposure times. There should be a property "Exposure max", which allows to limit the maximum used exposure time. If you lower that property value, the the motion blur will be much less, regardless, which frame rate is set. 
However, the rolling shutter effect will stay.

Next Monday I will be back in the office and can check the x226 camera model in Linux. You may trigger me on Monday!

Stefan

Moj-Dev, 2020-10-09:
I'm using auto exposure, here is my config for exposure:
    camera.set_tcam_property("Exposure Auto", True)
    camera.set_tcam_property("Exposure Auto Upper Limit Auto", True)
Thanks a lot for your support.

TIS-Stefan, 2020-10-12:
Hello

I can see something similar extremely short, if i use a 16 bit video format and have a fast change from bright to dark light. But then these dots are away.

Can you try to show all camera properties? Maybe with tcam-capture screenshots? This will be helpful.

Stefan

TIS-Stefan, 2020-10-13:
Hello

Today I tried to reproduce the issue, but had no success. The dots are missing on my camera
In Windows I use RGB64 with 20 fps. 
In Linux I use rggb with 15 fps.  So far, that is, as you write. 
30 fps are available only, if I use an 8 bit pixel format. I did so on my Linux computer. I was not able to see the dots

The line

`camera.set_tcam_property("Exposure Auto Upper Limit Auto", True)
`
Allows exposure times up to 1/15 second, if you set this to true. Therefore, set it to False and and set the property "Exposure Auto Upper Limit" to e.g. 200000. Then you should have less motion blurr.

Moj-Dev, 2020-10-15:
@TIS-Stefan  tnx for your help. 
I attached some screenshots from my configuration in windows. You can see the fps and also dots. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TIS-Stefan, 2020-10-16:
Hello

We would like to get the camera, because it shows the issue nicely. 
Please contact your camera dealer for sending the camera in. 

Stefan

Moj-Dev, 2020-10-16:
Thank you so much, Stefan. So this is a hardware issue?

TIS-Stefan, 2020-10-19:
Hello

We think so, because we have no cameras in our lab, that show the same issue. Yours seems to be a nice specimen for checking, what is going on.

Stefan

closed 2021-02-19 by TIS-Stefan