TheImagingSource/tiscamera #371: video/x-bayer rggb16 format in gstreamer pipeline

Moj-Dev, 2020-10-08:
I'm using the gstreamer to save the video file, here is my pipeline config:
pipeline = Gst.parse_launch("tcambin name=bin"
                                " ! video/x-raw,format=BGRx,width=4000,height=3000,framerate=15/1"
                                " ! tee name=t"
                                " ! queue"
                                " ! videoconvert"
                                " ! ximagesink"
                                " t."
                                " ! queue"
                                " ! videoconvert"
                                " ! avimux"
                                " ! filesink name=fsink")

When I change the format to " !video/x-bayer,format=rggb16, ..." or "! video/x-bayer,format=rggb , ..."I got an error:
GLib.Error: gst_parse_error: could not link bin to videoconvert0, videoconvert0 can't handle caps video/x-bayer, format=(string)rggb16, width=(int)3840, height=(int)2160, framerate=(fraction)30/1 (3)

While I have them in the list formats :
video/x-bayer rggb - 4000x3000 - 30/1 15/1 5/1 1/1 
video/x-bayer rggb - 3840x2160 - 30/1 15/1 5/1 1/1 
video/x-bayer rggb - 1920x1080 - 30/1 15/1 5/1 1/1 
video/x-bayer rggb - 640x480 - 30/1 15/1 5/1 1/1 
video/x-bayer rggb16 - 4000x3000 - 15/1 5/1 1/1 
video/x-bayer rggb16 - 3840x2160 - 15/1 5/1 1/1 
video/x-bayer rggb16 - 1920x1080 - 15/1 5/1 1/1 
video/x-bayer rggb16 - 640x480 - 15/1 5/1 1/1 
video/x-raw {  RGBx  xRGB  BGRx  xBGR  RGBA  ARGB  BGRA  ABGR  } - 4000x3000 - 30/1 15/1 5/1 1/1 
video/x-raw {  RGBx  xRGB  BGRx  xBGR  RGBA  ARGB  BGRA  ABGR  } - 3840x2160 - 30/1 15/1 5/1 1/1 
video/x-raw {  RGBx  xRGB  BGRx  xBGR  RGBA  ARGB  BGRA  ABGR  } - 1920x1080 - 30/1 15/1 5/1 1/1 
video/x-raw {  RGBx  xRGB  BGRx  xBGR  RGBA  ARGB  BGRA  ABGR  } - 640x480 - 30/1 15/1 5/1 1/1 

Any thoughts on this? How can I save video in rggb16 format?

TIS-Stefan, 2020-10-09:
Hello

Using 16bit per channel video formats needs all other modules to support them too. However, most of them do not support that. 
If you want to use 16 bits, you have two solutions
1.) You may use the tiscamera-dutils from https://www.theimagingsource.com/support/downloads-for-linux/
2.) You may use pipelines like the ones shown at https://theimagingsource.deskpro.com/en/kb/articles/linux-set-video-formats-in-camera-and-memory

If you use gst-inpect-1.0 videoconvert, you will see, that no 16 bit formats are supported. Thus saving 16 bit avis will work with raw data only, as in Windows.

Moj-Dev, 2020-10-16:
Also, rggb format doesn't work for me. I have this format in available gstreamer caps. Here is the result of tcam-ctrl -c 06020009 command:
Available gstreamer-1.0 caps:
video/x-bayer,format=rggb,width=4000,height=3000,framerate={30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=3840,height=2160,framerate={30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=1920,height=1080,framerate={30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb,width=640,height=480,framerate={30/1,15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=4000,height=3000,framerate={15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=3840,height=2160,framerate={15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=1920,height=1080,framerate={15/1,5/1,1/1};
video/x-bayer,format=rggb16,width=640,height=480,framerate={15/1,5/1,1/1}

But I can't launch gstreamer pipeline and got the following error:
    pipeline = Gst.parse_launch("tcambin name=bin"
gi.repository.GLib.Error: gst_parse_error: could not link queue0 to videoconvert0 (3)

How can I use x-bayer,rggb in my code?

TIS-Stefan, 2020-10-19:
Hello

Did you try the link, I showed in my post? It shows, how to get raw data.

Stefan

closed 2021-02-19 by TIS-Stefan