TheImagingSource/tiscamera #372: tcambin video FPS is very slow

devfox-se, 2020-10-14:
Hi,

I have SDI-to-USB 3.0 converter connected to my board and I am trying to do post processing on the video using OpenCV/Python. To communicate with the device I have built and installed tiscamera tcambin executable but the video output FPS is very low ~12 FPS while it should be capable of outputting ~30 FPS.

I have created a custom pipeline without tcambin to see if I can get higher framerate and I found out that the problem is in tcambin itself

So for the following GStreamer pipeline I get ~12 FPS
`gst-launch-1.0 tcambin serial=17710405 ! 'video/x-raw,format=YUY2,width=1920,height=1080,framerate=30/1' ! videoconvert ! fpsdisplaysink video-sink=ximagesink`

and for raw GStreamer pipeline the output is ~30 FPS
`gst-launch-1.0 v4l2src device=/dev/video0 ! 'video/x-raw,format=YUY2,width=1920,height=1080,framerate=30/1' ! videoconvert ! fpsdisplaysink video-sink=ximagesink`

Yet when I try to use my custom pipeline for the OpenCV to capture the video, the framerate drops again.
Is there a way to make tiscamera output solid 30 FPS so that I can use the provided TIS python class?

TIS-Stefan, 2020-10-14:
Hello

On my Linux computer I do not get images with 30 fps YUY2 at all. Neither with v4l2, nor with tcambin.
Therefore, I can not test. I will ask a colleague.

Stefan

devfox-se, 2020-10-14:
Hi @TIS-Stefan 

I must mention that I my unit is Nvidia Jetson NX arm64

TIS-Stefan, 2020-10-15:
Hello 

I spoke with the programmers. They told me, the support of DFG/HSDI is rudimentary.  E.g. there are no properties for the DFG/HDSDI.
Therefore, you can stay with the v4l2src. If you want to stay with the tis.py, you can exchange the tcambin with v4l2src. That should work.

I am sorry for not having a better answer.

Stefan

closed 2021-02-19 by TIS-Stefan