TheImagingSource/tiscamera #374: 'Device Lost' Error Persists Reboots

cSDes1gn, 2020-10-20:
I have a TIS module (DMM 27UJ003-ML) using V4L2 hooked up via USB3.0 to an RPi. I am creating and tearing down a number of different gst pipelines for different streaming requirements. 

I am experiencing a number of 'device lost' errors. These errors persist through system reboots and system power cycling. The only way I have been able to resolve this issue is by unplugging and replugging the USB3.0 cable. This leads me to believe it is some kind of firmware setting that is being reset upon a new connection. I have checked all the tcam properties using the python GST library and through the cli using `tcam-ctrl` and have not found any deviations. I need a way of reconnecting (through software) to the module without hotplugging the USB cable.

Any guidance would be appreciated.

TIS-Stefan, 2020-10-21:
Hello

Which PI version do you have? The camera needs an USB 3.0 connection, it wont run on USB 2.0.

Stefan

cSDes1gn, 2020-10-21:
RPi4 via USB3.0

TIS-Stefan, 2020-10-22:
Hello 

Device Lost is something physical and has no relation to firmware.
Did you try another USB cable?
A reboot does not unpower the camera nor it changes anything on the cable. What happens, if you unpower the PI4 instead?

Stefa

cSDes1gn, 2020-10-22:
> Device Lost is something physical and has no relation to firmware.

That's helpful information. I assumed there was some setting mismatch between the firmware and the Pi's gst settings and so the connection could not be re-established.

> What happens, if you unpower the PI4 instead?

Even if I power cycle the Pi the connection is not always guaranteed to be re-established with the module. Someone else who has wrestled with this issue has mentioned that he suspects the USB cable is at fault. All the cables in question are USB3.0 compatible with the blue insert. We have a single cable that has never given us problems but it is too long for our use case. We opted to replace it with a shorter cable which has given us mixed results regarding the device lost issue. 

Upon deeper investigation of the USB cables, the cables in question are type UL20276 and the working cable is UL2725. Im not sure what these types mean and if that would be something to look deeper into.

My intuition tells me if a cable is able to stream video it should be able to do so unconditionally. The fact that these cables jump in and out concerns me. They are also brand new cables and so are not physically damaged.

TIS-Stefan, 2020-10-23:
USB cables have different quality. The soldering to the connectors and the shieldings can be different. I case you can test a different cable, I would like you to do so.

Stefan

cSDes1gn, 2020-10-23:
Apparently, the working UL2725 cable came with the TIS module. I will try another UL2725 just to verify and let you know.

closed 2021-02-19 by TIS-Stefan