TheImagingSource/tiscamera #375: error building/cmake?

UCN0, 2020-10-25:
Hello. I am trying to build tiscamera, I have cloned and installed the dependencies except libzip4, I have libzip5. I am using ubuntu-20.10 arm64 on raspi4 and this is the error I get

 root@machine:/home/machine/Documents/tiscamera/build# cmake -DBUILD_ARAVIS=OFF ..
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found GObject: /usr/include/glib-2.0  
-- Found PYTHON_MODULE_PyQt5: /usr/lib/python3/dist-packages/PyQt5  
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   OFF
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            OFF
-- Build tests                    OFF
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /etc/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing data files to:      /usr/share/theimagingsource/tiscamera
-- Installing gstreamer-1.0 to:   /usr/lib/aarch64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/aarch64-linux-gnu/girepository-1.0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/machine/Documents/tiscamera/build
root@machine:/home/machine/Documents/tiscamera/build# make -j4
make[1]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
Scanning dependencies of target pugixml_obj
Scanning dependencies of target tcam-dfk73
Scanning dependencies of target generate-version-file
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
Scanning dependencies of target 33update
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
[  1%] Building CXX object dependencies/PugiXml/CMakeFiles/pugixml_obj.dir/pugixml.cpp.o
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
[  2%] Built target generate-version-file
[  2%] Building C object src/v4l2/CMakeFiles/tcam-dfk73.dir/dfk73.c.o
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
Scanning dependencies of target tcam-base
[  2%] Building CXX object tools/firmware-update/33u/CMakeFiles/33update.dir/Archive.cpp.o
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
[... 53 lines cut ...]
[ 21%] Building CXX object tools/firmware-update/33u/CMakeFiles/33update.dir/GenCPFacade.cpp.o
[ 22%] Building CXX object tools/firmware-update/33u/CMakeFiles/33update.dir/MachXO3.cpp.o
[ 23%] Building CXX object tools/firmware-update/33u/CMakeFiles/33update.dir/JedecFile.cpp.o
[ 24%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zAlloc.c.o
[ 24%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zCrc.c.o
[ 25%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zFile.c.o
[ 26%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Bra86.c.o
[ 27%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/CpuArch.c.o
[ 27%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/LzmaDec.c.o
[ 28%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zArcIn.c.o
[ 29%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zCrcOpt.c.o
[ 30%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zStream.c.o
[ 30%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Bra.c.o
[ 31%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Delta.c.o
[ 32%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Ppmd7.c.o
[ 33%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zBuf.c.o
[ 33%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/7zDec.c.o
[ 34%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Bcj2.c.o
[ 35%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/BraIA64.c.o
[ 36%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Lzma2Dec.c.o
[ 36%] Building C object tools/firmware-update/33u/CMakeFiles/33update.dir/lib/7z/Ppmd7Dec.c.o
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
[ 36%] Built target 33update
make[1]: Leaving directory '/home/machine/Documents/tiscamera/build'
make: *** [Makefile:152: all] Error 2

TIS-Edgar, 2020-10-26:
Fixed in 939b27a.

Please pull the master branch an try building to verify this issue is fixed.

UCN0, 2020-10-26:
Thanks, but now it makes up to 74%:

``
root@machine:/home/machine/Documents/tiscamera/build# cmake -DBUILD_ARAVIS=ON ..
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found GObject: /usr/include/glib-2.0  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Found PYTHON_MODULE_PyQt5: /usr/lib/python3/dist-packages/PyQt5  
-- Found PYTHON_MODULE_setuptools: /usr/lib/python3/dist-packages/setuptools  
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   ON
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            OFF
-- Build tests                    OFF
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /etc/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing data files to:      /usr/share/theimagingsource/tiscamera
-- Installing gstreamer-1.0 to:   /usr/lib/aarch64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/aarch64-linux-gnu/girepository-1.0
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/machine/Documents/tiscamera/build
root@machine:/home/machine/Documents/tiscamera/build# make -j4
make[1]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
Scanning dependencies of target aravis
Scanning dependencies of target tcam-dfk73
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
Scanning dependencies of target generate-version-file
Scanning dependencies of target pugixml_obj
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[2]: Entering directory '/home/machine/Documents/tiscamera/build'
[  0%] Creating directories for 'aravis'
[  1%] Building C object src/v4l2/CMakeFiles/tcam-dfk73.dir/dfk73.c.o
[  2%] Building CXX object dependencies/PugiXml/CMakeFiles/pugixml_obj.dir/pugixml.cpp.o
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
[  2%] Built target generate-version-file
[... 714 lines cut ...]
[ 69%] Building CXX object src/CMakeFiles/tcam.dir/PipelineManager.cpp.o
[ 69%] Building CXX object src/CMakeFiles/tcam.dir/ImageSource.cpp.o
[ 70%] Building CXX object src/CMakeFiles/tcam.dir/PropertyHandler.cpp.o
[ 71%] Building CXX object src/CMakeFiles/tcam.dir/public_utils.cpp.o
[ 71%] Building CXX object src/CMakeFiles/tcam.dir/v4l2/V4l2Device.cpp.o
[ 72%] Building CXX object src/CMakeFiles/tcam.dir/v4l2/V4L2PropertyHandler.cpp.o
[ 72%] Building CXX object src/CMakeFiles/tcam.dir/v4l2/v4l2_utils.cpp.o
[ 73%] Building CXX object src/CMakeFiles/tcam.dir/v4l2/v4l2_api.cpp.o
[ 74%] Building CXX object src/CMakeFiles/tcam.dir/aravis/AravisDevice.cpp.o
[ 74%] Building CXX object src/CMakeFiles/tcam.dir/aravis/aravis_utils.cpp.o
/home/machine/Documents/tiscamera/src/aravis/AravisDevice.cpp: In constructor ‘tcam::AravisDevice::AravisDevice(const tcam::DeviceInfo&)’:
/home/machine/Documents/tiscamera/src/aravis/AravisDevice.cpp:153:20: error: ‘runtime_error’ is not a member of ‘std’
  153 |         throw std::runtime_error("Error while creating ArvCamera");
      |                    ^~~~~~~~~~~~~
/home/machine/Documents/tiscamera/src/aravis/AravisDevice.cpp: In member function ‘void tcam::AravisDevice::determine_packet_request_ratio()’:
/home/machine/Documents/tiscamera/src/aravis/AravisDevice.cpp:669:13: warning: unused variable ‘err’ [-Wunused-variable]
  669 |     GError* err = nullptr;
      |             ^~~
make[2]: *** [src/CMakeFiles/tcam.dir/build.make:261: src/CMakeFiles/tcam.dir/aravis/AravisDevice.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/machine/Documents/tiscamera/build'
make[1]: *** [CMakeFiles/Makefile2:694: src/CMakeFiles/tcam.dir/all] Error 2
make[1]: Leaving directory '/home/machine/Documents/tiscamera/build'
make: *** [Makefile:152: all] Error 2
``

TIS-Edgar, 2020-10-26:
Overlooked something. Compiled with all relevant compile flags. Pull and try please.

UCN0, 2020-10-26:
Perhaps this is a bug on ubuntu 20.10 arm raspiberry4. I compile the same code without errors on Jetson nano with a custom ubuntu 20.04 OS. I will try again

UCN0, 2020-10-26:
Thanks. This time worked:
``
root@ath:/home/ath/Documents/tiscamera/build# cmake -DBUILD_ARAVIS=ON ..
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found GObject: /usr/include/glib-2.0  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Found PYTHON_MODULE_PyQt5: /usr/lib/python3/dist-packages/PyQt5  
-- Found PYTHON_MODULE_setuptools: /usr/lib/python3/dist-packages/setuptools  
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   ON
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            OFF
-- Build tests                    OFF
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /etc/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing data files to:      /usr/share/theimagingsource/tiscamera
-- Installing gstreamer-1.0 to:   /usr/lib/aarch64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/aarch64-linux-gnu/girepository-1.0
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ath/Documents/tiscamera/build
root@ath:/home/ath/Documents/tiscamera/build# make -j4
make[1]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
Scanning dependencies of target aravis
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
Scanning dependencies of target generate-version-file
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
Scanning dependencies of target pugixml_obj
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
[  0%] Creating directories for 'aravis'
Scanning dependencies of target tcam-dfk73
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[  0%] Built target generate-version-file
[  1%] Building CXX object dependencies/PugiXml/CMakeFiles/pugixml_obj.dir/pugixml.cpp.o
[  2%] Building C object src/v4l2/CMakeFiles/tcam-dfk73.dir/dfk73.c.o
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
[... 848 lines cut ...]
[ 95%] Building CXX object src/gstreamer-1.0/CMakeFiles/gsttcamwhitebalance.dir/gsttcamwhitebalance.cpp.o
[ 96%] Linking CXX shared library libgsttcambiteater.so
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[ 96%] Built target gsttcambiteater
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
Scanning dependencies of target tcam-ctrl
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
make[2]: Entering directory '/home/ath/Documents/tiscamera/build'
[ 97%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/main.cpp.o
[ 98%] Linking CXX shared library libgsttcambin.so
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[ 98%] Built target gsttcambin
[ 98%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/properties.cpp.o
[ 98%] Linking CXX executable tcam-uvc-extension-loader
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[ 98%] Built target tcam-uvc-extension-loader
[ 99%] Building CXX object tools/tcam-ctrl/CMakeFiles/tcam-ctrl.dir/formats.cpp.o
[100%] Linking CXX shared library libgsttcamwhitebalance.so
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[100%] Built target gsttcamwhitebalance
[100%] Linking CXX executable tcam-ctrl
make[2]: Leaving directory '/home/ath/Documents/tiscamera/build'
[100%] Built target tcam-ctrl
make[1]: Leaving directory '/home/ath/Documents/tiscamera/build'
``

closed 2020-10-26 by UCN0