TheImagingSource/tiscamera #376: tiscamera-dutils for Rasbpery Pi 4 32bit

Moj-Dev, 2020-10-30:
I don't have Tone Mapping and Sharpness on Raspbian 32-bit. I think the  tiscamera-dutils should be installed to get these properties, but there is no package for ARM 32 architecture. I could find only 64 bit version. Am I right? Do you have any thoughts on this?

TIS-Stefan, 2020-11-02:
These properties are avaialble with the library from https://www.theimagingsource.com/support/downloads-for-linux/install/tiscameraarm6432/. They can be seen, **after** the pipeline has been started (GST_STATE_PLAYING) . tcam-ctrl wont show them.

Moj-Dev, 2020-11-03:
Hi,
The tiscamera is already the newest version (0.12.0). These are the available properties after pipeline.set_state(Gst.State.PLAYING):

['Brightness', 'Whitebalance Auto', 'Whitebalance Red', 'Whitebalance Blue', 'Gain', 'Exposure Time (us)', 'Exposure Auto', 'Exposure Auto Reference', 'Gain Auto', 'Trigger Mode', 'Software Trigger', 'Trigger Delay (us)', 'Strobe Enable', 'Strobe Polarity', 'Strobe Exposure', 'Strobe Duration', 'Strobe Delay', 'GPOut', 'GPIn', 'Offset X', 'Offset Y', 'Offset Auto Center', 'Trigger Polarity', 'Trigger Operation', 'Auto Functions ROI Control', 'Whitebalance', 'Whitebalance Auto Preset', 'Whitebalance Green', 'White Balance Temperature Preset', 'Whitebalance Temperature', 'Highlight Reduction', 'Exposure Auto Upper Limit Auto', 'Exposure Auto Lower Limit', 'Exposure Auto Upper Limit', 'Override Scanning Mode', 'Auto Functions ROI Preset', 'Gain Auto Lower Limit', 'Gain Auto Upper Limit', 'Trigger Global Reset Release']

While with exactly the same code I got these following properties on ubuntu18.04 64-bit that tiscamera-dutils is installed:
['Hue', 'Saturation', '**Sharpness**', 'Denoise', 'Contrast', 'Gamma', '**Tonemapping**', 'Tonemapping Intensity', 'Tonemapping Global Brightness', 'Brightness', 'Whitebalance Auto', 'Whitebalance Red', 'Whitebalance Blue', 'Gain', 'Exposure Time (us)', 'Exposure Auto', 'Exposure Auto Reference', 'Gain Auto', 'Trigger Mode', 'Software Trigger', 'Trigger Delay (us)', 'Strobe Enable', 'Strobe Polarity', 'Strobe Exposure', 'Strobe Duration', 'Strobe Delay', 'GPOut', 'GPIn', 'Offset X', 'Offset Y', 'Offset Auto Center', 'Trigger Polarity', 'Trigger Operation', 'Auto Functions ROI Control', 'Whitebalance', 'Whitebalance Auto Preset', 'Whitebalance Green', 'White Balance Temperature Preset', 'Whitebalance Temperature', 'Highlight Reduction', 'Exposure Auto Upper Limit Auto', 'Exposure Auto Lower Limit', 'Exposure Auto Upper Limit', 'Override Scanning Mode', 'Auto Functions ROI Preset', 'Gain Auto Lower Limit', 'Gain Auto Upper Limit', 'Trigger Global Reset Release']

I believe that we should install tiscamera-dutils to set "Tonemapping" and "Sharpness" and the problem is that you don't have iscamera-dutils for ARM 32-bit.

[Here is my code.](https://github.com/Moj-Dev/TIS-CAM/blob/main/SaveStreamV2.py)

TIS-Stefan, 2020-11-04:
You are right, there is no tiscamera_dutils for 32 bit. Maybe in the 0.13.0 tiscamera release, we will have a 32 bit version.

So you may try a 64 bit Linux on the PI?

I am sorry for pointing the wrong direction, but with complexity and amount of possible hardware, OS and software combinations, support becomes more complicated I simply do not know all anymore. 

Stefan

TIS-Stefan, 2021-02-19:
We released the tiscamera-dutils for Raspbian 32 bit at https://www.theimagingsource.com/support/downloads-for-linux/install/tiscameradutilsarmhfpi/

closed 2021-02-19 by TIS-Stefan