TheImagingSource/tiscamera #378: Linux 5.9.5 on arm32: Device did not provide ANY valid caps. Refusing playback.

g0hl1n, 2020-11-06:
Hi,
after updating my system to Linux 5.9.5, GLIBC_2.30 and tiscamera 0.12.0 tcamsrc is unable to get any caps from my camera and therefore fails:

<pre>root@host:~# GST_DEBUG=tcamsrc:9 gst-launch-1.0 tcamsrc serial=&quot;08810319&quot; type=&quot;v4l2&quot; ! fakesink
0:00:00.172613333  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1769:gst_tcam_src_init: Versions:
        Tcam:   0.12.0_master/2ce3ee84_rev_2777
        Aravis: not_configured_version_not_configured
0:00:00.173013667  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1824:gst_tcam_src_set_property: Set camera name to 08810319
0:00:00.173197000  4585  0x1bc4c20 DEBUG                tcamsrc gsttcamsrc.cpp:1250:gst_tcam_src_init_camera:&lt;tcamsrc0&gt; Initializing device.
0:00:00.173283333  4585  0x1bc4c20 DEBUG                tcamsrc gsttcamsrc.cpp:1266:gst_tcam_src_init_camera: Opening device. Serial: &apos;08810319 Type: &apos;unknown&apos;
0:00:00.213954333  4585  0x1bc4c20 DEBUG                tcamsrc gsttcamsrc.cpp:764:gst_tcam_src_get_all_camera_caps: Found 0 pixel formats
0:00:00.214065333  4585  0x1bc4c20 <b>ERROR  </b>              tcamsrc gsttcamsrc.cpp:770:gst_tcam_src_get_all_camera_caps: Device did not provide ANY valid caps. Refusing playback.
0:00:00.214180667  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:774:gst_tcam_src_get_all_camera_caps: Device provides the following caps: EMPTY
0:00:00.214341333  4585  0x1bc4c20 DEBUG                tcamsrc gsttcamsrc.cpp:1859:gst_tcam_src_set_property: Setting device type to v4l2
0:00:00.221217333  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1000:gst_tcam_src_get_caps: Available caps = EMPTY
0:00:00.221402667  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1000:gst_tcam_src_get_caps: Available caps = EMPTY
0:00:00.221713000  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1000:gst_tcam_src_get_caps: Available caps = EMPTY
0:00:00.222046667  4585  0x1bc4c20 INFO                 tcamsrc gsttcamsrc.cpp:1000:gst_tcam_src_get_caps: Available caps = EMPTY
WARNING: erroneous pipeline: could not link tcamsrc0 to fakesink0
root@host:~# uname -a
Linux host 5.9.5-1 #1 SMP PREEMPT Thu Nov 5 22:27:38 UTC 2020 armv7l armv7l armv7l GNU/Linux
root@host:~# dmesg | tail
[   97.940945] usb 1-1.4: new high-speed USB device number 4 using ci_hdrc
[   99.633323] usb 1-1.4: New USB device found, idVendor=199e, idProduct=8307, bcdDevice= 8.13
[   99.633349] usb 1-1.4: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[   99.633364] usb 1-1.4: Product: DFM 5U2-SKIDATA-2.9MM
[   99.633377] usb 1-1.4: Manufacturer: SKIDATA
[   99.633390] usb 1-1.4: SerialNumber: 08810319
[   99.634974] uvcvideo: Unknown video format 4c545250-0000-0010-8000-00aa00389b71
[   99.635021] uvcvideo: Found UVC 1.03 device DFM 5U2-SKIDATA-2.9MM (199e:8307)
[  100.144136] uvcvideo: Failed to query (GET_INFO) UVC control 4 on unit 1: -110 (exp. 1).
[  100.651919] input: DFM 5U2-SKIDATA-2.9MM as /devices/platform/soc/2100000.bus/2184200.usb/ci_hdrc.0/usb1/1-1/1-1.4/1-1.4:1.0/input/input3
</pre>

If more information is needed please come back to me!

Any help is greatly appreciated.

regards;rl

g0hl1n, 2020-11-23:
Hi,
do you have any information/news on this problem?
Or is there anything I should try?

TIS-Stefan, 2020-11-23:
Hallo

I missed your issue somehow, I am sorry. In between we updated tiscamera to 0.13.0 You may try that.  The save way of rebuilding is
```
cd tiscamera/build
gige-daemon stop
sudo make uninstall
git pull
rm * -rf
cmake -DBUILD_ARAVIS=OFF ..
make -j 32
sudo make install
```
Please come back to me, if the error stays.

Stefan

g0hl1n, 2020-11-24:
Hi @TIS-Stefan,
using tiscamera v0.13.0 on a Linux 5.9.10 works again!

Thank you!

closed 2020-11-24 by g0hl1n