TheImagingSource/tiscamera #380: Device Lost - device descriptor/read8, error -110

freelancer1845, 2020-11-16:
Hey,

we have a DFK 33UX183.

Connected to a system running:
```
Ubuntu 18.04
4.15.0-123-generic
```

I can successfully create GStreamer pipelines etc. but at some point the camera usually fails and can only be reconnected by unplugging.

dmesg reports the following errors

```
[156890.561990] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156895.809921] usb 2-6: device descriptor read/8, error -110
[156895.917904] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156901.185778] usb 2-6: device descriptor read/8, error -110
[156901.518074] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156906.561907] usb 2-6: device descriptor read/8, error -110
[156906.669874] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156911.937717] usb 2-6: device descriptor read/8, error -110
[156912.270065] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156917.313704] usb 2-6: device descriptor read/8, error -110
[156917.421864] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156922.689968] usb 2-6: device descriptor read/8, error -110
[156923.021894] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156928.065949] usb 2-6: device descriptor read/8, error -110
[156928.173857] usb 2-6: reset SuperSpeed USB device number 2 using xhci_hcd
[156933.441953] usb 2-6: device descriptor read/8, error -110
[156933.633787] usb 2-6: USB disconnect, device number 2
```

and the camera can not be found by tcam-ctrl anymore.

My pipeline looks like this (python):
```
"tcambin name=bin ! video/x-raw,format=YUY2,width=<width>,height=<height>,framerate=<framerate>"
                                    " ! videoconvert"
                                    " ! videoscale ! video/x-raw,width=1920,height=1080"
                                    " ! videoconvert"
                                    " ! x264enc tune=zerolatency"
                                    " ! rtph264pay"
                                    " ! udpsink host=localhost port=5000"
```

Any suggestions?

I'm on 68dbbc9112c2dafcb37c6c8b324cb930776eb67c

TIS-Stefan, 2020-11-16:
I am very sorry, but I have no idea, why your USB controller is reset in between. 
I say something similar some years ago, but never found a reason nor a solution for hat.

Which computer model do you use?

freelancer1845, 2020-11-17:
The camera is connected to an HP Z4 G4 Workstation.
<details>

<summary>This is the output from lsusb -v of the camera</summary>

```
Bus 002 Device 002: ID 199e:9094 The Imaging Source Europe GmbH
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x199e The Imaging Source Europe GmbH
  idProduct          0x9094
  bcdDevice            0.00
  iManufacturer           1 The Imaging Source Europe GmbH
  iProduct                2 DFK 33UX183
  iSerial                 3 44020271
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         1988
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0
      iFunction               4 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0
      iInterface              0
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 06 01 02 00 07 08 03 05 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
[... 978 lines cut ...]
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 6: Dev 2, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 2, If 2, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 6: Dev 2, If 0, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 6: Dev 2, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 6: Dev 2, If 1, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 7: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 8: Dev 4, If 0, Class=Hub, Driver=hub/2p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
    |__ Port 11: Dev 13, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 13: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 13: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

I also get those errors during streaming at high resolution + high framerate

```
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: warning [initialize_device] Unknown device speed: 10000 Mbps
libusb: error [udev_hotplug_event] ignoring udev action bind
libusb: error [udev_hotplug_event] ignoring udev action bind
```

I updated to Kernel 5.4.0-54-generic and will gather some information and report back then...

freelancer1845, 2020-11-17:
So Updating the kernel didn't help much... I changed the format to x-bayer/rggb but it still breaks down after ~2min of playing.

<details>
<summary>When recreating the pipeline (2xtimes), this is the TCAM_LOG=DEBUG</summary>

```
>>>> First Pipeline creation
2020.11.17T13:13:19:893        <INFO> tcamgstbase.cpp:755: Fixating assumed largest caps: video/x-bayer, format=(string)rggb, width=(int)5472, height=(int)3648, framerate=(fraction)18/1
2020.11.17T13:13:19:898        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2020.11.17T13:13:19:898        <DEBUG> V4l2Device.cpp:577: Stopped stream
2020.11.17T13:13:19:898        <DEBUG> V4l2Device.cpp:181: Requested format change to 'format=RGGB,width=5472,height=3648,framerate=18.000000' 42474752
2020.11.17T13:13:19:899        <DEBUG> V4l2Device.cpp:403: Setting framerate to '18.000000' =  18 / 1
2020.11.17T13:13:19:901        <INFO> V4l2Device.cpp:436: Current framerate is 18 / 1 fps
2020.11.17T13:13:19:901        <DEBUG> V4l2Device.cpp:247: Active format is: 'format=RGGB,width=5472,height=3648,framerate=18.000000'
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RGGB' - 42474752
2020.11.17T13:13:19:901        <INFO> format.cpp:128: No string for fourcc 0x0
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc '' - 0
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'RG16' - 36314752
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'YUYV' - 56595559
2020.11.17T13:13:19:901        <INFO> format.cpp:128: No string for fourcc 0x0
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc '' - 0
2020.11.17T13:13:19:901        <DEBUG> PipelineManager.cpp:222: Found device fourcc 'BGR24' - 33524742
2020.11.17T13:13:19:901        <DEBUG> V4l2Device.cpp:181: Requested format change to 'format=RGGB,width=5472,height=3648,framerate=18.000000' 42474752
2020.11.17T13:13:19:903        <DEBUG> V4l2Device.cpp:403: Setting framerate to '18.000000' =  18 / 1
2020.11.17T13:13:19:904        <INFO> V4l2Device.cpp:436: Current framerate is 18 / 1 fps
2020.11.17T13:13:19:904        <DEBUG> V4l2Device.cpp:247: Active format is: 'format=RGGB,width=5472,height=3648,framerate=18.000000'
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> ImageBuffer.cpp:38: allocating data buffer
2020.11.17T13:13:19:904        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2020.11.17T13:13:19:904        <INFO> PipelineManager.cpp:486: source ->  sink
2020.11.17T13:13:19:904        <INFO> ImageSink.cpp:51: Pipeline started playing
2020.11.17T13:13:19:904        <DEBUG> V4l2Device.cpp:1606: Will use 10 buffers
2020.11.17T13:13:19:904        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f46cf6010) with length 19961856
2020.11.17T13:13:19:912        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:912        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f459ec010) with length 19961856
2020.11.17T13:13:19:919        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:920        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f446e2010) with length 19961856
2020.11.17T13:13:19:925        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:925        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f3ecf6010) with length 19961856
2020.11.17T13:13:19:930        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:930        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f3d9ec010) with length 19961856
2020.11.17T13:13:19:935        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:935        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f3c6e2010) with length 19961856
2020.11.17T13:13:19:939        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:939        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f3b3d8010) with length 19961856
2020.11.17T13:13:19:943        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:943        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f3a0ce010) with length 19961856
2020.11.17T13:13:19:946        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:946        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f38dc4010) with length 19961856
2020.11.17T13:13:19:950        <DEBUG> V4l2Device.cpp:1642: Successfully queued v4l2_buffer
2020.11.17T13:13:19:950        <DEBUG> V4l2Device.cpp:1633: Queueing buffer(0x7f1f37aba010) with length 19961856
[... cut ...]
2020.11.17T13:15:33:893        <DEBUG> V4l2Device.cpp:181: Requested format change to 'format=RGGB,width=5472,height=3648,framerate=18.000000' 42474752
2020.11.17T13:15:39:121        <ERROR> V4l2Device.cpp:231: Error while setting format 'Input/output error'
Debug ERROR  : Unable to set format in device
Debug WARN    tcambin-source: error: Internal data stream error.
Debug WARN    tcambin-source: error: streaming stopped, reason not-negotiated (-4)

```
</details>

I will dismount the camera and try it on a different pc :)

freelancer1845, 2020-11-17:
Ok the problem is definitely the usb port/hub on the mainboard. It works on other ports not on the same hub and other PCs. I didn't expect that form a HP Workstation mainboard and newer Linux Kernels....
If you want any more information I'm glad to provide it. But for me the issue is resolved... Thanks for noticing :)

TIS-Stefan, 2020-11-17:
I had a suspect about that,  because other users encountered similar issues, e.g. 75 fps expected, but only 4 achieved. Adding an extra USB 3.0 PCIe board solved the issue. 
However, that does not happen on all computers of this manufacturer.

closed 2020-11-17 by TIS-Stefan