TheImagingSource/tiscamera #381: Frame capture time in Python

martin-hora, 2020-11-20:
Hi,
Is it possible to extract the [frame capture time](https://www.theimagingsource.com/documentation/tiscamera/timestamps.html#capture-time) in Python?

My application runs a GStreamer pipeline with an appsink similar to [Example 7](https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py). So far, I have been using `GStreamer.Buffer` presentation timestamps (PTS) to estimate the time when a frame was taken. However, these timestamps are not precise enough for my purpose. I was thinking about using the capture times for a better estimation, but I do not know how to acquire those.

I have tried:
 * using the [get_reference_timestamp_meta](https://gstreamer.freedesktop.org/documentation/gstreamer/gstbuffer.html?gi-language=python#gst_buffer_get_reference_timestamp_meta) method of `GStreamer.Buffer`. But it always returns `None`. Maybe, I am not providing the correct caps `reference` parameter.
 * extracting the frame metadata using the [get_meta](https://gstreamer.freedesktop.org/documentation/gstreamer/gstbuffer.html?gi-language=python#gst_buffer_get_meta) method, but I do not know what to plug in for the `api` parameter.

I would be grateful for your assistance.

System information:
 * Camera: DMK 37BUX265
 * GStreamer version: 1.16.2
 * tiscamera version: 0.12.0
 * Python version 3.8
 * Operation system: Ubuntu 20.04

TIS-Stefan, 2020-11-20:
I am sorry, it is not possible doing that in Python. Only in C we can read the meta data.

martin-hora, 2020-11-20:
Ok, thanks for the answer.

closed 2021-02-19 by TIS-Stefan