TheImagingSource/tiscamera #382: Linux -DMM 42BUC03-ML- Missing parameters (Strobe Enable and trigger)

JacobSimuT, 2020-11-23:
Can you please advise on how to get the strobe enable and trigger parameters to work with the DMM 42BUC03-ML
I hav tried uvcdynctrl -i tisEUVC.xml -d /dev/video0 but parameters are not added.
Also the tcam-uvc-extension-loader seems to fail.

It is not clear in the Wiki which firmware to use for this camera:
In the "Getting Started with USB 2.0 CMOS Cameras" it states the dmk42uc03_3014.euvc and in the "compatibility-list" section it states dmk42uc03_143.euvc
But have tried both in mode UVC with no success to add the parameters.

In issue #309 @TIS-Arne  has a link to a python script, will that work with the DMM 42BUC03-ML camera and firmware?
If not can you advise how to do a similar script for the  DMM 42BUC03-ML camera and for both strobe and trigger parameters. 

The OS is Raspberry Pi OS Kernel version 5.4
BR
Jacob Simu

TIS-Stefan, 2020-11-23:
Hello

You may try that script made by Arne. 

Also we changed the uvc extension handling. Please use it as shown at to run the uvc extension manually as shown at https://www.theimagingsource.com/documentation/tiscamera/tcam-uvc-extension-loader.html.

Stefan

JacobSimuT, 2020-11-23:
I will try again with the 0.13.0 release and also Arnes script. 

How about the FW in the camera is the dmk42uc03_3014.euv or the dmk42uc03_143.euvc and should the uvc mode or the proprietary mode be used? 

The wiki sends mixed messages about this.

BR
Jacob

JacobSimuT, 2020-11-24:
With version 0.12.0 the tcam-uvc-extension-loader I still do not get it to work:
With fresh installation with the debian package the tcam-uvc-extension loader is called from the /etc/udev/rules.d/80-theimagingsource-cameras.rules, as I parse it, it will run the usb2.json:
I get the following when trying to run it manually:
pi@raspberrypi:~ $ sudo tcam-uvc-extension-loader -f Downloads/tiscamera-v-tiscamera-0.12.0/data/uvc-extensions/usb2.json -v
Error while mapping 'ExtIO': errno: 2 - No such file or directory
Error while mapping 'GainR': errno: 2 - No such file or directory
Error while mapping 'GainG': errno: 2 - No such file or directory
Error while mapping 'GainB': errno: 2 - No such file or directory
Error while mapping 'Trigger': errno: 17 - File exists
Error while mapping 'Trigger Global Reset Shutter': errno: 17 - File exists
Error while mapping 'Software Trigger': errno: 2 - No such file or directory
Error while mapping 'Binning': errno: 17 - File exists
Error while mapping 'Skipping': errno: 2 - No such file or directory
Error while mapping 'X Offset': errno: 17 - File exists
Error while mapping 'Y Offset': errno: 17 - File exists
Error while mapping 'Strobe Enable': errno: 2 - No such file or directory
Error while mapping 'Strobe Polarity': errno: 2 - No such file or directory

With the 0.13.0 tcam-capture does not start up the camera, I will send log next time I test. The tcam-uvc-extension-loader gives same output as in 0.12.0.

The Arne script turn on the strobe but I need also to activate the trigger function and if possible change polarity of strobe.
BR,
Jacob

TIS-Stefan, 2020-11-24:
Hello

You can ignore the error messages, they are shown, because your camera does not have these properties.
For the other things, I must check tomorrow.

Stefan

TIS-Stefan, 2020-11-24:
I just tested a DFK 42BUC03 with tiscamera 0.13.0 and firmware 3014 (dfk42uc03_3014.euvc). The camera shows trigger and software trigger on my system. 
[image]
So it should do on your computer too.

Stefan

TIS-Stefan, 2020-11-24:
It works the same for the DMK 42BUC02:
[image]

The firmware for the DMM and DMK is the same

Stefan

closed 2021-02-19 by TIS-Stefan