TheImagingSource/tiscamera #383: Camera not working

andreabonacini, 2020-11-25:
Hello,

I have installed tiscamera on Ubuntu 20.04 (Virtual Machine) in order to use the camera model DFK 33GX174. I followed the tutorial for the installation but testing some python examples it does not work. On debug of the 02 example, the flow stops when trying to get the exposure node.

**This is the output:**

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.690: Name 'tcambin-src_caps' is not unique in bin 'tcambin0', not adding

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.690: Name 'tcambin-exposure' is not unique in bin 'tcambin0', not adding
Expposure Auto has value: True

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.690: Name 'tcambin-src_caps' is not unique in bin 'tcambin0', not adding

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.690: Name 'tcambin-exposure' is not unique in bin 'tcambin0', not adding
Gain Auto has value: True

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.691: Name 'tcambin-src_caps' is not unique in bin 'tcambin0', not adding

(02-set-properties.py:13046): GStreamer-WARNING **: 16:00:39.691: Name 'tcambin-exposure' is not unique in bin 'tcambin0', not adding
Traceback (most recent call last):
  File ".../tiscamera/examples/python/02-set-properties.py", line 100, in <module>
    sys.exit(main())
  File ".../tiscamera/examples/python/02-set-properties.py", line 84, in main
    print_properties(camera)
  File ".../tiscamera/examples/python/02-set-properties.py", line 60, in print_properties
    category, group) = camera.get_tcam_property("Exposure")
TypeError: unknown type (null)

(02-set-properties.py:13046): GStreamer-CRITICAL **: 16:00:39.957: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed

(02-set-properties.py:13046): GStreamer-CRITICAL **: 16:00:39.957: gst_bin_remove: assertion 'GST_IS_ELEMENT (element)' failed

Also i tried to verify the installation with tcam-capture and tcam-ctrl -l. 

**First one error:**
/usr/lib/python3/dist-packages/tcam_capture/ImageProvider.py:31: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if name is "default":
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

**Second one output:**
No devices found.

Note: 
- I can ping the camera from both windows and ubuntu. 
- On windows i can use it with the IC Capture app.
- At the end of the successful installation of dependencies i get a warning about libzip4 is not found. In include dir i have libzip5

TIS-Stefan, 2020-11-26:
Hello
first of all, please make sure with "tcam-ctrl --version", that you use the current 0.13.0 version.  If you do not have this version, then please update (git pull).
I tried the 02-set-properties.py on my Ubuntu 20.04 on my desktop x86 computer and the only error message I receive is, that Brightness property is null. That is expected, because the DFK 33GX174 has no Brightness property.
The 02-set-properties.py shows only how to handle properties. The used properties are only examples, showing how tis is expected to work. 
I am a little bit surprised about the warning you see, I do not see them

Stefan

andreabonacini, 2020-11-26:
Actualy, the version was 0.12.0. 
Uptdating it the tcam-ctrl -l command works and i can see the camera, while tcam-capture gives this error (but at least now it starts):

[2020-11-26:16:05:00:596] [   ERROR] [         TcamView.py:509] - ERROR: bin : Could not create element 'tcambayertransform'.

Anyway now the 02 example gives the same message as yours.

Thanks for the reply.

Andrea

TIS-Stefan, 2020-11-27:
Hello Andrea

[2020-11-26:16:05:00:596] [ ERROR] [ TcamView.py:509] - ERROR: bin : Could not create element 'tcambayertransform'.

That is not an error. The tcambayertransform is a module, that is used, if a MIPI camera and our (to be released) PIMIPIsrc is in use on a Raspberry PI. I suppose, you have neither of them. Therefore, this "ERROR" can be ignored. It will be changed to "INFO" in future. 

> Anyway now the examples work and the 02 gives me the same message as yours.

That is expected with your camera, because it has no Brightness property. However, the samples are for showing how to use a property (that exists on your device),

Stefan

andreabonacini, 2020-11-27:
Anyway i'm still not able to save an image using tcam-capture. Clicking on the save image button give me this error:

Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 420, in save_image_action
    self.view.save_image(self.settings.get_image_type())
  File "/usr/lib/python3/dist-packages/tcam_capture/TcamView.py", line 190, in save_image
    self.imagesaver.save_image(get_encoder_dict()[image_type])
  File "/usr/lib/python3/dist-packages/tcam_capture/MediaSaver.py", line 246, in save_image
    self._create_pipeline()
  File "/usr/lib/python3/dist-packages/tcam_capture/MediaSaver.py", line 138, in _create_pipeline
    self.caps.to_string(),
AttributeError: 'NoneType' object has no attribute 'to_string'

And then i have a crash reporting while the viewer stays in: "In trigger mode. Waiting for first image...".
Also the hotkeys do not work.
I checked the tmp default saving dir but there are no images.

andreabonacini, 2020-11-27:
Also, trying to change some properties in that example i noticed that i can't set them and the printing function shows the properties before.

v-prgmr, 2021-01-29:
@andreabonacini ,  i got the same issue now. While using tcam-capture, i get the same error. Did you find a fix?

TIS-Stefan, 2021-01-29:
Did you update from git hub, so you have latest 0.13.0 version?

v-prgmr, 2021-01-29:
@TIS-Stefan , yes I checked the version and it was the latest version.

TIS-Stefan, 2021-02-19:
The message ""In trigger mode. Waiting for first image..."  is a little bit optimistic. I says only, that the camera could be detected but no images are delivered. 
In a virtual machine this can be an issue of the network connection from VM to the host's hardware. In VMWare I made the network connection as "Bridged (connect VMs directly to the external Network)".

As long, as no images are received, image saving can not work too. 

Please let me know, what the current status of your camera situation is.

closed 2021-07-28 by TIS-Stefan