TheImagingSource/tiscamera #384: Whitebalance auto cannot be disabled

Kotesh81, 2020-11-29:
Hi team,  

I have been working on TIS cameras for a while now. One of the camera is Model: DBK 33UX264 Serial: 19910245 Type: v4l2. I am using tcam-capture application built for linux to capture images. When I try to disable whitebalance auto by unchecking the field in color tab, the whitebalance values of green, red, and blue channels still changed automatically. I have walked through the logic and found the properties were set correctly but we still see this issue. Please let me know if there are any leads to fix this issue.

TIS-Stefan, 2020-11-30:
Hello

Are we in contact by email already?

Stefan

Kotesh81, 2020-11-30:
Yes the same one

tehkillerbee, 2020-12-01:
We are experiencing the same issue with the following two cameras:
Model: DFK 33UX264 Serial: 45710071 Type: v4l2
Model: DFK 33UX264 Serial: 45710072 Type: v4l2

I see some of the relevant Whitebalance properties mentioned in the user guide (Whitebalance Mode and others) but these properties do not appear to be available when listing supported properties ( tcam-ctrl -p xx)

TIS-Stefan, 2020-12-01:
Could you please answer to the email thread too? (So we both have double work :-) ) (I can better forward this to programmers. )

TIS-Stefan, 2020-12-03:
We narrowed the issue. It is caused by setting properties in unexpected sequence and this causes the camera to be confused. In particular, this is caused by tcam-capture. Therefore, the "Apply cached properties when opening a device" option should be disabled in the tcam-capture options. 

The complete sequence of the workaround is:
- start tcam-capture
- disable Apply cache in tcam-capture
- close tcam-capture
- disconnect camera 
- reconnect camera
- start tcam-capture

now all automatics are enabled and you get a nice image. Now:
- disable exposure auto, if you do not want automatic
- disable gain auto ,if you do not want automatic
- disable Whitebalance Auto
- move R/G/B sliders.

The "'Whitebalance" checkbox changes only between white balance auto mode of "Grayworld" and "Temperature". It has no effect, if Whitebalance Auto is turned off.

As far as I know, only the USB cameras of 33U, 37U and 38U are affected. I am not sure about the GigE 33G and 37G cameras
The USB 2.0 CMOS cameras are not affected, because they have not automatics calculation on board.

The reason for this behavior is that v4l2 / UVC does not know dependencies between properties, e.g.  if Whitebalance Auto is true, it is not possible to set the RGB sliders. Unfortunately this is also not implemented in the UVC interface of the camera.  GigEVision and USB3Vision should know about these dependencies. (Therefore, using the USB cameras with Aravis instead of v4l2, this issue should not exist. But I did not test that explicitly).

If restoring properties from a file I suggest to make this kind of dependencies on your own. If e.g. exposure auto is on , then do not touch exposure time. Same for gain. If Whitebalance Auto is on, then do not change R/G/B and Temperature. I am aware, I should provide Python and C++ scripts for that, but I am running out of time at the moment.

Stefan

Kotesh81, 2020-12-03:
Thanks alot. The routine suggested kind of solved the issue for us. Definitely if there is a feature to save the configuration/properties/dependencies to a file in tcam-capture. Please update us with the python scripts mentioned in your reply.

TIS-Stefan, 2020-12-07:
I will make scripts like that, if I have some sparetime or found somebody else doing that. If there are scripts, they will be available at https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples. I am sorry for not being able to make that immediately.

Stefan

closed 2021-02-19 by TIS-Stefan