TheImagingSource/tiscamera #385: compile problem

Daniel-CSU, 2020-12-07:
when make
/home/wzl/tiscamera/src/gstreamer-1.0/gsttcambin.cpp: In function ‘GstStateChangeReturn gst_tcam_bin_change_state(GstElement*, GstStateChange)’:
/home/wzl/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1257:14: error: ‘GST_STATE_CHANGE_PLAYING_TO_PLAYING’ was not declared in this scope
         case GST_STATE_CHANGE_PLAYING_TO_PLAYING:
              ^
In file included from /usr/include/gstreamer-1.0/gst/gst.h:54:0,
                 from /home/wzl/tiscamera/src/gstreamer-1.0/gsttcambin.h:20,
                 from /home/wzl/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:17:
/home/wzl/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1259:75: error: ‘gst_state_change_get_name’ was not declared in this scope
             GST_INFO("Changing state: %s", gst_state_change_get_name(trans));
                                                                           ^
/usr/include/gstreamer-1.0/gst/gstinfo.h:546:31: note: in definition of macro ‘GST_CAT_LEVEL_LOG’
         (GObject *) (object), __VA_ARGS__);    \

TIS-Stefan, 2020-12-07:
That is strange, because it is defined in element.h:
https://github.com/alessandrod/gstreamer/blob/master/gst/gstelement.h, see line 257.

Did you install all dependencies with `./scripts/dependency-manager install` after cloning? 
Which Ubunutu version do you use? I tested on 18.04, 20.04 and 20.10

Stefan

TheoCarrier, 2020-12-09:
Hi, 

Same problem here. I've run all the dependencies

> In function ‘GstStateChangeReturn gst_tcam_bin_change_state(GstElement*, GstStateChange)’:
/home/corealis/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1257:14: error: ‘GST_STATE_CHANGE_PLAYING_TO_PLAYING’ was not declared in this scope
         case GST_STATE_CHANGE_PLAYING_TO_PLAYING:
              ^
In file included from /usr/include/gstreamer-1.0/gst/gst.h:54:0,
                 from /home/corealis/tiscamera/src/gstreamer-1.0/gsttcambin.h:20,
                 from /home/corealis/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:17:
/home/corealis/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1259:75: error: ‘gst_state_change_get_name’ was not declared in this scope
 GST_INFO("Changing state: %s", gst_state_change_get_name(trans));
                                                               ^
/usr/include/gstreamer-1.0/gst/gstinfo.h:546:31: note: in definition of macro ‘GST_CAT_LEVEL_LOG’
         (GObject *) (object), __VA_ARGS__);    \
                               ^
/home/corealis/tiscamera/src/gstreamer-1.0/gsttcambin.cpp:1259:13: note: in expansion of macro ‘GST_INFO’
             GST_INFO("Changing state: %s", gst_state_change_get_n
             ^
src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/build.make:62 : la recette pour la cible « src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/gsttcambin.cpp.o » a échouée
make[2]: *** [src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/gsttcambin.cpp.o] Erreur 1
CMakeFiles/Makefile2:716 : la recette pour la cible « src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/all » a échouée
make[1]: *** [src/gstreamer-1.0/CMakeFiles/gsttcambin.dir/all] Erreur 2
Makefile:149 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2

I'm running 16.04 LTS 32 bits version. Could it be the problem ? 

Thank you for your time, Théo

TIS-Stefan, 2020-12-09:
Hello

> I'm running 16.04 LTS 32 bits version. Could it be the problem ?
Maybe, but I think, it is rather not the issue. Did you run an apt-get update and apt-get updated in order to update the installed packages?

Stefan

TheoCarrier, 2020-12-09:
Hello,

Yes the installation of the linux has been made just before and the procedure has been followed multiples times and always the same error.

Thank you for your rapid answer, have a nice day
Theo

TIS-Stefan, 2020-12-09:
Hello Theo

I searched in the internet for that error and found https://askubuntu.com/questions/1144952/can-not-open-setting-in-ubuntu-18-04
maybe this helps.
Stefan

TheoCarrier, 2020-12-23:
Problem was solved installing 20.04 LTS on a new 64 bits PC work perfectly. I've followed exact same procedure. I'm thinking if it's not the 16.04 or the 32 bits who was the problem. 

Thanks you, Théo

closed 2021-02-19 by TIS-Stefan