TheImagingSource/tiscamera #386: whitebalance segfault

TheophileBlard, 2020-12-07:
When using `tcamwhitebalance` in our application, a segfault is thrown when calling the `next_line` function.
After some investigation (gdb), we found the `next_line` function of libacl is called instead of the `next_line` function declared in `bayer.h`, which cause the segfault. Our application is linked with libarchive, which is itself linked to libacl.

To solve this issue we explicitly linked our executable with libtcam-algorithms, which highlighted another issue:  libtcam-algorithms.so is not linked to libtcam.so (undefined symbols when use ldd -r). We then also linked our executable to libtcam.so and `tcamwhitebalance` is now working properly. Using `LD_PRELOAD` also work.

However, this is not satisfying for us, as we don't want to specifically link to tcam librariries. Do you have a quick fix/patch to ensure the right  `next_line` is called ? We are using v0.12.0.

TIS-Stefan, 2020-12-07:
Please let me know, which camera model you use.
Can you please show me the pipeline, you use?

Thank you in advance!
Stefan

TheophileBlard, 2020-12-08:
Output of `tcam-ctrl -l`:
```shell
Available devices:
Model		Type	Serial

DFK Z12GX236	aravis	46910269
```

Pipeline:
```shell
LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libarchive.so \
gst-launch-1.0 tcambin ! "video/x-raw, width=1280, height=720, framerate=30/1" ! videoconvert ! ximagesink
````

We are using `LD_PRELOAD` to emulate that our application is linked to libarchive.

Output of `gdb`:
```shell
Thread 16 "tcambin-source:" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f957fa1f0 (LWP 10169)]
_IO_fgets (buf=0x7f6c000b20 "", n=4096, fp=0x1) at iofgets.c:47
47	iofgets.c: No such file or directory.
(gdb) bt
#0  0x0000007fb7bcc138 in _IO_fgets (buf=0x7f6c000b20 "", n=4096, fp=0x1) at iofgets.c:47
#1  0x0000007fb7fbf848 in next_line () at /lib/aarch64-linux-gnu/libacl.so.1
#2  0x0000007fb7482210 in tcam::algorithms::whitebalance::para_wb_callback::call(tcam_image_buffer const&, tcam_image_buffer const&) ()
    at /usr/lib/libtcam-algorithms.so.0
#3  0x0000007fb74842d4 in tcam::algorithms::parallel::parallel_state::split_image_context::do_one() () at /usr/lib/libtcam-algorithms.so.0
#4  0x0000007fb7483b94 in tcam::algorithms::work_pool::work_pool::worker_thread_function(int) () at /usr/lib/libtcam-algorithms.so.0
#5  0x0000007fb75a4e94 in  () at /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000007fb7cca088 in start_thread (arg=0x7fadff78ef) at pthread_create.c:463
#7  0x0000007fb7c3a4ec in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

kalvdans, 2020-12-08:
Seems to be fixed on the libacl side in 2014:  https://git.savannah.nongnu.org/cgit/acl.git/commit/?id=809584283a3ce5b0250656faefb23a821782adbe

The same should be done for libtcam, either using attribute(hidden) or a unique prefix.

TIS-Edgar, 2020-12-08:
Currently looking into this.

Not reproducible with debian testing (amd64). libarchive version 3.4.3.

Which distribution are you using?

>  libtcam-algorithms.so is not linked to libtcam.so 

That is indeed a bug.

Is there anything special about your setup? I am somewhat surprised as this is the first time this issue has come up.

I am looking into options to patch this in a reliable manner....

TheophileBlard, 2020-12-09:
Our target is a Jetson Xavier with NVIDIA L4T 32.4.4 (based on Ubuntu 18.04), but I have also reproduced this issue on my x86 host computer (Ubuntu 18.04). libarchive version is 3.2.2 on both systems.

TIS-Stefan, 2021-02-19:
Hello 

For the tiscamera 0.13.1 and tiscamera-dutils 0.13.0 release I made many tests on TX Nano. I did not get an error at all. So you may update your tiscamera version and check, whether the error is gone?

Stefan

TheophileBlard, 2021-02-22:
As pointed out by https://github.com/TheImagingSource/tiscamera/issues/386#issuecomment-740531305, this is not directly an issue with tiscamera, but a compatibility issue with the version of libacl shipped with Ubuntu 18.04. Did you try the example pipeline I provided, with `LD_PRELOAD` ? On my end, I will update tiscamera to the latest version, and see if it works (but the `next_line` function has not been modified). In fact, I should do a pull a request with https://github.com/TheImagingSource/tiscamera/issues/386#issuecomment-740531305 fix suggestion.

TIS-Edgar, 2021-02-22:
We are still unable to reproduce this issue. If you have a working fix, feel free to provide it.
We are planning on a permanent fix for the next stable release by changing the default visibility to hidden and moving all functions to namespaces. Since that release is still months away a transitional solution is welcomed.

TIS-Edgar, 2022-06-07:
Considered obsolete. tcamwhitebalance is deprecated. Calculation now performed in tcammainsrc and application in tcamconvert/tcamdutils.

closed 2022-06-07 by TIS-Edgar