TheImagingSource/tiscamera #387: USB cable for DFM 37UX226-ML

Moj-Dev, 2020-12-12:
I'm using [this USB cable](https://www.londondrugs.com/certified-data-premium-usb-3.1-type-c-to-usb-3.0-a-female---guc05-1000/L6473672.html) for 37UX226 camera and couldn't get frames over 14FPS in RGB32(4000x3000). The problem is the cable? Should I use only the[ TIS USB3.1 cable](https://www.theimagingsource.com/products/cables/usb-3.1/causb31acbls3/)?

TIS-Stefan, 2020-12-14:
Hello

Its a good question. Which software do yo use? Maybe the software or computer is too slow for debayering the 12Megapixel image.

Stefan

Moj-Dev, 2020-12-14:
Software: IC Capture 2.5 on Windows 10
Laptop: Lenovo Y7000P-1060

TIS-Stefan, 2020-12-15:
This is the Linux issue area. 

However, check for power saving mechanism and try to disable them. 
Also let me know, what the four numbers in the status bar, right corner of IC Capture, show.

Moj-Dev, 2020-12-15:
I wanted to test what you have said, but I got an error. "The device 'DFM 37UX226-ML' is in live mode, but it does not deliver images". 
[image]

I changed the format and frame rate, none of them worked and I can not get any frames!
This camera is the MOST USERENEMY PRODUCT I've ever seen! First, there were lots of blue dots in the frames and as I discussed with you, I returned it and got another one. Then, your claim about getting 30 FPS in full resolution was not correct, and now it doesn't work for no reason!

TIS-Stefan, 2020-12-16:
Hello

You are not the only user of this camera. But you are the only one, having these issues.
Which USB cable do you use? If you use a cable provided with a mobil phone, then you wont get images, because these cables are usually USB 2.0 only.

At least, there is always the option to give the camera back, if you are not able to get images.

Moj-Dev, 2020-12-17:
The camera worked for a while with [this USB3.0 cable ](https://www.londondrugs.com/certified-data-premium-usb-3.1-type-c-to-usb-3.0-a-female---guc05-1000/L6473672.html). 
I've ordered your cable and going to test it.

TIS-Stefan, 2020-12-18:
If a camera worked for a while and then does not, then something must have changed.
Question is, what?
I would have a look by teamviewer on your computer, but I suppose, we are on different time zones, teamviewer logins are shared by email and I will be on holidays for next two weeks.

Some more hints:
Windows 10

- Check with "winver" the Windows version.
If 1607, Anniversary Update: It blocks USB cameras. It is fixed by Windows Updates, thus perform updates.

- Virus Scanner
Check for virus scanner blocking webcam streams. e.g. Kaspersky Endpoint Security and ESET do so.

Moj-Dev, 2020-12-21:
I tried the camera with TIS USB cable both on Windows 10 (Version 1809) and Ubuntu 18.04, It didn't work! I tried with all formats and framerate, none of them worked! Also, tried another PC with Windows OS, it was the same!

TIS-Stefan, 2021-02-19:
Hello

Did you sent the camera in to your camera dealer for checking? Maybe it is broken somehow?

Stefan

closed 2021-07-28 by TIS-Stefan