TheImagingSource/tiscamera #390: Camera works 1st time after plugging in, but never after that [Device lost]

pintariching, 2021-01-17:
I have a DFK27AUJ003 camera connected via the USB3.0 cable to a 3.0 USB port on the motherboard.
I'm running Ubuntu 20.04, however the same problem occurs when I move to a RaspberryPi 4.

The strange thing is, upon connecting the camera and starting tcam-capture, everything works great. I get a picture, I can change settings and everything. However once I close tcam-capture and it closes the stream, if I try to open it once again, an error pops up saying: Device has been lost. Please reconnect or re-open it.

Below is the log that I get
```
2021.01.17T14:34:43:886        <DEBUG> Logger:108: 
The following library versions are used:
	Tcam:	0.13.0_master/635b652a_rev_2999
	Aravis:	not_configured_version_not_configured
2021.01.17T14:34:48:250        <INFO> V4l2Device.cpp:410: Current framerate is 6 / 1 fps
2021.01.17T14:34:48:250        <WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Exposure Time (us)'(199e201)
2021.01.17T14:34:48:250        <WARNING> v4l2_utils.cpp:145: Unable to find std property. Passing raw property identifier through. 'Gain (dB/100)'(199e204)
2021.01.17T14:34:48:250        <INFO> V4l2Device.cpp:1170: Encountered write only control.
2021.01.17T14:34:48:250        <WARNING> v4l2_utils.cpp:226: Detected stepsize 0 for property Strobe Delay. Setting to 1.
2021.01.17T14:34:48:250        <DEBUG> v4l2_utils.cpp:278: Enum Trigger Polarity does not have any entries. Ignoring.
2021.01.17T14:34:48:250        <ERROR> V4l2Device.cpp:1208: Property 'Trigger Polarity' is null
2021.01.17T14:34:48:250        <DEBUG> V4l2Device.cpp:673: Found format: GRBG
2021.01.17T14:34:48:250        <DEBUG> V4l2Device.cpp:673: Found format: Y800
2021.01.17T14:34:48:250        <INFO> format.cpp:128: No string for fourcc 0x0
2021.01.17T14:34:48:250        <DEBUG> V4l2Device.cpp:673: Found format: 
2021.01.17T14:34:48:250        <DEBUG> V4l2Device.cpp:673: Found format: Y16
2021.01.17T14:34:48:250        <DEBUG> V4l2Device.cpp:243: Returning 4 formats.
2021.01.17T14:34:48:250        <DEBUG> PipelineManager.cpp:127: Received 4 formats.
2021.01.17T14:34:48:250        <DEBUG> PropertyHandler.cpp:273: Created 19 properties
2021.01.17T14:34:48:250        <INFO> tcamgstbase.cpp:1448: Format has empty fourcc. Ignoring
2021.01.17T14:34:48:258        <INFO> tcamgstbase.cpp:882: Fixating assumed largest caps: video/x-bayer, format=(string)grbg, width=(int)3872, height=(int)2764, framerate=(fraction){ 6/1, 5/1, 4/1, 3/1, 2/1 }
2021.01.17T14:34:48:258        <INFO> tcamgstbase.cpp:923: Largest caps are: video/x-bayer, framerate=(fraction)6/1, width=(int)3872, height=(int)2764, format=(string)grbg
2021.01.17T14:34:48:258        <INFO> tcamgstbase.cpp:882: Fixating assumed largest caps: video/x-bayer, framerate=(fraction)6/1, width=(int)3872, height=(int)2764, format=(string)grbg
2021.01.17T14:34:48:258        <INFO> tcamgstbase.cpp:923: Largest caps are: video/x-bayer, framerate=(fraction)6/1, width=(int)3872, height=(int)2764, format=(string)grbg
2021.01.17T14:34:48:258        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2021.01.17T14:34:48:258        <DEBUG> V4l2Device.cpp:518: Stopping stream
2021.01.17T14:34:48:258        <DEBUG> V4l2Device.cpp:543: Stopped stream
2021.01.17T14:34:48:258        <DEBUG> V4l2Device.cpp:163: Requested format change to 'format=GRBG,width=3872,height=2764,framerate=6,000000' 47425247
2021.01.17T14:34:48:261        <DEBUG> V4l2Device.cpp:376: Setting framerate to '6,000000' =  6 / 1
2021.01.17T14:34:48:264        <INFO> V4l2Device.cpp:410: Current framerate is 6 / 1 fps
2021.01.17T14:34:48:264        <DEBUG> V4l2Device.cpp:222: Active format is: 'format=GRBG,width=3872,height=2764,framerate=6,000000'
2021.01.17T14:34:48:264        <DEBUG> PipelineManager.cpp:219: Found device fourcc 'GRBG' - 47425247
2021.01.17T14:34:48:264        <DEBUG> PipelineManager.cpp:219: Found device fourcc 'Y800' - 30303859
2021.01.17T14:34:48:264        <INFO> format.cpp:128: No string for fourcc 0x0
2021.01.17T14:34:48:264        <DEBUG> PipelineManager.cpp:219: Found device fourcc '' - 0
2021.01.17T14:34:48:264        <DEBUG> PipelineManager.cpp:219: Found device fourcc 'Y16' - 20363159
2021.01.17T14:34:48:264        <DEBUG> V4l2Device.cpp:163: Requested format change to 'format=GRBG,width=3872,height=2764,framerate=6,000000' 47425247
2021.01.17T14:34:48:267        <DEBUG> V4l2Device.cpp:376: Setting framerate to '6,000000' =  6 / 1
2021.01.17T14:34:48:270        <INFO> V4l2Device.cpp:410: Current framerate is 6 / 1 fps
2021.01.17T14:34:48:270        <DEBUG> V4l2Device.cpp:222: Active format is: 'format=GRBG,width=3872,height=2764,framerate=6,000000'
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> ImageBuffer.cpp:38: allocating data buffer
2021.01.17T14:34:48:270        <INFO> PipelineManager.cpp:476: Pipeline creation successful.
2021.01.17T14:34:48:270        <INFO> PipelineManager.cpp:486: source ->  sink
2021.01.17T14:34:48:270        <INFO> ImageSink.cpp:51: Pipeline started playing
2021.01.17T14:34:48:270        <DEBUG> V4l2Device.cpp:1515: Will use 10 buffers
2021.01.17T14:34:48:270        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc6165c9010) with length 10702208
2021.01.17T14:34:48:273        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:273        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc615b94010) with length 10702208
2021.01.17T14:34:48:275        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:275        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc61515f010) with length 10702208
2021.01.17T14:34:48:277        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:277        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc61472a010) with length 10702208
2021.01.17T14:34:48:279        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:279        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60f5cb010) with length 10702208
[... 1 lines cut ...]
2021.01.17T14:34:48:280        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60eb96010) with length 10702208
2021.01.17T14:34:48:282        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:282        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60e161010) with length 10702208
2021.01.17T14:34:48:284        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:284        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60d72c010) with length 10702208
2021.01.17T14:34:48:286        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:286        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60ccf7010) with length 10702208
2021.01.17T14:34:48:288        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:288        <DEBUG> V4l2Device.cpp:1542: Queueing buffer(0x7fc60c2c2010) with length 10702208
2021.01.17T14:34:48:290        <DEBUG> V4l2Device.cpp:1551: Successfully queued v4l2_buffer
2021.01.17T14:34:48:291        <INFO> V4l2Device.cpp:509: Starting stream in work thread.
2021.01.17T14:34:48:291        <DEBUG> ImageSource.cpp:57: PLAYING....
2021.01.17T14:34:48:291        <INFO> PipelineManager.cpp:83: All pipeline elements set to PLAYING.
2021.01.17T14:34:50:293        <ERROR> V4l2Device.cpp:1409: Timeout while waiting for new image buffer.
2021.01.17T14:34:50:293        <INFO> V4l2Device.cpp:556: notifying callbacks about lost device
2021.01.17T14:34:50:297        <INFO> ImageSource.cpp:68: Source changed to state STOPPED
2021.01.17T14:34:50:297        <DEBUG> V4l2Device.cpp:518: Stopping stream
[... cut ...]
After un-plugging and re-plugging the camera back in the USB port, the behavior is the same. On the first try, it functions properly, but after exiting and re opening tcam-capture, it times out on every time.

The only solution I can come up with to try and get around this, is by connecting the camera to an USB hub that can power down each USB port (My motherboard doesn't support it for some reason) and cycle the power on the port every time I open and close the software, so that I don't have to re-plug the USB cable every time.

TIS-Stefan, 2021-01-20:
Hello

I am sorry for the delay of my answer.
Can you please show the output of dmesg after the error occurred? Also I would like to see the output of "lsusb -t"
Does the same issue happen with qv4l2 or guvcview too?

Sorry for asking questions first.

Stefan

pintariching, 2021-01-20:
Hi

No need to apologize I understand. I'll send you the logs later today as I'm currently busy.

Kind regards

pintariching, 2021-01-20:
Okay after some tinkering tcam-capture no longer displayed neither the options of the camera nor the image, but on the upside, I was using the camera in a python flask project where it was giving me some trouble, printing the error `Error starting pipeline`. I then found out that running the flask project with debug=false resulted in the python code working perfectly. 

For anybody wondering, here is how I build a flask RESTFUL api that sends the image, captured from a DFK27AUJ camera.

```
from flask import Flask, Response
from flask_restful import Api, Resource
import TIS  # Importing module from Linux-tiscamera-Programming-Samples/python-common/TIS.py
import cv2
import base64
import json
import signal
import sys

app = Flask(__name__)
api = Api(app)

Tis = TIS.TIS()
Tis.openDevice("13020300", 1280, 960, "30/1", TIS.SinkFormats.BGRA, False) 

Tis.Start_pipeline()

# I found that setting properties after starting the pipeline seemed to work
Tis.Set_Property("Exposure Auto", False)
Tis.Set_Property("Gain Auto", False)

# Had to define a gracefull exit to stop the pipeline after pressing ctrl+c to stop the flask app
def gracefullExit(sig, frame):
    print("Stopping TIS pipeline")
    Tis.Stop_pipeline()
    sys.exit(0)

signal.signal(signal.SIGINT, gracefullExit)

def getCameraImage():
    if Tis.Snap_image(1) is True:
        image = Tis.Get_image()  # Get the image from the camera
        return image

class ImageRequest(Resource):
    def get(self):
        image = getCameraImage()
        retval, buffer = cv2.imencode(".jpg", image) # Encode the image from a numpy array (I think) to a JPG string
        jpg_as_bytes = base64.b64encode(buffer) # Encode the JPG string as a byte array
        jpg_as_text = jpg_as_bytes.decode("utf-8") # Decode the byte array into a string, that can be displayed on a website 
                                                                             # By using the tag <img src="data:image/jpg;base64, jpg_as_text"

        return {"image": jpg_as_text} 

api.add_resource(ImageRequest, "/getimage")

if __name__ == "__main__":
    app.run()  # If debug=True, the app has trouble starting a pipeline 
```

closed 2021-01-22 by pintariching