TheImagingSource/tiscamera #391: Missed frame from camera with strobe_enabled=1

dzhura, 2021-01-21:
Hi

I have two DFK 38UX267 I want to work synchronously, where one camera, master, sends a trigger signal to the other one, slave, via a cable.

I was successful to make the cameras trigger at the same time with correct exposure value. However, the problem is that slave delivers an extra frame at the start of streaming. On the other side, it looks like the master camera misses a caputer or slower to start streaming than the salve camera. I put a timer in front of the cameras to check if two images are made at the same time.

ATM, I am using v4l API directly, though I have the same bug when use tcamsrc and gstreamer.

I set strobe_enable to true straight after calling ioctl with VIDIOC_STREAMON by calling  ioctl(fd_master, VIDIOC_S_CTRL, enable_strobe), where enable_strobe is v4l2_param with parameter id and 1 (it works because oscilloscope start to receive signals)

I turn off strobe straight after VIDIOC_STREAMOFF. After that the oscilloscope shows no signal.  

Here is an illustration. The framerate is 1 fps. Frame # is the number of arrived frame from a stream.

frame #                                                  1      2      3   ... 
Time shown on master frames (sec):     10    11    12  ...   
Time shown on slave frames (sec)         9      10    11  ... 

If I set the master to do 1 capture and slave to do 10, then:
frame #                                                  1      2      3   ... 
Time shown on master frames (sec):     10       
Time shown on slave frames (sec)         9      10    
No surprise, slave timeouts because it does not receive any signal. 

The question is: why the master camera does not deliver a frame captured at the time 9?

The same problem when use tcamsrc and gstreamer and set strobe_enabled to 1 before running the application with the pipeline.

TIS-Stefan, 2021-02-19:
First of all, sorry for the delay of my answer.
I guess, you get an extra image on slave, because the master sets the strobe out according the strobe polarization property. The slave reacts on this change.
Since this is not a software issue, I would like to ask, whether you can create the setup using Windows and IC Capture 2.5? With that you will have access to all camera properties.

For interest I would like to know: You made a correct Open Drain circuit from the master's strobe out to the slave's trigger in, right?

Stefan

ChristopherLenzCMD, 2021-02-23:
Hi Stefan,

I am having a similar problem when using a Window/Halcon based set up. We have configured our circuit to act as an open-drain supplying a falling edge to the slave cameras on a rising edge from the camera (the FET changing to drain).

We are seeing multiple strobes coming out of our master camera. We have varied the FPS and this only delays the pulses. Despite this, we are getting the last or middle frame from the master's strobes. In other words, we are missing frames from the master.

in addition to this the time to "snap" a single frame on master is roughly 3 seconds at 1 FPS. The slave takes 3 photos and one of them (mostly the last one) matches the single frame that comes out of the master.

dzhura, 2021-02-24:
Hi Stefan,

Thanks for your reply. I have the same issue on Windows. I guess, it means the problem does not relate to tiscamera.

I am pretty sure the circuit is correct - the output signal was measured by oscilloscope. BTW, interesting fact: a camera starts to generate a signal as soon as the strobe is on.

Cheers,
Maxim

TIS-Stefan, 2021-02-24:
Hello

I would like you both show me your OpenDrain circuits drawing as well as to which camera pins the lines are connected. I will forward to my engineers in order to check them.

Also please make sure your trigger signals have a clear 3.3V or above level and no noise. That is to avoid you have more than one trigger signal.

I tested the DFK 38UX267 and I get for each trigger exactly one rising and falling edge. 

Stefan

TIS-Stefan, 2021-02-25:
Christopher

according to the email you send to me:

> Each time we "snap" the camera to get a single image, we receive multiple strobe output pulses. this is causing a slave/sync camera to capture multiple images.
What do you mean by "snap"? What are you doing?

Stefan

ChristopherLenzCMD, 2021-02-25:
we are taking a single frame, not running in live

Kind Regards
Christopher Lenz

TIS-Stefan, 2021-02-25:
I continued that by E-Mail, so we do not have double workload.

TIS-Stefan, 2021-03-10:
Hello Maxim

that is correct and I think, I was the one, who answered to them,

Stefan

TIS-Stefan, 2021-03-11:
Hello Maxim

no, there is not as far as I know.
However, we suggested a workaround by using software trigger for the master.  This would avoid the issue and could be faster than the Halcon script, that was provided to us.

Stefan

dzhura, 2021-03-12:
Hi Stephan,

Thanks you for your help.

Cheers,
Maxim

closed 2021-07-28 by TIS-Stefan