TheImagingSource/tiscamera #392: Unable to use multiple cameras on Ubuntu -  Unable to display caps.

lior3032, 2021-01-27:
I'm trying to use two ImagingSource Ethernet cameras in Ubuntu. Both cameras are from the same model (33Gp1300e) and connected to the same network card. Unfortunately, I'm unable to record from both cameras parallely. when trying to record from more than 1 camera via gstreamer I get an error message saying there are no available caps.

```
Debug ERROR  : Device did not provide ANY valid caps. Refusing playback.
Debug ERROR  : Device did not provide ANY valid caps. Refusing playback.
0:00:07.588805536 27720 0x7efc78299320 ERROR                tcamsrc gsttcamsrc.cpp:770:gst_tcam_src_get_all_camera_caps: Device did not provide ANY valid caps. Refusing playback.
```
Running the command:
`tcam-ctrl -c serial_number `

Returns a valid input for one camera,  but an error message for the second camera.

`Unable to display caps. Conversion failed. `

One thing to note, is that I get the error message for the second camera that I connect, the problem is not with one of the two cameras but with using more than one camera at the same time.

Any help would be appreciated.

TIS-Stefan, 2021-01-27:
Hello

If both cameras are connected to the same network, then you must take care of network bandwidth. Therefore, you reduce the frame rates of both cameras to the half of the possible maximum.

Also make sure, the camera's and network controllers IP addresses are configured correctly. Use the tcam-gigetool for getting the information about IP addresses. It should look like follows:
```
$ tcam-gigetool list

------------------------ Interface: enp1s0 [192.168.2.5] ------------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DFK 33GX174          | 14610619         | ?                    | 192.168.2.176    
DMK 33GP1300         | 23519996         |                      | 192.168.2.6      
DFK 23GP031          | 30210020         |                      | 192.168.2.3      
DFK 33GP1300         | 33510567         | Kam Cam              | 192.168.2.4      
DMK 33GP031          | 34510079         |                      | 192.168.2.8      
DFK Z12GX236         | 38810653         |                      | 192.168.2.9      
DFK 39GX265-Z20      | 39910315         |                      | 192.168.2.10     
DFK Z30GP031         | 41910044         |                      | 192.168.2.2      
DMK Z30GP031         | 41910051         |                      | 192.168.2.7      
DMK 23G618           | 43210118         | ?                    | 192.168.0.44   

```
That is the configuration on my test computer.

lior3032, 2021-01-28:
Hello,

I don't think it's a bandwidth issue. When I have two camera connected, only one camera works, and by works I mean able to record. I'm not even able to retrieve the caps from both cameras.

In addition, if I disconnect the camera that works, the other camera still doesn't work, I need to restart the device so I'll be able to record from it. It seems that when I have one camera connected, the other camera doesn't work from some reason.

This is my tcam-gigetool output:

```
$ tcam-gigetool list
----------------------- Interface: enp4s0 [192.168.2.100] -----------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP1300e        | 13810206         | ?                    | 169.254.24.175   

----------------------- Interface: enp5s0 [192.168.2.101] -----------------------

     Model Name      |  Serial Number   |  User Defined Name   |     Current IP   
DMK 33GP1300e        | 13810225         | ?                    | 169.254.101.161  

```

lior3032, 2021-01-28:
I managed to solve the problem. I connected the cameras to a switch and both are identified. I still think the issue should remain open, since when the cameras are connected to different interfaces only one works. Let me know if you need any further clarifications from me.

TIS-Stefan, 2021-01-28:
if the subnetmask is 255.255.255.0, then the operating system can not distinguish between your network controllers and the cameras wont run. You may use 
enp4s0 : 192.168.2.100:255.255.255.0 
enp5s0 : 192.168.3.100:255.255.255.0 

Also the cameras run on DHCP, while you have fixed IP adresses in your network controllers. The addresses do not match, so no connection
You set the IP adressess of the cameras to fixed (persistant) 192.168.2.101 and 192.168.3.101. Subnetmask would be 255.255.255.0 for both of them.

Stefan

closed 2021-02-19 by TIS-Stefan