TheImagingSource/tiscamera #393: Unable to retrieve device info for.....

v-prgmr, 2021-02-01:
Hi,

I am using two gigE cameras. Model: DFK 33GX273. 
I have built tiscamera with Aravis and now when i list the devices using tcam-ctrl -l, the device gets detected the first time. 

Running tcam-capture right after lets me choose the camera but after that the applications freezes. Closing the application and running tcam-ctrl -l again gives the following 

`$ tcam-ctrl -l`
`Unable to retrieve device info for 'H��tWdH3,%0'`

What could be the reason?

TIS-Stefan, 2021-02-02:
Hello

I am sorry, but I do not know the reason for this. However, if you disconnect the camera again, make it powerless and reconnect it, is it shown correctly again? 
What is the serial number of the camera? It is printed under the barcode on the lable.

Stefan

v-prgmr, 2021-02-02:
The serial number of the camera is 25910029, I checked it with another camera which is also the same model. When I now try to list the cameras on the network, one of the cameras gets detected while for the other one, I get the same error "Unable to retrieve device info ... " 

I will check it once again by removing the power and get back to you. However, what I always tried was to reboot the system. And now the camera gets detected and sometimes works with tcam-capture and sometimes tcam-capture freezes (most often)
.

TIS-Stefan, 2021-02-03:
You may run
tcam-gigetool list
and show me the output.

The tool is documented at https://www.theimagingsource.com/documentation/tiscamera/tcam-gigetool.html

TIS-Stefan, 2021-02-19:
Hello

Do you have any new information for me?

Stefan

v-prgmr, 2021-02-22:
Have been away from the hardware setup, will get back to you asap @TIS-Stefan

closed 2021-02-22 by TIS-Stefan