TheImagingSource/tiscamera #397: Installation problem on Ubuntu 20.04 & tone mapping function in linux SDK

hftsai, 2021-03-01:
Hi, Sorry i'm a novice in linux.
I have a system in ubuntu20.04 with anaconda3. 

I followed the guide on building tiscamera on github. 
However if "cd build" it didn't work. 

for  configuration
i did
`cmake -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DBUILD_LIBUSB=ON
`
then 
```
make
sudo make install.
```

at first it should error that 
` ImportError: No module named gi`

then later found it's due to installation path and the anaconda path not linked
i did
`ln -s /usr/lib/python3/dist-packages/gi /home/davince/anaconda3/lib/python3.8/site-packages/`

and it didn't have a problem with gi but i still cannot initiate tcam-capture, this error appears.
(base) davince@davince-Alienware-m15:~/Downloads/tiscamera$ tcam-capture
```
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 28, in <module>
    from tcam_capture.TcamDevice import TcamDevice, TcamDeviceIndex
  File "/home/davince/anaconda3/lib/python3.8/site-packages/tcam_capture/TcamDevice.py", line 23, in <module>
    from gi.repository import Tcam, Gst
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/importer.py", line 146, in load_module
    dynamic_module = load_overrides(introspection_module)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/overrides/__init__.py", line 118, in load_overrides
    override_mod = importlib.import_module(override_package_name)
  File "/home/davince/anaconda3/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/overrides/Gst.py", line 608, in <module>
    real_functions = [o for o in inspect.getmembers(Gst) if isinstance(o[1], type(Gst.init))]
  File "/home/davince/anaconda3/lib/python3.8/inspect.py", line 350, in getmembers
    value = getattr(object, key)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/module.py", line 163, in __getattr__
    setattr(wrapper, value_name, wrapper(value_info.get_value()))
ValueError: invalid enum value: 51202
```

I wonder if you can help me? or is ubuntu20.04 not supported. (my system wifi doesn't play well with 18.04)

Another question i wonder is the tone mapping function and the automatic white balancing available in the python sdk?

TIS-Stefan, 2021-03-01:
If "cd build" does not work, then you did not made "mkdir build" before.
Please follow the steps:

```
git clone https://github.com/TheImagingSource/tiscamera.git
cd tiscamera
sudo ./scripts/install-dependencies.sh --compilation --runtime
mkdir build
cd build
cmake -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DBUILD_LIBUSB=ON  ..
make
sudo make install
```
Stefan

hftsai, 2021-03-01:
ok i see. 
I didn't include the .. 

I have redone all the steps again. the error persists:
```
Traceback (most recent call last):
  File "/usr/bin/tcam-capture", line 28, in <module>
    from tcam_capture.TcamDevice import TcamDevice, TcamDeviceIndex
  File "/home/davince/anaconda3/lib/python3.8/site-packages/tcam_capture/TcamDevice.py", line 23, in <module>
    from gi.repository import Tcam, Gst
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/importer.py", line 146, in load_module
    dynamic_module = load_overrides(introspection_module)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/overrides/__init__.py", line 118, in load_overrides
    override_mod = importlib.import_module(override_package_name)
  File "/home/davince/anaconda3/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/overrides/Gst.py", line 608, in <module>
    real_functions = [o for o in inspect.getmembers(Gst) if isinstance(o[1], type(Gst.init))]
  File "/home/davince/anaconda3/lib/python3.8/inspect.py", line 350, in getmembers
    value = getattr(object, key)
  File "/home/davince/anaconda3/lib/python3.8/site-packages/gi/module.py", line 163, in __getattr__
    setattr(wrapper, value_name, wrapper(value_info.get_value()))
ValueError: invalid enum value: 51202

```

TIS-Stefan, 2021-03-01:
Hello

Please let me know, which camera model you use.
Also: Anaconda creates many problems. Please use a non Anaconda Python. 

Stefan

hftsai, 2021-03-01:
it's a DFK 37BUX178

it is an issue our software needs to build around anaconda python....
i'll try on standard python

TIS-Stefan, 2021-03-01:
Hello 
Please show the stack trace. 

Stefan

hftsai, 2021-03-02:
Sorry how do i do that?

hftsai, 2021-03-02:
i saw this here about USB3vision.
https://theimagingsource.deskpro.com/en/kb/articles/usb3vision-instead-of-v4l2-for-usb-cameras
should DFK37BUX178 installation follow this one as well?

hftsai, 2021-03-03:
On Xavier NX with the driver binary (0.13.1) it's working. 

Another question:
Is tone mapping available in linux with ARMv8 architecture?
it's not shown in the tcam-capture.
I wonder if it's available. it will be a very necessary function for us.

TIS-Stefan, 2021-03-03:
It seems, tcam-capture wont run in Anaconda environment.
The call stack is the trace back you posted above.
It is not related to python 3.8. It runs well on on all my Linux computers.

I did not found any hints to the gi/moduly.py error "ValueError: invalid enum value: 51202" in the internet. The error does not relate to our modules, but to Anaconday and gi. Therefore, you may ask the Anaconda programmers about this error and its meaning. 

The camera uses v4l2, not USB3Vision by default. 

You wrote "still the same error message." Which is that? If it is the same as you wrote above, then you are still using the Anaconda environment.

However, if you start Python in Anaconda (I never used that, therefore I might be not exact), what happens after you enter
```
import gi
from gi.repository import GLib, GObject, Gst, Tcam
gi.require_version("Gst", "1.0")
gi.require_version("Tcam", "0.1")
```
Please enter that line by line in Python and see, what happens.

Stefan

closed 2021-03-20 by hftsai