TheImagingSource/tiscamera #398: Best way to stream live video to browser?

andrewzhang35, 2021-03-02:
Hello,

I hope your day is going well. I am seeking some insight into how to stream live video from one of The Imaging Source cameras to a browser/web-app. Currently, I can view a live feed from the camera in a display sink using both the Python 03-live-stream and 08-save-stream examples found on the tiscamera GitHub repo (https://github.com/TheImagingSource/tiscamera/tree/master/examples/python).

I also have code set up in Node.js that displays a live video feed in a browser using WebRTC, however it currently just streams a feed from my webcam. Essentially, I am looking for advice on how to integrate the live video streaming from the camera locally with the video feed online, so that I can ultimately stream what the camera sees in a browser. Do you happen to have any suggestions on how to approach this, resources that may help, example code for a similar scenario, etc.? That would be greatly appreciated. Please feel free to ask any clarifying questions you may have.

Thanks in advance.

TIS-Stefan, 2021-03-03:
The problem of webrtc is, that it does not set a video format. (I do not know, why they miss that point.) Thus, it takes the first format from the camera, which usually is Y800. And can not handle it. The only solution is to limit the available video formats to RGB32 in the camera driver. 

Another nice way is writing a small Python webserver, which streams the images to the client. Many years ago there was a script available for that on github, but all these links have gone.

The following sample may gives a idea how to do that:
[web.zip](https://github.com/TheImagingSource/tiscamera/files/6074485/web.zip)
It is very old (from 2015) and I did not test that actually.

Stefan

andrewzhang35, 2021-03-03:
Hi,

Thank you very much for the sample, I'll take a look at it and get back to you if I have any questions.

On an unrelated note, I've noticed that in some of the examples you've provided in the past in GitHub issues and in some of the tiscamera examples in general, there is sometimes a "pipeline.get_state(Gst.CLOCK_TIME_NONE)" line that shows up. Would you mind explaining what this line does?

It is causing some issues when I try to run a program in a docker container, and through googling I have only found that Gst.CLOCK_TIME_NONE is a constant to define an undefined clock time. Is it a necessary line of code? I have not noticed any differences in the functionality of my program after removing it.

Thanks once again,
Andrew

TIS-Stefan, 2021-03-04:
Hello Andrew

For gst_state you may look at https://gstreamer.freedesktop.org/documentation/gstreamer/gstelement.html

Howveer CLOCK_TIME_NONE not a good parameter, because in case get_state does return, it waits infinite.  However, if that function does not return, then something in the pipeline went wrong.

I am not familiar with webrtc.

a gstreamer 

> pipeline to a web-app/online, d

I do not know, I would have to search in the internet for that

Stefan

andrewzhang35, 2021-03-10:
Hi Stefan,

Thanks so much for all the help you've given so far. I'm writing this message because I have an unrelated issue to what we had discussed earlier which I thought you might be able to help with.

Currently I'm trying to run an application for an Imaging Source GigE camera within a docker container. The application itself works fine, but one issue I have encountered is being able to connect to the camera/ethernet device from within the docker container. I'm trying to find a way to map the ethernet connection from the host device/computer to the docker container so that the container can access the camera (the camera is plugged into the computer). I'm not sure if you'll necessarily have an answer for me, but I thought it'd be worth a shot to ask if you happened to know any way of doing this.

Thanks once again,
Andrew

TIS-Stefan, 2021-03-11:
Hi Andrew

I am totally sorry, but I am a docker noob. I used that only once, without understanding, what I was doing. 
So you may search in the internet. 
But as a start, you may try to ping the camera from the docker. If you get an answer, then there should be a connection.

Stefan

closed 2021-07-28 by TIS-Stefan