TheImagingSource/tiscamera #399: PAUSED GST pipeline does not resume playing on >=10 secs pause

copterspace, 2021-03-26:
Dear friends, I test DMM 22BUC03-MLA on Raspberry PI 4 Raspbian. 
Tiscam repo version I use is 0.12.0 (because of https://github.com/TheImagingSource/tiscamera/issues/389)
For my application I need to pause and resume Gstreamer pipeline several times to pause frames acquisition.
I modified main cycle of https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py in the following manner (from line 125):

```python
    startingframe = 0
    try:
        while True:
            time.sleep(1)
            print('FPS = {}'.format(framecount - startingframe))
            startingframe = framecount 
            # add PAUSE
            pipeline.set_state(Gst.State.PAUSED)
            time.sleep(10)
            # resume playing
            print('resume playing...')
            pipeline.set_state(Gst.State.PLAYING)
    except KeyboardInterrupt:
        pass
    finally:
        pipeline.set_state(Gst.State.NULL)

```
The problem appears if pause length >=10 secs - gst player does not resume the frame acquisition properly:
```
Press Ctrl-C to stop.
FPS = 58 frame 57, Pixel Value=14 Timestamp=981722425
resume playing...
FPS = 68 frame 125, Pixel Value=10 Timestamp=1982971221
resume playing...
FPS = 10 frame 135, Pixel Value=11 Timestamp=2126382443
resume playing...
FPS = 0
resume playing...
FPS = 0
resume playing...
FPS = 0
```
Instead of PAUSED I also tried Gst.State.READY and Gst.State.VOID_PENDING - it did not help, same result.
The only workaround helped - pause pipeline with Gst.State.NULL - but in this case framerate of the first second pipeline playing degrades (58-60 instead of 70 fps), because new startup of the pipeline takes some time.

Here is full modified example code: [Tcam appsink_pause.zip](https://github.com/TheImagingSource/tiscamera/files/6209573/Tcam.appsink_pause.zip)

How to pause gstreamer pipeline properly for more then 10 secs?

If I specify pause 9 secs (`time.sleep(9)`) - it works OK:
```
Press Ctrl-C to stop.
FPS = 59 frame 58, Pixel Value=12 Timestamp=986407481
resume playing...
FPS = 69 frame 127, Pixel Value=9 Timestamp=19858345724
resume playing...
FPS = 69 frame 196, Pixel Value=12 Timestamp=2981905053
resume playing...
FPS = 71 frame 267, Pixel Value=11 Timestamp=3990136627
resume playing...
FPS = 69 frame 336, Pixel Value=10 Timestamp=4988812849
resume playing...
FPS = 70 frame 406, Pixel Value=11 Timestamp=5990881533
```

TIS-Stefan, 2021-03-26:
Hello

The main trick is setting the "sync" property of the appsink to "False":

``` Python
    sink.set_property("emit-signals", True)
    sink.set_property("max-buffers",5)
    sink.set_property("sync",False)
    sink.set_property("drop",1)
    user_data = "This is our user data"

    # tell appsink what function to call when it notifies us
    sink.connect("new-sample", callback, user_data)
```
The pipeline is:
``` Python
pipeline = Gst.parse_launch("tcambin  name=source"
                                " ! video/x-raw,format=GRAY8,width=640,height=480,framerate=60/1"
                                " ! appsink name=sink")
```

The main loop uses NULL state, but we get all wanted frames:
``` Python
        while True:
            time.sleep(1)
            print('FPS = {}'.format(framecount - startingframe))
            startingframe = framecount 
            # add PAUSE
            pipeline.set_state(Gst.State.NULL)
            time.sleep(20)
            # resume playing
            print('resume playing...')
            pipeline.set_state(Gst.State.PLAYING)
            pipeline.get_state(2000*1000*1000)
```
My output is:

```
Press Ctrl-C to stop.
PS = 61
resume playing...
FPS = 61
resume playing...
FPS = 61
resume playing...
FPS = 61
resume playing...
FPS = 61
resume playing...
FPS = 61
resume playing...
FPS = 61
```

This should solve your issue.

copterspace, 2021-03-30:
Yes, this helped. Even works with my original pipeline.
Stefan, thank you for fast reply!

I also tested it on 42BUC03 camera - it requires magick line `pipeline.get_state(2000*1000*1000)` - without it it breakes.
22BUC03 works OK even without this line.

TIS-Stefan, 2021-03-30:
It is not that magic. It is waiting for the pipeline changing the state from NULL / READY to playing with a two seconds timeout. If the timeout does not occur, you can be sure, that the pipeline has started correctly and you can start your image processing.
You can check the return values of gst_state() in order to be on the safe side.

Stefan

closed 2021-03-30 by TIS-Stefan