TheImagingSource/tiscamera #401: Cannot find camera after 3-5 hours

yaokmallorca, 2021-03-30:
Hello, 

We installed 3 cameras in Ubuntu 18.04, and the camera models were DFK 33UX178, DFK 33UX178, DFK AFUX178-M12 respectively. Our application needed to use three cameras at the same time. The application was based on QT5 C++. I followed the QT demo (https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/QT5%20Device%20Selection%20and%20Property%20Dialogs%20in%20C%2B%2B). After running 3-5 hours, we could not receive data from DFK 33UX178 model, and cannot find the camera using device listing. We needed to restart the software (delete class TcamCamera pointer), then camera could work well after restarting the application. After many times, DFK 33UX178 camera always had this problem, while other two cameras could work well. 

Then, we removed other two cameras (DFK 33UX178, DFK AFUX178-M12), and only used the QT demo for DFK 33UX178 camera (https://github.com/TheImagingSource/Linux-tiscamera-Programming-Samples/tree/master/QT5%20Device%20Selection%20and%20Property%20Dialogs%20in%20C%2B%2B). The same problem happened. We did not receive any print for this error. 

We obtained camera data using gst_app_sink_try_pull_sample function. When the problem happened, this function returned a null pointer. 

How do we solve the problem? Thank you for your helping.

TIS-Stefan, 2021-07-28:
Hello 
I am sorry for missing your issue.
I understand, you do not need to reconnect the camera, a software restart is fine. 
Does the camera work again, if you delete the tcamcamera object in your software and select the camera again?

It sounds for me, that the camera stream stops, but Gstreamer keeps on running. The sample you use has no bus message handler, so a device lost message is not received. Therefore, GStreamer keeps the device alive all the time and you need to restart the complete program.

Does the same happen, if you run the camera with gst-launch-1.0?

Do you get a dmesg output giving some more information?

Again, I am sorry for not handling your issue immediately.

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar