TheImagingSource/tiscamera #403: 42BUC does not give out correct FPS after startup on RPI

copterspace, 2021-04-06:
Dear friends, I continue test gstreamer paused pipeline I described in https://github.com/TheImagingSource/tiscamera/issues/399 .
Now I test DMM 42BUC03-MLA on Raspberry PI 4 Raspbian.
Tiscam repo version I use is 0.12.0 (because of https://github.com/TheImagingSource/tiscamera/issues/389)
For my application I need to pause and resume Gstreamer pipeline several times to pause frames acquisition.
For test I modified [07-appsink.py](https://github.com/TheImagingSource/tiscamera/blob/master/examples/python/07-appsink.py) , here is full code: 
[Tcam appsink_pause_repaired2.zip](https://github.com/TheImagingSource/tiscamera/files/6262417/Tcam.appsink_pause_repaired2.zip)

With 42BUC camera I met new issue: it does not give proper FPS after Raspberry startup.
I set Exposure Auto and Gain auto = False, Exposure = 13000, Gain = 34 , and expect camera will give me 25 FPS (as it gives on IC Capture, for example), but after RPI startup I get only 19-20 fps:
```
FPS = 21 frame 20, Pixel Value=1 Timestamp=2187385868
resume playing...
FPS = 19 frame 39, Pixel Value=1 Timestamp=3137742478
resume playing...
FPS = 19 frame 58, Pixel Value=0 Timestamp=4176930052
resume playing...
FPS = 19 frame 77, Pixel Value=1 Timestamp=5127237274
resume playing...
FPS = 20 frame 97, Pixel Value=1 Timestamp=6166486606
resume playing...
```

Then I simply unplug and replug camera to USB 2.0 port - it starts to give out 38-39 FPS , more then expected, but it is OK:
```
FPS = 39 frame 38, Pixel Value=0 Timestamp=1996329183
resume playing...
FPS = 38 frame 76, Pixel Value=0 Timestamp=2991120961
resume playing...
FPS = 38Captured frame 115, Pixel Value=0 Timestamp=4019235201
resume playing...
FPS = 39 frame 153, Pixel Value=0 Timestamp=5014084720
resume playing...
FPS = 38 frame 191, Pixel Value=0 Timestamp=6008930738
```
Only physical replug helps, `sudo reboot now`, for example, does not help.
For my application I'd like to have camera 42BUC work correctly right after startup, not to replug it each time after I power up the device.
With 22BUC I did not have such problem.
Camera: DMM 42BUC03-MLA V4L2    10120024
For work with RPI I updated camera firmware as described [here](https://www.theimagingsource.com/documentation/tiscamera/firmware-update.html).

TIS-Edgar, 2021-04-08:
Hi,

The applications should use a `appsink name=sink sync=false` to ensure the gstreamer clock does not interfere with the measurements.

To questions:
1. Does the issue exist when using tcam-capture?
2. Does the issue exist when using qv4l2? You will get a fps count at the bottom of the main window once you start a live stream.

qv4l2 is a good test application to verify that there is no underlying issue in the kernel/hardware/etc.

copterspace, 2021-04-09:
I use [headless PRI setup](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip), neither tcam-capture nor qv4l2 works.

But the problem persists with `gst-launch-1.0 -v tcambin ! videoconvert ! fpsdisplaysink text-overlay=0 video-sink=fakesink` , so at least it is not python-related...

TIS-Stefan, 2021-04-12:
I can reproduce the issue using v4l2 on a Pi 4.
The very same situation gives correct results on a TX Nano. 

Thus I conclude, the Pi 4 starts has an issue with USB controller start. 

I am in contact with our firmware engineers.

Stefan

TIS-Stefan, 2021-04-21:
I am very sorry, but currently I can not present a solution. I am aware, that this is not helpful, but I ask for progress each day.

copterspace, 2021-04-26:
I've found some workaround: instead of physically replug USB camera I can power off and then back on RPI4 USB port with [uhubctl](https://github.com/mvp/uhubctl):
```
sudo uhubctl -l 1-1 -a off
sudo uhubctl -l 1-1 -a on
```
After this, `gst-launch-1.0 -v tcambin ! videoconvert ! fpsdisplaysink video-sink=fakesink text-overlay=0` gives out correct fps.
It ain't good solution for production - but may help to debug further this problem...

TIS-Stefan, 2021-04-26:
It seems, the Pi 4 starts with USB 1.1 speed.  Thank you for the hint.

I sent a new firmware of DFK 42 to you this morning.

Stefan

TIS-Stefan, 2021-07-28:
If I remember correctly, we sent a new firmware

closed 2021-07-28 by TIS-Stefan