TheImagingSource/tiscamera #404: Reverse video

yaokmallorca, 2021-04-14:
Hello. 

Our application needs to reverse video online. 
We have 3 cameras, and every camera has different properties. 
More precise, DFK AFUX178-M12 only has "Reverse X", 
DFX 33UX178 has "Reverse X" and "Reverse Y", and DFX 33UX183 has "Reverse Y".
However, we need to reverse X and Y for all three cameras online. 

Alternatively, we can use "gstreamer-videoflip", but we need to restart the camera.
How can I solve this problem? Thanks.

yaokmallorca, 2021-04-19:
Nice, no response. 
I reverse the video by software. This is an example of flipping along Y. 
```
void 
TcamCamera::reverse_y() {
    GstElement *video_flip = gst_bin_get_by_name (GST_BIN (displaybin_), "video-flip");
    if (video_flip == NULL) {
        return;
    }
    gint method;
    g_object_get (G_OBJECT (video_flip), "method", &method, NULL);
    if (method == 2)
        g_object_set (G_OBJECT (video_flip), "method", 0, NULL);
    else 
        g_object_set (G_OBJECT (video_flip), "method", 2, NULL);
    gst_object_unref (video_flip);
}
```
It is strange for me that some cameras can flip X **and** Y, and some only can flip X **or** Y.
For example, this is the output of "list_properties.py".
It is clear DFK 33UX183 only can reverse Y,  DFK AFUX178-M12 only can reverse X, and DFK 33UX178 can reverse X and Y.
What are your reasons for doing this?  You haven't finished the drive yet?

```
Available devices:
0 - Use default device
1 - 03020148 - DFK 33UX183
Brightness(integer) value: 300 default: 50 min: 0 max: 1023 grouping: Exposure - Brightness
Saturation(integer) value: 64 default: 64 min: 0 max: 255 grouping: Color - Saturation
Hue(integer) value: 0 default: 0 min: -180 max: 180 grouping: Color - Hue
Whitebalance Auto(boolean) value: True default: True grouping: Color - Whitebalance
Whitebalance Red(integer) value: 121 default: 64 min: 0 max: 255 grouping: Color - Whitebalance
Whitebalance Blue(integer) value: 107 default: 64 min: 0 max: 255 grouping: Color - Whitebalance
Gamma(integer) value: 100 default: 100 min: 1 max: 500 grouping: Image - Gamma
Gain(integer) value: 0 default: 0 min: 0 max: 1957 grouping: Exposure - Gain
Exposure(integer) value: 55555 default: 333 min: 50 max: 30000000 grouping: Exposure - Exposure
Exposure Auto(boolean) value: True default: True grouping: Exposure - Exposure
Exposure Auto Reference(integer) value: 128 default: 128 min: 0 max: 255 grouping: Exposure - Exposure
Gain Auto(boolean) value: True default: True grouping: Exposure - Gain
Trigger Mode(boolean) value: False default: False grouping: Special - Trigger Mode
Software Trigger(button) grouping is Special -  Trigger Mode
Trigger Delay (us)(integer) value: 31 default: 0 min: 0 max: 1000000 grouping: Special - Trigger Mode
Strobe Enable(boolean) value: False default: False grouping: Special - Strobe Enable
Strobe Polarity(boolean) value: False default: False grouping: Special - Strobe Enable
Strobe Exposure(boolean) value: True default: True grouping: Special - Strobe Enable
Strobe Duration(integer) value: 100 default: 100 min: 0 max: 32767 grouping: Special - Strobe Enable
Strobe Delay(integer) value: 0 default: 0 min: 0 max: 32767 grouping: Special - Strobe Enable
GPOut(integer) value: 0 default: 0 min: 0 max: 1 grouping: Special - GPIO
GPIO(integer) value: 0 default: 0 min: 0 max: 1 grouping: Special - GPIO
Offset X(integer) value: 0 default: 0 min: 0 max: 5216 grouping: Partial Scan - Offset Auto Center
Offset Y(integer) value: 0 default: 0 min: 0 max: 3644 grouping: Partial Scan - Offset Auto Center
Offset Auto Center(boolean) value: True default: True grouping: Partial Scan - Offset Auto Center
Trigger Polarity(enum) value: FallingEdge default: RisingEdge grouping Special - Trigger Mode
Entries: 
	 FallingEdge
	 RisingEdge
Trigger Operation(enum) value: Default default: Default grouping Special - Trigger Mode
Entries: 
	 Default
	 GlobalResetRelease
Trigger Burst Count(integer) value: 1 default: 1 min: 1 max: 1000 grouping: Special - Trigger Mode
Trigger Debounce Time (us)(integer) value: 0 default: 0 min: 0 max: 1000000 grouping: Special - Trigger Mode
Trigger Mask Time (us)(integer) value: 0 default: 0 min: 0 max: 1000000 grouping: Special - Trigger Mode
Trigger Noise Surpression Time (us)(integer) value: 0 default: 0 min: 0 max: 1000000 grouping: Special - Trigger Mode
Auto Functions ROI Control(boolean) value: True default: True grouping: Special - Auto Functions ROI Control
Auto Functions ROI Left(integer) value: 1368 default: 0 min: 0 max: 5456 grouping: Special - Auto Functions ROI Control
Auto Functions ROI Top(integer) value: 912 default: 0 min: 0 max: 3632 grouping: Special - Auto Functions ROI Control
Auto Functions ROI Width(integer) value: 2736 default: 0 min: 16 max: 5472 grouping: Special - Auto Functions ROI Control
Auto Functions ROI Height(integer) value: 1824 default: 0 min: 16 max: 3648 grouping: Special - Auto Functions ROI Control
Whitebalance(boolean) value: False default: False grouping: Color - Whitebalance
[... 118 lines cut ...]

```
0 - Use default device
1 - 4910292 - DFK AFUX178-M12
Brightness(integer) value: 300 default: 60 min: 0 max: 4095 grouping: Exposure - Brightness
Gain(integer) value: 70 default: 0 min: 0 max: 480 grouping: Exposure - Gain
Focus(integer) value: 312 default: 500 min: 0 max: 1000 grouping: Lens - Focus
Exposure(integer) value: 50000 default: 33333 min: 15 max: 30000000 grouping: Exposure - Exposure
Gain (dB/100)(integer) value: 0 default: 0 min: 0 max: 4800 grouping: Unknown - INVALID_PORPERTY
Trigger Mode(boolean) value: False default: False grouping: Special - Trigger Mode
Software Trigger(button) grouping is Special -  Trigger Mode
Trigger Delay (us)(integer) value: 150 default: 0 min: 0 max: 10000000 grouping: Special - Trigger Mode
Strobe Enable(boolean) value: False default: False grouping: Special - Strobe Enable
Strobe Polarity(boolean) value: False default: False grouping: Special - Strobe Enable
[... cut ...]
GPIO(integer) value: 0 default: 0 min: 0 max: 1 grouping: Special - GPIO
Offset X(integer) value: 0 default: 0 min: 0 max: 2688 grouping: Partial Scan - Offset Auto Center
Offset Y(integer) value: 0 default: 0 min: 0 max: 1728 grouping: Partial Scan - Offset Auto Center
Offset Auto Center(boolean) value: True default: True grouping: Partial Scan - Offset Auto Center
**Reverse X(boolean) value: False default: False grouping: Unknown - INVALID_PORPERTY**
Override Scanning Mode(integer) value: 0 default: 0 min: 0 max: 2 grouping: Unknown - INVALID_PORPERTY
```

TIS-Stefan, 2021-04-19:
Hello

> Nice, no response.
I am sorry,  I have a lot of workload and also we had two days weekend. 

> DFX 33UX178 has "Reverse X" and "Reverse Y", and DFX 33UX183 has "Reverse Y".
> However, we need to reverse X and Y online for all cameras.
> 
> Alternatively, we can use "gstreamer-videoflip", but we need to restart the camera.
> How can I solve this problem? Thanks.

The camera use different hardware and sensors with different capabilities. Therefore, the properties, in particular the Reverse ones, are different. I am sorry, but this can not be changed.

Regarding the videoflip module, I would expect, that GST_VIDEO_ORIENTATION_HORIZ and GST_VIDEO_ORIENTATION_VERT would not need a restart, since the memory of the GstBuffer must not be reorganized. But I think, setting video-direction uses one function for all possible values, therefore a restart is necessary.
I suppose, you tried that already on your own. 
I am afraid, I have no solution for your issue, except, I do not dare to say, writing your own flipping filter. 

Stefan

yaokmallorca, 2021-04-20:
Hi, @TIS-Stefan.
Thanks for your reply. I understand your considerations. 
Currently, the camera works well. We need to do more testing. 
Anyway, thank you for your help.

TIS-Stefan, 2021-04-20:
I just tried on my own a simple C++ sample:

```
int main(int argc, char **argv)
{
    gst_init(&argc, &argv);

    cv::Mat OpenCVImage;
    // Initialize our TcamCamera object "cam" with the serial number
    // of the camera, which is to be used in this program.
    std::cout << "Start";
    TcamImage cam("38910083");
    //TcamCamera cam("00001234");
    
    // Set a color video format, resolution and frame rate
    cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{30,1});

    // Comment following line, if no live video display is wanted.
    cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    
    // Start the camera
    cam.start();
    std::cout << "Live started";

    sleep(1);
    printf("Start Snap\n");
    // Snap an Image with 60 ms timeout. Should be set accordingly to the
    // frame rate.
    if( cam.snapImage(60) )
    {
        // On succes do something with the image data. Here we create
        // a cv::Mat and save the image
        OpenCVImage.create( cam.getHeight(),cam.getWidth(),CV_8UC(cam.getBytesPerPixel()));
        memcpy( OpenCVImage.data, cam.getImageData(), cam.getImageDataSize());
        cv::imwrite("test.jpg",OpenCVImage);
    }
    else
    {
        printf("Timeout at snapImage()\n");
    }

    printf("Press enter key to end program.");
    // Simple implementation of "getch()", wait for enter key.
    char dummyvalue[10];
    scanf("%c",dummyvalue);

    cam.stop();
    std::cout << "Live ended\n";

    return 0;
}
```

I get all output to std::cout. So there must be something different between my and your computer. It it was UTF8, then the outputs before the stream is started, should not show.  Therefore, I am clueless.

Stefan

closed 2021-07-28 by TIS-Stefan