TheImagingSource/tiscamera #405: tiscamera does block std::cout prints

mbcel, 2021-04-18:
I am experiencing a strange behaviour when using the tiscamera library.

When I start the camera via `TcamCamera::start() ` and after setting exposure properties somehow no `std::cout` prints are shown anymore. Using `printf()` does still work. This is really strange and I am not really sure what is going on. Furthermor just before the cout's are not working anymore this is printed:

Cam id: Exposure Time (us) �U   ��
�U   0             M               �
�U   	                                  p��U                                     �~�U                                     ��
�U      �                          �Z
�U      �                                                          	       ��U   ���U           !          U   �	�U       U   `�	�U       U    �	�U                           �                                                                                                                                                         1       ��>|   �X
�U   �\
�U                  !       src     �a
�U     de    1      h��v         ȶ�v   ��
�U   ��
�U          	 �Trigger Mode                                                       	 �        

Does anyone know what is going on? I used version 0.11 and 0.13 the same happens in both version. 

On stackoverflow [(LINK)](https://stackoverflow.com/questions/11173422/cout-no-output-but-printf-does) I read that apparently when a string is set to 0 this behaviour can happen. Is this happening anywhere? Unfortunately, I cannot find the line/file where the above output is generated...

TIS-Stefan, 2021-04-19:
Hello
It depends on the thread, in which the cout and printf is done. If it is in the callback, then you get the output after your program has ended.
However, you may try std::flush() after your std::cout call.

mbcel, 2021-04-19:
No this happens on all threads. No std::cout is working in the callback nor on the main thread that started the camera. Printf is working as expected in both threads. There is also no std::cout print after the program has finished, instead they are completely ignored/vanished. Std::flush() also does not change anything.

TIS-Stefan, 2021-04-20:
Hello
Which kind of text do you write? I can not read anything of your output, because you may use a different font.  (I do not have all fonts installed on my current computer)

Does the output work, before the live stream is started?

mbcel, 2021-04-20:
This ouput is copied from my CLion editor. This is configured to use "utf-8". However the same output is generated in a linux terminal. I am also not sure what this text is supposed to be. It is also not generated by me but instead from some underlying library from tcamcamera's Property->set(), I guess. I did a search but could not find any file where this "Cam id: ..." comes from. 

Everything is output normal before the stream is started. And if I do not start the camera stream a parallel thread loop outputs everything fine. However, when the stream is started this strange output comes from above (from some underlying library I assume) and after that no thread does print any std::cout prints anymore.

TIS-Stefan, 2021-04-20:
Hello
I tried that code:

```
int main(int argc, char **argv)
{
    gst_init(&argc, &argv);

    cv::Mat OpenCVImage;
    // Initialize our TcamCamera object "cam" with the serial number
    // of the camera, which is to be used in this program.
    std::cout << "Start";
    TcamImage cam("38910083");
    //TcamCamera cam("00001234");
    
    // Set a color video format, resolution and frame rate
    cam.set_capture_format("BGRx", FrameSize{640,480}, FrameRate{30,1});

    // Comment following line, if no live video display is wanted.
    cam.enable_video_display(gst_element_factory_make("ximagesink", NULL));
    
    // Start the camera
    cam.start();
    std::cout << "Live started";

    sleep(1);
    printf("Start Snap\n");
    // Snap an Image with 60 ms timeout. Should be set accordingly to the
    // frame rate.
    if( cam.snapImage(60) )
    {
        // On succes do something with the image data. Here we create
        // a cv::Mat and save the image
        OpenCVImage.create( cam.getHeight(),cam.getWidth(),CV_8UC(cam.getBytesPerPixel()));
        memcpy( OpenCVImage.data, cam.getImageData(), cam.getImageDataSize());
        cv::imwrite("test.jpg",OpenCVImage);
    }
    else
    {
        printf("Timeout at snapImage()\n");
    }

    printf("Press enter key to end program.");
    // Simple implementation of "getch()", wait for enter key.
    char dummyvalue[10];
    scanf("%c",dummyvalue);

    cam.stop();
    std::cout << "Live ended\n";

    return 0;
}
```
I get all outputs to std::cout.
Does this code above shows, how you use the std::cout?

Stefan

TIS-Edgar, 2024-04-12:
Closed due to inactivity.

closed 2024-04-12 by TIS-Edgar