TheImagingSource/tiscamera #406: can't detect my DMK Z12GP031 after installation and

chang9148, 2021-04-28:
Hallo,

I try to install the tiscamera on my Jetson Xavier NX on Ubuntu 18.04 for my DMK Z12GP031 camera. Buti met the issue ,that  the "tcam-capture" software can't detect my camera.  

**First The cmake process are normal**

```
student@nvidiajetson:~/Downloads/tiscamera/build$ sudo cmake -DBUILD_ARAVIS=ON -DBUILD_GST_1_0=ON -DBUILD_TOOLS=ON -DBUILD_V4L2=ON -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found GObject: /usr/include/glib-2.0  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found PYTHON_MODULE_PyQt5: /usr/lib/python3/dist-packages/PyQt5  
-- Found PYTHON_MODULE_setuptools: /usr/lib/python3/dist-packages/setuptools  
-- 
-- Build gstreamer-1.0 plugins:   ON
-- Support for GigE via aravis:   ON
-- Support for USB cameras:       ON
-- Support for LibUsb cameras:    ON
-- Build additional utilities:    ON
-- Build documentation            OFF
-- Build tests                    OFF
-- 
-- 
-- Installation prefix:           /usr
-- Installing binaries to:        /usr/bin
-- Installing libraries to:       /usr/lib
-- Installing header to:          /usr/include
-- Installing udev rules to:      /etc/udev/rules.d
-- Installing systemd units to:   /lib/systemd/system
-- Installing uvcdynctrl rules to:/usr/share/uvcdynctrl/data/199e
-- Installing desktop files to:   /usr/share/applications
-- Installing data files to:      /usr/share/theimagingsource/tiscamera
-- Installing gstreamer-1.0 to:   /usr/lib/aarch64-linux-gnu/gstreamer-1.0
-- Installing gobject-introspection to: /usr/share/gir-1.0
-- Installing introspection typelib to: /usr/lib/aarch64-linux-gnu/girepository-1.0
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/student/Downloads/tiscamera/build
```

**Second** ,_by make process only the follow  warning_ 
```
`
student@nvidiajetson:~/Downloads/tiscamera/build$ sudo make

[  4%] Performing build step for 'aravis'
Making all in src
  GEN      arvenumtypes.h
  GEN      arvenumtypes.c
**WARNING**: No file "arvenumtypes.h" found.
**WARNING**: No file "arvenumtypes.h" found.
  CC       arvenums.lo
  CC       arvdebug.lo
>>>
[ 34%] Generating Tcam-0.1.gir
g-ir-scanner: link: cc -o /home/student/Downloads/tiscamera/src/gobject/tmp-introspectndcyqeel/Tcam-0.1 /home/student/Downloads/tiscamera/src/gobject/tmp-introspectndcyqeel/Tcam-0.1.o -L. -Wl,-rpath,. -Wl,--no-as-needed -ltcamprop -L/home/student/Downloads/tiscamera/build/src/gobject -Wl,-rpath,/home/student/Downloads/tiscamera/build/src/gobject -lgio-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lgobject-2.0 -lglib-2.0
tcamprop.c:147: **Warning**: Tcam: tcam_prop_get_tcam_menu_entries: unknown parameter 'name' in documentation comment, should be 'menu_name'
[ 34%] Generating Tcam-0.1.typelib
```

**Then There is no further warning.**

**Third** ,after i 100% install tiscamera , i try the gige-daemon,but it is not work as follow.

`
```
student@nvidiajetson:~/Downloads/tiscamera/build$ sudo systemctl daemon-reload     
student@nvidiajetson:~/Downloads/tiscamera/build$ sudo systemctl enable gige-daemon.service 
Failed to enable unit: Unit file gige-daemon.service does not exist.
student@nvidiajetson:~/Downloads/tiscamera/build$ sudo systemctl start gige-daemon.service 
Failed to start gige-daemon.service: Unit gige-daemon.service not found.
`
But at the installation process , i have found the gige-daemon as follow.
`Scanning dependencies of target gige-daemon
[ 96%] Building CXX object tools/gige-daemon/CMakeFiles/gige-daemon.dir/main.cpp.o
[ 97%] Building CXX object tools/gige-daemon/CMakeFiles/gige-daemon.dir/CameraListHolder.cpp.o
[ 97%] Building CXX object tools/gige-daemon/CMakeFiles/gige-daemon.dir/DaemonClass.cpp.o
[ 98%] Linking CXX executable gige-daemon
[ 98%] Built target gige-daemon
Scanning dependencies of target tcam_gigewrapper
```

**For further Attempt with** `gst-launch-1.0 tcambin ! videoconvert ! ximagesink` It return the follow answer:

```
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...
```
For ifconfig command 

```
`eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 48:b0:2d:15:d7:62  txqueuelen 1000  (Ethernet)
        RX packets 15  bytes 2046 (2.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 567  bytes 103371 (103.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 37  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 2274  bytes 162437 (162.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2274  bytes 162437 (162.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rndis0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b2:01:94:70:ee:f1  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b2:01:94:70:ee:f3  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.31.74  netmask 255.255.255.0  broadcast 192.168.31.255
        inet6 fe80::b516:c10f:d99a:fb81  prefixlen 64  scopeid 0x20<link>
        ether d8:c0:a6:50:e7:9d  txqueuelen 1000  (Ethernet)
        RX packets 37525  bytes 46437723 (46.4 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 17327  bytes 4079283 (4.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

How can i solve this issue? 

Look forward to your answer.

TIS-Stefan, 2021-04-28:
I already answered to your email, therefore I close the case here.
It is always OK to write an email  or post here. You do not need to make a double request.

Stefan

chang9148, 2021-04-28:
Hallo Stefan，

The answer is for my early E-Mail  , which  I have finished already . The cmake parameter „avaris“ is set to “ON”as you Email and then I can all install rightly . The question on GitHub is different, which is occurred after install. The give-daemon is no work. The gst-Launch is not work too.For this one , i really don’t have idea. The details are in #406.

Look forward to your answer
Your sincerely
Tao

from Tao Chang

Closed #406<https://github.com/TheImagingSource/tiscamera/issues/406>.

TIS-Stefan, 2021-04-28:
Is the camera powered? Does the green LED on the RJ45 connector light constantly, while the yellow one blinks? If not, then you need either a Power Supply, a PoE Injector or a PoE Switch.
Then you must check the IP Adresses too. You may have a look at the document at https://s1-dl.theimagingsource.com/api/2.5/packages/publications/whitepapers-cameras/wpgigecamerastartup/4df6c9cb-9fb5-50d0-8b42-82259fd2124b/wpgigecamerastartup_1.0.en_US.pdf. There are many hints and trouble shooting. I hope, this will give an idea. If not. please continue asking!

Stefan

TIS-Stefan, 2021-04-30:
After rebuilding tiscamera the issue is sorted out.
Stefan

closed 2021-04-30 by TIS-Stefan