TheImagingSource/tiscamera #407: Can't diable "Exposure Auto" and "Gain Auto" on DFK 23G618 in Python

kanawat, 2021-04-28:
Hello,
I am using DFK 23G618 (GigE) camera with python3 on Ubuntu 20.04 LTS.
I am able to run the tcam-capture example fine, with the "Exposure Auto" and "Gain Auto" check box to enable/diable the features.

However, when I ran this peice of code in Python3 , expecting the gain and exposure to be fixed.  It didn't.  The brightness of the video keep oscillating.  

```
def main():
    Gst.init(sys.argv)  # init gstreamer
    serial = None
    pipeline = Gst.parse_launch("tcambinname=bin"
                                " ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1"
                                " ! videoconvert"
                                " ! ximagesink")

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")

    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)

    #print(camera.get_tcam_property("Exposure Auto"))
    camera.set_tcam_property("Exposure", 31451)
    camera.set_tcam_property("Gain", 16.04)
    camera.set_tcam_property("Exposure Auto", False)
    camera.set_tcam_property("Gain Auto", False)
    
    pipeline.set_state(Gst.State.PLAYING)
    print("Press Ctrl-C to stop.")

    # We wait with this thread until a
    # KeyboardInterrupt in the form of a Ctrl-C
    # arrives. This will cause the pipline
    # to be set to state NULL
    try:
        while True:
            time.sleep(1)
            
    except KeyboardInterrupt:
        pass
    finally:
        pipeline.set_state(Gst.State.NULL)

```
Here are the list of what I tried:
1. Changing from `camera.set_tcam_property("Exposure Auto", False)` to `camera.set_tcam_property("Exposure Auto", "Off")` like what suggested in the other post had no effect.
2. I listed the property of the camera with tcam-ctrl , no parameter named "Exposure Auto" and "Gain Auto" can be found.
```
Exposure                                (int) min=10 max=30000000 step=1 default=31451 value=31451 category=Exposure group=Exposure
Trigger Selector                        (enum)  default=FrameStart category=Special group=Trigger Mode
						value=FrameStart
							 FrameStart
Trigger Mode                            (enum)  default=Off    category=Special group=Trigger Mode
						value=Off
							 Off
							 On
Trigger Source                          (enum)  default=Any    category=Special group=Trigger Mode
						value=Any
							 Any
							 Line1
							 Software
Trigger Activation                      (enum)  default=RisingEdge category=Special group=Trigger Mode
						value=RisingEdge
							 FallingEdge
							 RisingEdge
Software Trigger                        (button) category=Special group=Trigger Mode
Gain                                    (double) min=0 max=36.6234 step=0.001 default=16.0384 value=16.0384 category=Exposure group=Gain
Balance Ratio Selector                  (enum)  default=Red    category=Color group=Whitebalance
						value=Red
							 Blue
							 Green
							 Red
BalanceRatioRaw                         (int) min=0 max=255 step=1 default=64 value=64 category=Color group=INVALID_PORPERTY
GPIO                                    (int) min=0 max=3 step=1 default=2 value=2 category=Special group=GPIO
GPIn                                    (int) min=0 max=1 step=1 default=1 value=1 category=Special group=GPIO
GPOut                                   (int) min=0 max=1 step=1 default=0 value=0 category=Special group=GPIO
Strobe Enable                           (enum)  default=Off    category=Special group=Strobe Enable
						value=Off
							 Off
							 On
Strobe Polarity                         (enum)  default=ActiveLow category=Special group=Strobe Enable
						value=ActiveLow
							 ActiveHigh
							 ActiveLow
Strobe Operation                        (enum)  default=Exposure category=Special group=Strobe Enable
						value=Exposure
							 Exposure
							 FixedDuration
Strobe Duration                         (int) min=0 max=10000 step=1 default=100 value=100 category=Special group=Strobe Enable
Strobe Delay                            (int) min=-10000 max=10000 step=1 default=0 value=0 category=Special group=Strobe Enable

```
3. I can disable the property using these commands:
```
tcam-ctrl -p -s "Gain Auto=false" <SERIAL NO.>
tcam-ctrl -p -s "Exposure Auto=false" <SERIAL NO.>
```
4. Running example 01-list-properties.py also doesn't show parameter named "Exposure Auto" and "Gain Auto".
5. Running example 02-set-properties.py gives this error:
```
Traceback (most recent call last):
  File "02-set-properties.py", line 104, in <module>
    sys.exit(main())
  File "02-set-properties.py", line 84, in main
    print_properties(camera)
  File "02-set-properties.py", line 38, in print_properties
    category, group) = camera.get_tcam_property("Exposure Auto")
TypeError: unknown type (null)

```

Any idea how can I disable Exposure Auto and Gain Auto in python?

TIS-Stefan, 2021-04-28:
As shown in the property output, the automatics are string / enum values. So you use
camera.set_tcam_property("Exposure Auto", "Off")
camera.set_tcam_property("Gain Auto", "Off")

I am annoyed about the USB and GigE interfaces boing different too. USB is boolean, GigE is string. Maybe in a future version this will be unifyed.

I hope, this helps

Stefan

kanawat, 2021-04-28:
Thank you for the quick response.  But using "Off" instead of False doesn't work.  I also didn't see these 2 properties in the property output.  It seems like in Python, the "Exposure Auto" and "Gain Auto" are inaccessible.

Kanawat

kanawat, 2021-04-29:
As I posted in my 1st post, I am unable to use Python script to list "Exposure Auto" and "Gain Auto", I can see all other properties but not these two.  When executing  `camera.get_tcam_property("Exposure Auto")`, it gives the following error (this is the result from running the included example python script).

```
Traceback (most recent call last):
  File "02-set-properties.py", line 104, in <module>
    sys.exit(main())
  File "02-set-properties.py", line 84, in main
    print_properties(camera)
  File "02-set-properties.py", line 38, in print_properties
    category, group) = camera.get_tcam_property("Exposure Auto")
TypeError: unknown type (null)
```

Kanawat.

TIS-Stefan, 2021-04-29:
Hello

Please, can you run the list properties Python sample?

Stefan

kanawat, 2021-04-30:
Hello Stefan,

Nevermind the confustions.  I know that the camera model is probably too old to be working smoothly with recent software.  But your last advice did the trick! -- Thank you!

After setting camera into PLAYING state and list out the properties, I can see several more parameters, including the automatic gain & exposure.  

Just in case anybody experiencing the same problem, here's the working code --- the key is to set the automatic properties ONLY AFTER the pipeline is set into PLAYING state:

```
def main():
    Gst.init(sys.argv)  # init gstreamer

    serial = None

    pipeline = Gst.parse_launch("tcambin name=bin"
                                " ! video/x-raw,format=BGRx,width=640,height=480,framerate=15/1"
                                " ! videoconvert"
                                " ! ximagesink")

    # retrieve the bin element from the pipeline
    camera = pipeline.get_by_name("bin")

    # serial is defined, thus make the source open that device
    if serial is not None:
        camera.set_property("serial", serial)

    pipeline.set_state(Gst.State.PLAYING)

    print("Press Ctrl-C to stop.")
    time.sleep(1)

    # with 23G camera model Automactics parameters are in software, camera must be in PLAYING state for software to be able to set these automatics
    camera.set_tcam_property("Exposure", 31451)
    camera.set_tcam_property("Gain", 16.04)
    camera.set_tcam_property("Exposure Auto", False)
    camera.set_tcam_property("Gain Auto", False)

    print_properties(camera) # this function is copied from 01-list-properties.py example

    # We wait with this thread until a
    # KeyboardInterrupt in the form of a Ctrl-C
    # arrives. This will cause the pipline
    # to be set to state NULL
    try:
        while True:
            time.sleep(1)
            
    except KeyboardInterrupt:
        pass
    finally:
        pipeline.set_state(Gst.State.NULL)
```

closed 2021-04-30 by kanawat

TIS-Stefan, 2021-04-30:
Hello

Thank you for your code!

Stefan