TheImagingSource/tiscamera #409: Tcam-Capture error: displaysink: could not initilaize xv output

mbcel, 2021-05-10:
I just installed the tiscamera repo. However I am not getting any camera live view with the tcam-capture program. No output stream frame rates are selectable/shown in the gui but the camera itself is recognized. 

Running the gui in a terminal outputs the error: 

`Tcamview.py:509 : - ERROR: displaysink : Could not initialise Xv output`

TIS-Stefan, 2021-05-10:
Hello 

it seems, you use  ssh forwarding, which has no graphic board hardware support.
My suggestion: Search in the tcam-capture source for "xvimagesink" and replace it with "ximagesink". (Without "v".) The latter one does not use graphic boards hardware and therefore works with SSH.

Which camera model do you use?
Which computer model to you use?

Stefan

mbcel, 2021-05-11:
I installed the tiscamera lib through ssh but when running I do not use ssh and instead run it on the computer itself. It's a amd64 based laptop running Ubuntu 18.04 and also equipped with an nvidia graphics card. Futhermore I blacklisted the nouveau driver since else I am not able to boot into Ubuntu. Could this be the reason? 

Camera is: Model: DFM 37UX273-ML Serial: xxxxxx Type: v4l2

mbcel, 2021-07-15:
Yep, it does work. However I had to remove some properties that only exist in the xvimagesink.

TIS-Stefan, 2021-07-15:
Very well!

closed 2021-07-15 by TIS-Stefan