TheImagingSource/tiscamera #410: IMX Low-Latency Mode

dzhura, 2021-05-31:
Is it possible to read and/or set IMX Low-Latency Mode for DFK 38UX267 in Linux? I found that /tiscamera/data/uvc-extensions/usb37.xml contains this field, so my guess is that it is possible with DFK 37U cameras.

TIS-Stefan, 2021-05-31:
Yes

(Short answer... ) 

You may check with tcam-capture.

Stefan

dzhura, 2021-05-31:
Hi Stefan,

I checked all the properties, including the special tab, in tcam-capture, but did not find this option. Where this option is supposed to be? 

BTW, tcam-ctrl --save does not list the option. I am using 0.13.1 version of tiscamera.

Maxim

TIS-Stefan, 2021-06-01:
You are right, it is missing in the UVC Extension for 38U cameras. I asked engineers to add this. 
You could compile tiscamera with
cmake -DTCAM_ARAVIS_USB_VISION=ON -DBUILD_ARAVIS=ON ..
and access the camera by the USB3Vision interface by passing the serialnumber as "nnnnnnnn-aravis" to the tcambin / tcamsrc. You can select the camera in tcam-capture accordingly.  Doing so, you will see all available camera properties. 

Stefan

dzhura, 2021-06-02:
Thanks Stefan. May I ask to add Trigger overlap option to UVC, please? Or should I create another issue for this?

I follow the build instructions and use the definitions you suggested, but build fails at 
[  0%] Performing configure step for 'aravis'
/bin/sh: 1: /home/user/projects/tiscamera/external/aravis/autogen.sh: not found
external/CMakeFiles/aravis.dir/build.make:125: recipe for target 'external/aravis-prefix/src/aravis-stamp/aravis-configure' failed
make[2]: *** [external/aravis-prefix/src/aravis-stamp/aravis-configure] Error 127
CMakeFiles/Makefile2:656: recipe for target 'external/CMakeFiles/aravis.dir/all' failed
make[1]: *** [external/CMakeFiles/aravis.dir/all] Error 2
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2

However autogen.sh does exists:
$ ls -l /home/user/projects/tiscamera/external/aravis/autogen.sh
-rwxrwxr-x 1 user user 1289 Jun  2 10:51 /home/user/projects/tiscamera/external/aravis/autogen.sh

$ cmake --version
cmake version 3.19.4

BTW, I updated the repository to the latest master revision.

Maxim

TIS-Stefan, 2021-06-02:
I forwarded the trigger overlap request to the colleagues.
We found, that the DFK 38UX267 is the only DFK 38U camera, that does not export the IMX Low Latency Property by UVC. To be fixed. I will come back to you.

Regarding building: You may delete the build directory content first:
cd build
```
rm * -rf
cmake -DTCAM_ARAVIS_USB_VISION=ON -DBUILD_ARAVIS=ON ..
make -j30
sudo make install
```

(-j30 allows make to use up to 30 threads).

Stefan

TIS-Stefan, 2021-06-02:
Hi Maxim 

One addition: The firmware team states, that Trigger Overlap is not an UVC control. Therefore, it wont be added. 

Stefan

dzhura, 2021-06-03:
Hi Stefan,

Alas, I could not compile tiscamera:
```
$ mkdir build_usb3_vision
$ cd build_usb3_vision
$ cmake -DTCAM_ARAVIS_USB_VISION=ON -DBUILD_ARAVIS=ON ..
...
-- Configuring done
-- Generating done
$ make -j30
....
[ 45%] Performing configure step for 'aravis'
/bin/sh: 1: /home/user/projects/tiscamera/external/aravis/autogen.sh: not found
external/CMakeFiles/aravis.dir/build.make:125: recipe for target 'external/aravis-prefix/src/aravis-stamp/aravis-configure' failed
make[2]: *** [external/aravis-prefix/src/aravis-stamp/aravis-configure] Error 127
CMakeFiles/Makefile2:656: recipe for target 'external/CMakeFiles/aravis.dir/all' failed
make[1]: *** [external/CMakeFiles/aravis.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

TIS-Stefan, 2021-06-03:
Hi Maxim

that is strange. It worked perfectly on my Ubuntu.

I suggest to clone the project again:
git clone https://github.com/TheImagingSource/tiscamera.git tis-usb3vision
```
cd tis-usb3vision
sudo ./scripts/install-dependencies.sh --compilation --runtime
mkdir build
cd build
cmake -DTCAM_ARAVIS_USB_VISION=ON -DBUILD_ARAVIS=ON ..
make -j 30
sudo make install
```

Stefan

dzhura, 2021-06-07:
Hi Stefan,

The cause of the build failure is that my git config had core.autocrlf set to true for some reasons. It changed line endings and broke all the scripts in the repository. So I could build and install tiscamera with aravis. However, I still could not find IMX option in tcam-capture. I looked for this option in the Special tab.

The new firmware will be nice. Will the export of IMX Low Latency Mode be the only change in this update?

Maxim

TIS-Stefan, 2021-06-07:
Hello Maxim

Please send me the output of tcam-ctrl -l

Stefan

dzhura, 2021-06-07:
Hi Stefan

$ tcam-ctrl -l
Model: DFK 38UX267 Serial: 18914001 Type: v4l2
Model: DFK 38UX267 Serial: 18914003 Type: v4l2
Model: DFK 38UX267 Serial: 18914001 Type: aravis
Model: DFK 38UX267 Serial: 18914003 Type: aravis

Maxim

TIS-Stefan, 2021-06-08:
Hello Maxim

that looks very good.
Now you should be able to use
`tcam-ctrl -p 18914001-aravis`
and see much more properties compared with the output of
`tcam-ctrl -p 18914001-v4l2`

Stefan

dzhura, 2021-06-09:
Hi Stefan

Indeed, I get much more properties, including IMX low latency and Trigger Overlap. The most of these properties are not documented. Are they supposed to be available to user?

All these properties are in INVALID_PORPETY group. What does it mean? Is it possible to change IMX low latency and Trigger Overlap properties?

Maxim

TIS-Stefan, 2021-06-09:
Hi Maxim

"invalid property group" means only, that tcam-ctrl does not how to group these properties. But it is possible to change the values. Uhm... Trigger Overlap is readonly.

Do you see the same properties in tcam-capture too?

Stefan

dzhura, 2021-06-10:
Hi Stefan,

Only when I run:
$tcam-capture --serial=18914001-aravis

I do not see these options when run:
$tcam-capture

When change TriggerOverlap, tcam-capture froze. After that tcam-ctrl -p 18914001-aravis stopped to return anything, but tcam-ctrl -p 18914001 does work.

The same happens when use tcam-ctrl --load 18914001-aravis conf.json, where conf.json is output of tcam-ctrl --save with the IMX property set to true and TriggerOveralp set to Off:

$tcam-ctrl --load 18914001-aravis conf.json # runs for quite a while
Loading: aravis.json 
$echo $?
0
$tcam-ctrl --save 18914001-aravis
{
    "properties": null,
    "serial": "18914003",
    "version": "v0.1"
}

Maxim

TIS-Stefan, 2021-06-10:
Hello Maxim

tcam-capture should show the camera twice, once with aravis and once with v4l2. You should be able to select it from the device combobox. Or your can start tcam-capture with "tcam-capture --reset", then you will see the standard device selection dialog.

You can not set Trigger Overlap. That is a read only property and shows only, whether you can trigger now. 

Stefan

dzhura, 2021-06-10:
Hi Stefan,

I can not set IMX low latency trigger mode as well. Is it what you are expecting?

Maxim

TIS-Stefan, 2021-06-11:
Hello Maxim

you should be able to set that as boolean value. Does tcam-capture show it?  However, I learned, that the DFK 38UX267 has an outdated firmware. Please send me a request to support at theimagingsource.com, so I can send the new firmware to you. The zip file will contain the fwpack file, which is the firmware file and the "33update.exe", which is the Windows firmware update program.
After firmware update, the IMX Low Latency Mode is available in v4l2 too.

I wonder, why this IMX Low Latency Mode is so important to you. 

Stefan

dzhura, 2021-06-15:
Hi Stefan,

Thank you a lot for your support! I have sent the request via the contact form on theimagingsource.com. I want to stream pairs of synchronous frames from two cameras, so there is a requirement that frames arrive with low latency and the cameras start exposure at the same time (as much as possible).

Kind Regards,
Maxim

TIS-Stefan, 2021-06-16:
Hello Maxim

I will handle your request as soon as possible.
In v4l2 IMX Low Latency will be available, Trigger Overlay not. In USB3Vision both are available. 
I suppose, you use an external trigger source and have the camera wired to it via the Hirose connector. On software trigger the IMX Low Latency Mode has no effect.

Stefan

closed 2021-06-16 by dzhura