TheImagingSource/tiscamera #411: Building in a virtual environment

Aytoris, 2021-06-10:
Hello,
I am using Ubuntu 18.04 and miniconda (Python 3.8).

I would like to build this repository in a conda virtual environment, in order to have a more standalone Python project using the Python libraries of this repository.

### My first attempt: 
Here are the main parts of my virtual environment's directory tree:
/home/aytoris/condaENV
   '--> bin
   '--> lib
   '--> src
   ` `   '-- > tiscamera
   ` `   '-- > MyPythonProject

To build tiscamera I entered the following commands inside /condaENV/src:

`./scripts/dependency-manager install`
`cd build`
`cmake -DCMAKE_INSTALL_PREFIX=/home/aytoris/condaENV ..`
`make -j8`
`sudo make install`

From there, I create a .py file in MyPythonProject with this piece of code inside:
`import gi`
When I try to execute it from my virtual environment terminal (condaENV) with **python3**, this module is missing. Was I supposed to install it manually? How should I proceed? 

==> _A `pip install PyGObject` does not work (PEP517 error or something) but a `sudo pip install PyGObject` seems to solve this issue for the gi module. Using sudo is not satisfying as I want a standalone project in my virtual environment._

However, I have the same issue with the modules Tcam and Gst.

Nevertheless, when I launch `tcam-capture` from my virtual environment terminal (condaENV), I can correctly see a live stream from my camera.

Thank you in advance for your help

TIS-Edgar, 2021-06-11:
Hi,
We are lacking experience with conda, so please excuse if this may take some back and forth.
Your installation steps appear to be correct.

From my understanding `pip install PyGObject` should satisfy the dependency in the same way the system package `python-gi` should do.

My suspicion is that the local `gi` module does not have the correct search path and thus fails at finding the correct modules.
You can print the path with the following code:
```
gi.require_version('GIRepository', '2.0')
from gi.repository import GIRepository
GIRepository.Repository.get_search_path()
```
You can add to the search path with the followin code:
```
gi.require_version('GIRepository', '2.0')
from gi.repository import GIRepository
GIRepository.Repository.prepend_library_path('/my/custom/path')
```
or by setting an environment variable:
`export GI_TYPELIB_PATH="/usr/lib/x86_64-linux-gnu/girepository-1.0:${GI_TYPELIB_PATH}"`

As to why tcam-capture works when started from you environment ... I don't know.

Can you provide us with a step by step guide on how you set up the environment?
I would like to reproduce this to get a better understanding of what is happening.

Aytoris, 2021-06-11:
Hi Edgar,

Actually the tcam-capture doesn't work from my virtual env, my bad. It also says it need gi. Maybe nothing was actually installed correctly when I did `sudo make install`. Is "gi" installed thanks to this github repository? If that's the case, where can I find the files/python modules?
Or are we expected to install "gi" and other stuff ourselves? I tried `pip install PyGObject` but got this error: 
[image]

Is there an "official" way to keep track of all the packages used to pilot the camera like a requirements.txt from pip ? I also tried with venv but I don't know much about it.

I also tried `pip install pgi` ([PyPi link](https://pypi.org/project/pgi/) & [GitHub link](https://github.com/pygobject/pgi)) but did not get good results.

Aytoris, 2021-06-11:
OK I found a way to use the "gi" module from a virtualenv environment:
`(venv)$ pip install vext`
`(venv)$ pip install vext.gi`

But there's a trick: this pip package allows the venv to USE the BASE SYSTEM's "gi" module. So it's not exactly a standalone venv.

TIS-Edgar, 2021-06-14:
To fix the PyGObject installation error you likely need to install `libcairo2-dev`.

The `gi` python module is an external dependency and not part of tiscamera.
It is a python wrapper around the gobject-introspection c-libraries.
I am uncertain if those can be properly added to a conda setup, so docker might indeed be the better approach.

closed 2021-07-28 by TIS-Stefan

Sn0flingan, 2022-06-08:
> I found a way to use the "gi" module from a virtualenv environment (it should also work on a conda environment): `(venv)$ pip install vext` `(venv)$ pip install vext.gi`
> 
> But there's a trick: this pip package allows the venv to USE the BASE SYSTEM's "gi" module, which need to be ALREADY INSTALLED in the BASE SYSTEM. So it's not exactly a standalone venv.
> 
> --> [GitHub link for Vext and available packages](https://github.com/vext-python/vext#available-packages)
> 
> _A cleaner way to do this might be found by using symlinks, but this would not be visible in the requirements.txt_
> 
> I think I'll give docker a try when I have the time.

Another option is just to extend the sys.path. I prefer this to using Vext since it gives more control over the sys.path. Using Python 3.5 it would look like this:  
`sys.path = sys.path + ['/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages']`